micropython/micropython #8821: Please provide a starting point to find binary releases

harbaum, 2022-06-26:
As discussed in https://forum.micropython.org/viewtopic.php?f=2&t=12594 it would be very helpful for IDEs and the like if there was a starting point for automated download of binary releases. The IDEs could read that URL and parse its contents in order to e.g. allow the user to automatically flash the latest binary.

Using github releases for binaries would give something like that at e.g.  https://api.github.com/repos/micropython/micropython/releases/latest

It was also suggested to use some RSS feed or the like which would work equally well.

It would be nice to have one single entry point providing links or the like pointing to all of the latest binaries.

dpgeorge, 2022-06-26:
The https://micropython.org/download/ page does have everything.  If you follow the page to each board you will get the firmware for all boards.  The names of boards are based on entries in `board.json` from this repo, so that gives you all required information.

But it sounds like you want something simpler, like a flat, single page with a list of all available firmware?  One way would be to provide a .json to download which is a combination of all `board.json` files with links added to available firmware.

harbaum, 2022-06-26:
I am especially searching for something that can reliably and automatically be parsed. Parsing a web page might break if the general layout of the web page changes.

What mechanism creates the download page? Perhaps this might be extended to put a simple Json file to the same location as well.

dpgeorge, 2022-06-27:
> What mechanism creates the download page?

See `tools/autobuild/autobuild.sh` and `tools/autobuild/build-downloads.py`

jimmo, 2022-06-27:
> But it sounds like you want something simpler, like a flat, single page with a list of all available firmware? One way would be to provide a .json to download which is a combination of all board.json files with links added to available firmware.

What I suggested in the forum post was that in the same way we currently generate the HTML for a given board, we could also generate XML (RSS) for that same page. @harbaum the main thing I needed to know was whether you want one feed per board, or one giant feed of everything?

> What mechanism creates the download page? Perhaps this might be extended to put a simple Json file to the same location as well.

The board.json info is sufficient to give you the (expected) URL to the release firmware files. i.e. once you know the board info and a version number, you can construct the URL. But if you need nightlies, then there's no way currently to get the list of nightly build firmware files.

harbaum, 2022-06-27:
> The board.json info is sufficient to give you the (expected) URL to the release firmware files.

I must admit that I don't see any useful information in e.g. ports/esp32/boards/GENERIC/board.json.

Actually the only information I'd need is the version number of the latest release and the datecode to fill the gaps in e.g.
  https://micropython.org/resources/firmware/esp32-{DATECODE}-v{VERSION}.bin

harbaum, 2022-06-27:
I just realized that I might be able to extract that information from https://api.github.com/repos/micropython/micropython/releases/latest

tag_name: "v1.19"
created_at: "2022-06-16T05:11:24Z"

In theory that would give me:

https://micropython.org/resources/firmware/esp32-20220616-v1.19.bin

But this doesn exist as this points to release v1.19 while the binary download are v1.19.1

jimmo, 2022-06-27:
> I must admit that I don't see any useful information in e.g. ports/esp32/boards/GENERIC/board.json.

The key is that the "source of truth" for a given board is that directory that contains board.json. The board.json tells you how to map that to the downloads page (for example, due to historical reasons, the board name might be different -- for example https://micropython.org/download/esp32/ corresponds to esp32/boards/GENERIC (that's why the `id` field is overridden).

We removed v1.19 because it was superceded by 1.19.1 -- there's a corresponding 1.19.1 tag though.

@harbaum if you can describe the exact functionality you need, then we can go about adding it. What is the information you need, and how would you like it to be made available. For example, is RSS necessary? Some of the downloads pages have a bit more structure (i.e. [build variants for pyboard](https://micropython.org/download/pybv11/), or [IDF variants for esp32](https://micropython.org/download/esp32/)). What if it was a json file? Do you also need a way to query for all available boards, or does your use case already know the board names?

As a starting point, would this work for you:
https://micropython.org/download/json/ -- this would be the concatenation of all the board.json files, including the extra fields populated by build-downloads.py
https://micropython.org/download/BOARD/json/ -- this would be the board.json, but with a set of one or more firmware variants, each containing URLs to nightlies and releases.

harbaum, 2022-06-27:
What I am trying to achieve is the following:

I have a Micro-Python IDE which already includes the ability to flash uP binaries onto ESP32 und ESP8266 Boards. The user can select the variant appropriate for his board from a drop down list and then hit the 'flash' button.

Now instead of including the binaries into the IDE I would like to present the user with a choice to download and install the latest uP binary from the official releases.

In oder to do so my IDE needs to be able to automatically find the appropriate latest downloads. One was is to parse the official download page and try to extract the link to the latest releases from that. However parsing that page is not trivial and it would be nice to have an easy to use source of the necessary information.

So how should my IDE find the URL of e.g. the latest GENERIC esp32 binary release?

harbaum, 2022-06-29:
The user base won't hurt me as user run the IDE locally and only github would see the downloads happening.

But still an easy means to get hold of the current version number and the datecode used to generate the latest binaries would be handy.