micropython/micropython #8824: ESP32-C3: machine.freq(240000000) not accepted

scruss, 2022-06-26:
On C3 Mini:
```
import machine
machine.freq(240000000)
```
should succeed, but instead gives a *ValueError*. But
```
import machine
machine.freq(240000000-1)
```
returns *ValueError: frequency must be 20MHz, 40MHz, 80Mhz, 160MHz or 240MHz*

Setting a lower frequency works, but 240 MHz seems unattainable.

Version, implementation, platform:

```
'3.4.0; MicroPython v1.19.1 on 2022-06-18'
(name='micropython', version=(1, 19, 1), _machine='LOLIN_C3_MINI with ESP32-C3FH4', _mpy=6)
'MicroPython-1.19.1--IDFv4.4.1-with-newlib3.3.0'
```

beyonlo, 2022-06-27:
As [Espressif SOCs pages](https://espressif.com/en/products/socs), the ESP32-C3 support max 120MHz:

```
ESP32-C2 Series
32-bit RISC-V MCU & 2.4 GHz Wi-Fi & Bluetooth 5 (LE) 
32-bit RISC-V single-core processor that operates at up to 120 MHz
State-of-the-art power and RF performance
576 KB ROM, 272 KB SRAM (16 KB for cache) on the chip
14 programmable GPIOs: SPI, UART, I2C, LED PWM controller, General DMA controller (GDMA), SAR ADC, Temperature sensor
```

Maybe is a good idea fix on MicroPython port to show just the max clock for that specific espressif device.

dpgeorge, 2022-06-27:
> Maybe is a good idea fix on MicroPython port to show just the max clock for that specific espressif device

Yes, we should adjust `modmachine.c:machine_freq()` so it know the valid clock frequencies for all the esp32 variants.

dpgeorge, 2022-07-08:
Fixed by d7a162516e0a85ae9aa571813bb78428db56a93e

closed 2022-07-08 by dpgeorge