micropython/micropython #8828: mpremote should not echo command when writing file to stdout

bulletmark, 2022-06-28:
Here is an example of the problem:
```
$ mpremote cat main.py
cat :main.py
print('hello world')
```
mpremote should not echo the `cat :main.py` because it corrupts the output of the actual file contents, e.g. if you pipe it somewhere.

MicroPython 1.19.1 on ESP32.

dpgeorge, 2022-06-28:
Thanks for the report.  `cat` was intended to be a simple way to view a file without copying it to a local file.  But I agree that having the header makes difficult to pipe into another command.

Removing the header would make it match unix cat, which I think makes good sense.

(Also, cat doesn't currently work with binary files, that could also be fixed.)

jimmo, 2022-06-28:
Should we remove the header from all commands or just special-case `cat`? Maybe the header should go to stderr instead?

dpgeorge, 2022-06-28:
I think just special case `cat` for now.  It's use with pipes is an obvious use case.

Tico06, 2024-01-26:
Hi There,
I am writing bash script with mpremote and the mount command echoes to stdout which is borring. From my point of view, any command shouldn't echo to stdout exept exec or run if print is used in.
Thanks and regards,
Eric.

andrewleech, 2024-01-26:
@Tico06 your request probably should go on a new issue, this one is quite old and closed, even if it seems related.

That being said, I think you do have a good point. The message saying mount has worked is valuable when repl is active, but fit this kind of non-interactive use it would be better to skip the message.