micropython/micropython #8833: Add raw mode to console output / remove cooked output entirely

smurfix, 2022-06-28:
I find that using a binary protocol to run over the Micropython console is impossible because it is hardcoded to translate `\n` to `\r\n`.

This is just plain annoying.

Also, given that every terminal under the sun is able to do this translation on its own when reading serial text, I submit that the whole cooked-output dance has outlived any usefulness, assuming it ever had any, and should simply be removed from Micropython.

robert-hh, 2022-06-28:
sys.stdin.buffer.read() and sys.stdout.buffer.write() perform raw input/output. Only Ctrl-C must be disabled using micropython.kbd_intr(), because that one is handled at the driver level.

smurfix, 2022-06-28:
*Sigh* never mind, you're right, my code was confused.

closed 2022-06-28 by smurfix