micropython/micropython #8841: Trying to #include "esp_wpa2.h", Failing

MATTYGILO, 2022-06-30:
I'm using an esp32 and creating a custom micropython function, however the project fails to build when trying to #include "esp_wpa2.h". From esp-idf.

How do I #include a function from esp idf.

What do I need to add to cmake and make for this work?

jimmo, 2022-06-30:
Please provide more information -- see the [new issue template](https://github.com/micropython/micropython/issues/new?assignees=&labels=bug&template=bug_report.md&title=) as a guide. Did it work before you added a function, which commands did you execute, which versions of everything, etc.

My guess is you have the wrong IDF version or something wrong with the IDF environment. See ports/esp32/README.md

MATTYGILO, 2022-06-30:
@jimmo I get this error
`fatal error: esp_wpa2.h: No such file or directory`

jimmo, 2022-06-30:
Please see my questions above.

dpgeorge, 2022-07-08:
I'll close this issue because it doesn't have enough information on how to reproduce the problem.

closed 2022-07-08 by dpgeorge