micropython/micropython #8847: [bug]Ble service don't work on esp32c3

coinight, 2022-07-01:
I was trying to make a bluetooth mouse on esp32c3 board.And I try "https://github.com/T-vK/ESP32-BLE-Mouse" (which is working on other esp32 not c3) but when I tried the example code ,an error appeared

***ERROR*** A stack overflow in task ble has been detected.

I don't know how this happened,and how to fix it.It seems that the problem is on RTOS layer.(maybe it's a bug?)
the version of micropython is [v1.19.1-61-g5b66d0860 (2022-06-27)] but i tried [v1.19.1 (2022-06-18)] It also didn't work
hope the problem can be solved
Thanks a lot
[image]

dpgeorge, 2022-07-08:
> I was trying to make a bluetooth mouse on esp32c3 board.And I try "https://github.com/T-vK/ESP32-BLE-Mouse"

That looks like Arduino code.  Did you have an equivalent MicroPython library that you tried?

> which is working on other esp32 not c3

That could be because the C3 is a RISC-V architecture and it might use more stack than xtensa.

coinight, 2022-07-15:
> That looks like Arduino code. Did you have an equivalent MicroPython library that you tried?
>
>
> That could be because the C3 is a RISC-V architecture and it might use more stack than xtensa.

Sorry for wasting your time.I posted a wrong link.It's https://github.com/Heerkog/MicroPythonBLEHID 
(sorry)

coinight, 2022-07-15:
It seems that when the ble server call advertise to fast the overflow happened.

ChihoSin, 2022-08-04:
Same problem when BLE central connected.

[image]

ChihoSin, 2022-08-04:
[image]

It's working when disable some BLE feature, look like nimble event thread can't alloc memory on esp32c3.

iBoundary, 2023-01-31:
@coinight @chihosin @dpgeorge  I also encountered this problem. Have you solved it later?
[image]

Rumidom, 2025-07-02:
same issue, a simple ble scan will cause a crash on the esp32 c3: 

**Code:**
`import bluetooth
import time
import network

ble = bluetooth.BLE()
_IRQ_SCAN_RESULT = const(5)
_IRQ_SCAN_DONE = const(6)

def bt_irq(event, data):
    if event == _IRQ_SCAN_RESULT:
        # Process scan result
        addr_type, addr, adv_type, rssi, adv_data = data
        print("Found device:", bytes(addr),' rssi',rssi)
    elif event == _IRQ_SCAN_DONE:
        print("Scan complete")

ble.active(True)
ble.irq(bt_irq)

# Start scanning
ble.gap_scan(2000)  # Scan for 2 seconds
# Wait for scan to complete
time.sleep(2)
ble.gap_scan(None)
ble.active(False)`

**Output:**
`MPY version : v1.23.0 on 2024-06-02
IDF version : v5.0.4
Machine     : ESP32C3 module with ESP32C3

Guru Meditation Error: Core  0 panic'ed (Store access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x40394f4e  RA      : 0x40394f32  SP      : 0x3fca4190  GP      : 0x3fc97e00  
TP      : 0x3fc69bec  T0      : 0xff3a1000  T1      : 0x00000001  T2      : 0x00000060  
S0/FP   : 0x00000058  S1      : 0x3fca18dc  A0      : 0x0000000c  A1      : 0x00000058  
A2      : 0x00000001  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x3fca1694  
A6      : 0x00000000  A7      : 0x3fc9cbf0  S2      : 0x3fcb8f38  S3      : 0x3fca1694  
S4      : 0x3fca16b8  S5      : 0x3fca16ec  S6      : 0x00000007  S7      : 0x00000000  
S8      : 0x0000001c  S9      : 0x00000001  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x01000000  T4      : 0x00000000  T5      : 0x00000000  T6      : 0x00000000  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x0000000c  
MHARTID : 0x00000000  

Stack memory:
3fca4190: 0x00000000 0x00000000 0x00000058 0x00000004 0x00001800 0x0000005c 0x3fca0000 0xc0380000
3fca41b0: 0x00060000 0x00000000 0x3fca1680 0x403949e8 0x3fcdc710 0x3fcde3f4 0x00000004 0x00000058
3fca41d0: 0x00000001 0x00000000 0x3fca1a30 0x40380966 0x403900a4 0x00000003 0x3fca0000 0x00000030
3fca41f0: 0xffffffff 0x3fc9c9e0 0x00000010 0x00000001 0x00002000 0x00000058 0x00000036 0x40380a48
3fca4210: 0x3fc9ca2b 0x3fc9ca44 0x00000036 0x42091daa 0x00000000 0x3fc9cbf0 0x00000036 0x42092d68
3fca4230: 0x3fcde450 0x3fc9c9e0 0x3fc9d000 0x00000001 0x3fc9ca2b 0x3fc9ca44 0x3fc9c994 0x4209f732
3fca4250: 0x00000000 0x00000001 0x3fc9ca28 0x00000000 0x3fc9ca2b 0x3fc9ca44 0x3fc9c994 0x420a0fb2
3fca4270: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000003 0x3c163b20 0x3fca0000
3fca4290: 0x00001dba 0x42098a1a 0x3c163b20 0x42098a26 0x00001dba 0x42098a1a 0x3fc9fc28 0x42098bec
3fca42b0: 0x3fcde408 0x3fcde408 0x00000004 0x00000002 0x3fcde408 0x3fca0000 0xffffffff 0x420908e4
3fca42d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca42f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca4310: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x000002e4
3fca4330: 0x4038c5ba 0x40001738 0x4212e580 0x40001780 0x4212e612 0x4000179c 0x400017ac 0x40001774
3fca4350: 0x4210d508 0x40001768 0x4000175c 0x40001760 0x40001750 0x40001790 0x40001788 0x400017b0
3fca4370: 0x400017a8 0x400017a0 0x40001758 0x4000176c 0x4000178c 0x40001798 0x40001794 0x40001778
3fca4390: 0x400015cc 0x400017a4 0x40001784 0x400015c8 0x40001620 0x400015f8 0x40001604 0x400015ec
3fca43b0: 0x4212e054 0x4212e220 0x400015e8 0x4000173c 0x40001720 0x400016ac 0x4210b96a 0x4210b788
3fca43d0: 0x4210b83a 0x4210c820 0x40001728 0x40001724 0x400016e4 0x40001700 0x400016b0 0x400016c4
3fca43f0: 0x4000171c 0x400016b4 0x400016fc 0x40001714 0x40001710 0x4212e306 0x4212e456 0x400015fc
3fca4410: 0x400016cc 0x40001600 0x42107f8c 0x4210f920 0x400016e8 0x400016ec 0x400016f0 0x400016f4
3fca4430: 0x400015f0 0x421084cc 0x400016c0 0x400015d0 0x4000161c 0x40001618 0x4000172c 0x4210c1c4
3fca4450: 0x4210cc88 0x42108584 0x42108282 0x400015b4 0x400015c4 0x4210753a 0x400016d4 0x40001718
3fca4470: 0x40001818 0x40001814 0x4210aafe 0x400016a4 0x40001674 0x400016b8 0x400016a8 0x40001740
3fca4490: 0x4210c0ae 0x4212f3d2 0x40001708 0x4210c292 0x400016e0 0x421094a2 0x42109292 0x42108f60
3fca44b0: 0x400017f8 0x40001748 0x400015bc 0x400015c0 0x42107518 0x421074f6 0x4000170c 0x40001734
3fca44d0: 0x40001744 0x40001680 0x4000169c 0x4212fc76 0x4212ec90 0x400016f8 0x400017b8 0x400016c8
3fca44f0: 0x400016bc 0x400017b4 0x400017c8 0x400017bc 0x400017fc 0x4210dbe8 0x4210dbfc 0x40001624
3fca4510: 0x400017c4 0x4210e0c4 0x400015e4 0x42107974 0x42107916 0x400017f4 0x4210dd28 0x400017e8
3fca4530: 0x400017e4 0x400017d4 0x400017d8 0x400017dc 0x400017e0 0x400017cc 0x400017d0 0x40001b2c
3fca4550: 0x400017ec 0x400015d8 0x4000174c 0x40001730 0x40001684 0x40001660 0x421097ba 0x40001654
3fca4570: 0x4000164c 0x421096b0 0x421097aa 0x4000165c 0x40001658 0x400015e0 0x4212f572 0x40001650

ELF file SHA256: 04e0092a8eea4b02

Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x403807a8
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0xe8c
load:0x403cc710,len:0x6ec
load:0x403ce710,len:0x2b24
entry 0x403cc710`