micropython/micropython #8851: Pypi mpy-cross returns zero when it encounters IndentationError.

RSC-Games, 2022-07-02:
I was writing a toolchain in CPython to compile a lot of MicroPython code. I noticed that while compiling, when an IndentationError occurred, mpy-cross returned zero exit status, so my toolchain continued compiling the code even though there were errors. This does not happen with other errors (like SyntaxError).

Minimal Example (hello_world.py):
```

def hello_world():
\tprint("Hello World (Tab indent.)")

\t# Print another hello world.
    print("Hello World (4 spaces indent)")

if __name__ == "__main__":
\thello_world()

```

bash:
```

$ pip install mpy-cross-1.19.1
~snip~
$ python
>>> import os
>>> print(os.system("mpy-cross hello_world.py"))
Traceback (Most recent call last):
  File "hello_world.py", line 4
IndentationError: unindent does not match any outer indent level
0
>>>

```

If possible, could you guys backport the fix to mpy-cross 1.18 because some of my devices are running 1.18 and I do not have access to my micropython development machine right now. If not, then I can just find the time to rebuild my custom micropython
firmware.

dlech, 2022-07-02:
The Python package you are using is probably https://gitlab.com/alelec/mpy_cross, so bugs with that package should be reported there.

Also, it has a Python API, so if you are using it from Python, then you shouldn't call it with `os.system()`, just use the Python API directly.

dpgeorge, 2022-07-05:
Running plain `./mpy-cross` on a file with indentation error does return an error code (value 1) to the shell.  So the problem is not in this repository.

closed 2022-07-05 by dpgeorge

RSC-Games, 2022-07-19:
I only encountered this issue with my implementation on replit.com. When I downloaded the files and ran them on my Windows PC the code worked perfectly.