micropython/micropython #8852: urequests Module missing when compiling Pico-W myself

WaGi-Coding, 2022-07-02:
Hey i just tried to compile the firmware for the PICO_W from current source on Ubuntu. Everything works fine, it shows as Pico W board etc.

However, i cannot use urequests like i can in the official Pico W Build. I followed the Readme files. Did i might missed something?

Greetings

WaGi-Coding, 2022-07-02:
I saw it's part of the micropython-lib repo. But i just wonder how i can include it with the firmware like the official PicoW MP firmware does

daviesian, 2022-07-03:
I had this issue today too, and resolved it as follows: 

You need to clone the `micropython-lib` repo next to the `micropython` repo. So, if you have something like `~/dev/micropython`, you should also have `~/dev/micropython-lib`. Then delete, recreate and reinitialise your build directory and rebuild the firmware from scratch, and the `urequests` package [should be included](https://github.com/micropython/micropython/blob/master/ports/rp2/boards/PICO_W/manifest.py#L7). That worked for me!

WaGi-Coding, 2022-07-03:
> I had this issue today too, and resolved it as follows:
>
> You need to clone the `micropython-lib` repo next to the `micropython` repo. So, if you have something like `~/dev/micropython`, you should [...]

This did the trick. Many many thanks

closed 2022-07-03 by WaGi-Coding