micropython/micropython #8858: Pico W - AP always without password?

WaGi-Coding, 2022-07-04:
Hey there, i want to open a secured access point with MP on the new Pico W like so:
```py
ap = network.WLAN(network.AP_IF)
ap.active(True)
ap.config(essid='PICO', password='12349876')
```

The Acces Point is visible & i can connect to it, but it is not secured.

i tried to pass something like `authmode=network.AUTH_WPA_WPA2_PSK` but network does not have this (i guess) enums & when trying to set it to 3 which should refer to WPA2 PSK, but anyway, `authmode` is not even a valid config parameter, as it spits out the error: `ValueError: unknown config param`

Here is a thread, mentioning that problem as well: https://forums.raspberrypi.com/viewtopic.php?t=336901

Greetings from Vienna

WaGi-Coding, 2022-07-04:
tried to compile MP myself, edited main.c from rp2 port, added `cyw43_wifi_ap_set_auth(&cyw43_state, 3);` This shows as protected then, but idk, i can't connect. Just infinite looping when trying to connect. So it seems to be a deeper lying problem, no clue. Just wanted to share what i found out ^^

WaGi-Coding, 2022-07-04:
> You need to configure the SSID and password before you run `ap.active(True)`. When you activate it, that brings up the access point with the currently-configured values.
> 
> Maybe this is a problem with the documentation?

oh yeh, i thought the same and even tried to first set it to false, wait 5 seconds and then do the config and set it to true, but still the same issue.

The thing is that it's open by default & the c function "cyw43_wifi_ap_set_auth" is never used in the micropython firmware for the PicoW at all, just hardcode it into main.c as mentioned in my other post did not worked, it shows as secured then, but i was not able to connect at all, like the connection request is infinite, not matter what i put as password from user-side

dpgeorge, 2022-07-05:
I found a few problems with setting the authentication mode on rp2:
- a bug in cyw43-driver where the `ap_auth` variable was too small (8 bits) to hold the auth constants; fixed by https://github.com/georgerobotics/cyw43-driver/commit/7a20b316c70896f104cbfd12b8537a7f5b97f487
- a bug in rp2 setup where it didn't configure the default AP mode; fixed by 4b9a2abbdec0e905fbc08f247513e43fb6bff39e
- also added an option to configure the security (auth mode) in d660a0c3d13a5354c8cb10cff90988cff75a33bb

It should now work, at least it works for me (doing the config before the active).

WaGi-Coding, 2022-07-05:
thank you very much, i'll test that in the evening

WaGi-Coding, 2022-07-05:
> I found a few problems with setting the authentication mode on rp2:
> 
> * a bug in cyw43-driver where the `ap_auth` variable was too small (8 bits) to hold the auth constants; fixed by [georgerobotics/cyw43-driver@7a20b31](https://github.com/georgerobotics/cyw43-driver/commit/7a20b316c70896f104cbfd12b8537a7f5b97f487)
> * a bug in rp2 setup where it didn't configure the default AP mode; fixed by [4b9a2ab](https://github.com/micropython/micropython/commit/4b9a2abbdec0e905fbc08f247513e43fb6bff39e)
> * also added an option to configure the security (auth mode) in [d660a0c](https://github.com/micropython/micropython/commit/d660a0c3d13a5354c8cb10cff90988cff75a33bb)
> 
> It should now work, at least it works for me (doing the config before the active).

I tried to apply these changes, pull the latest micropython master, and submodules and applied the changes you mentioned but i am getting this when i want to make `make BOARD=PICO_W`

```
[ 45%] Building C object CMakeFiles/firmware.dir/home/taki/micropython/extmod/network_cyw43.c.obj
In file included from /home/taki/micropython/py/mpstate.h:35,

                 from /home/taki/micropython/py/runtime.h:29,

                 from /home/taki/micropython/extmod/network_cyw43.c:29:

/home/taki/micropython/extmod/network_cyw43.c: In function 'network_cyw43_config':

/home/taki/micropython/extmod/network_cyw43.c:370:45: error: implicit declaration of function 'cyw43_wifi_ap_get_auth'; did you mean 'cyw43_wifi_ap_set_auth'? [-Werror=implicit-function-declaration]

  370 |                 return MP_OBJ_NEW_SMALL_INT(cyw43_wifi_ap_get_auth(self->cyw));

      |                                             ^~~~~~~~~~~~~~~~~~~~~~

/home/taki/micropython/py/obj.h:90:67: note: in definition of macro 'MP_OBJ_NEW_SMALL_INT'

   90 | e MP_OBJ_NEW_SMALL_INT(small_int) ((mp_obj_t)((((mp_uint_t)(small_int)) << 1) | 1))

      |                                                             ^~~~~~~~~

cc1: all warnings being treated as errors

make[3]: *** [CMakeFiles/firmware.dir/build.make:2593: CMakeFiles/firmware.dir/home/taki/micropython/extmod/network_cyw43.c.obj] Error 1

make[2]: *** [CMakeFiles/Makefile2:1387: CMakeFiles/firmware.dir/all] Error 2

make[1]: *** [Makefile:91: all] Error 2

make: *** [Makefile:27: all] Error 2

```

I guess i miss something as usual and should wait for it getting merged^^

dpgeorge, 2022-07-05:
You needed to do `git submodule update` to fetch the latest cyw43-driver repo patches.

> EDIT2: Okay sorry that was my mistake. Started with another clone and compiling works fine. also your fix works well. Again, many many thanks

Great!  Thanks for testing and confirming.

closed 2022-07-05 by dpgeorge