micropython/micropython #8859: PicoW / Pico LED difference

DaveAtKitronik, 2022-07-04:
The PicoW has moved the LED to the WIFI chipset. This is accessible via the following code:
```
import machine
import utime

#Setup the onboard LED Pin
LED = machine.Pin("LED",machine.Pin.OUT)

while True:
	LED.value(True)
	utime.sleep_ms(20)
	LED.value(False)
	utime.sleep_ms(980)
```
The same code should work on the normal Pico I think, but it gives the following error:

```
Type "help()" for more information.
>>> %Run -c $EDITOR_CONTENT
Traceback (most recent call last):
  File "<stdin>", line 5, in <module>
TypeError: can't convert str to int
>>> 
```
Looks like "LED" is in the PicoW machine, but not the Pico one.

PIcoW running 
```
MicroPython v1.19.1-88-g99c258977 on 2022-06-30; Raspberry Pi Pico W with RP2040
```
Pico running 
```
MicroPython v1.19.1 on 2022-06-18; Raspberry Pi Pico with RP2040
```
Both images obtained from Raspberry Pi website this morning.

jimmo, 2022-07-04:
> Pico running `MicroPython v1.19.1 on 2022-06-18; Raspberry Pi Pico with RP2040`

The support for named pins was added when the PicoW was released. This firmware is from 18th June so it will not have it.

Looks like the build server is currently down and so a nightly build with this feature hasn't been made yet... hopefully a new binary should be available from https://micropython.org/download/rp2-pico/ soon.

DaveAtKitronik, 2022-07-04:
Thought it might be a timing of the release issue. 
Will try to build a micropython from the source to make sure. This can probably be closed then.

dpgeorge, 2022-07-04:
Nightly firmware is building now...

DaveAtKitronik, 2022-07-05:
[v1.19.1-90-g3d76292f3 (2022-07-04) .uf2](https://micropython.org/resources/firmware/rp2-pico-20220704-unstable-v1.19.1-90-g3d76292f3.uf2) 
fixes this :)
Closing as there is nothing other than the normal cycle for release  to do for  this.

closed 2022-07-05 by DaveAtKitronik