micropython/micropython #8860: RFC: make `micropython-lib` a submodule of this repository

dpgeorge, 2022-07-04:
This discussion was started here https://github.com/micropython/micropython/pull/8856#issuecomment-1173568845

andrewleech, 2022-07-04:
I think having it as a submodule would increase its visibility and certainly make the existing manifest including simpler, however it would also somewhat bind the releases of packages to the micropython releases, making it harder to release new library package updates. There may end up being a _lot_ of commits here just updating that submodule pin for release 0.0.2 of package blah etc.

As an alternative suggestion, I've been thinking of expending the capabilities of the manifest system. Currently manifest has `freeze` and `include`, how about adding a `require(identifier, src)` which knows how to download / cache / include packages?
It could handle a range of options for `(identifier, src)` like:
* "micropython-urequests", "pypi"
* "os.path", "mpy-lib"
* "https://github.com/openmv/openmv/blob/master/scripts/libraries/urequests.py", "git"

Should add a way of specifying version, at least for source types that support it.

The library packages being intended for use this way could themselves have a manifest.py in their root to specify to the tool which files should be included in a build, and can also `require` their own dependencies etc. Perhaps for a start it'd be worth also adding a function to handle parsing (running) a setup.py if it exists.

This system wouldn't necessarily help people who want to use upip directly on a micropython device (still not sure how best to resolve that) but could be a highly flexible way to specify dependencies for a build.

dpgeorge, 2022-07-04:
Here are the relevant quotes from #8856.

> @iabdalkader : That said, I have a concern with adding modules from micropython-lib to manifests, because that adds a dependency on cloning that repo first. Maybe micropython-lib should be a submodule.

> @mattytrentini : +1
>
> I was recently going to try and explain how urequests is included to respond to https://github.com/micropython/micropython/issues/8852. I knew it was included for ESP32 and have a solid idea how the manifest system works - but then I took a look...and it's quite opaque. Having micropython-lib as a submodule (and tidying up the build system so it's clear which manifest file is called!) would have helped.

> @jimmo : I'm inclined to agree.
>
> So if micropython-lib was a submodule, would we remove the if statements in the manifest scripts to see if the path exists? In other words, if the submodule wasn't initialised, then the build would fail? Should micropython-lib be included in make submodules ?
>
> Also, I sort of like the idea of being able to pin the micropython-lib version, and explicitly moving it forward (i.e. with a release).

> @jimmo : FWIW, longer term what I'd love to do is the equivalent of feature levels for micropython-lib. i.e. a board (or port's default) manifest would include the top-level micropython-lib manifest with a bunch of arguments that include a feature level and supported features (e.g. networking), and that top-level manifest figures out what to include from micropython-lib. That would avoid situations like this where not all boards with wifi automatically get urequests.

> @mattytrentini : Yes, it's not clear why there's a need for both - or why there's also manifest_test.py. Tracing the use of manifest_release.py through from build-boards.sh is not at all obvious either - you need to know that there's a whole separate build system (tools/autobuild). Further, my opinion is that we ought to be building using Github Actions so the build process is clear.

dpgeorge, 2022-07-04:
> Yes, it's not clear why there's a need for both - or why there's also manifest_test.py. Tracing the use of manifest_release.py through from build-boards.sh is not at all obvious either

The use of `manifest_release.py` is to match the builds before the shift to a manifest and the auto-build scripts.  I'm happy to remove this and just have a single `manifest.py` for esp32 that's used in default and release builds.

As for `manifest_test.py` : this is necessary to test freezing of native and viper code on esp32 (xtensawin arch).  It's part of the CI to prevent regressions.

stinos, 2022-07-05:
> There may end up being a lot of commits here just updating that submodule pin for release 0.0.2 of package blah etc

That's a valid concern, but could be fixed by only moving to a new submodule commit with each new release, for instance. Which should be possible becuse there aren't many (?) direct dependencies on micropython-lib.

jimmo, 2022-07-12:
I've raised https://github.com/micropython/micropython/pull/8894 to actually implement this.

We use submodules for all our other dependencies, I think we should use it for micropython-lib. One counter-suggestion was to merge the `micropython` and `micropython-lib` repos. I don't think we should do that, I like the idea of keeping issues/PRs separate as well as repo-level access controls.

dpgeorge, 2022-08-02:
Done in 9a1b7d8448346b353e3aa7ae11ff2b441b6d5313

closed 2022-08-02 by dpgeorge