micropython/micropython #8861: rp2: Add equivalent of stm32's PendSV pending exception dispatch.

jimmo, 2022-07-05:
On (for example) stm32, if you press Ctrl-C a second time and the main thread has not picked up the pending exception from the first one, then it schedules a PendSV interrupt to force an NLR with the pending exception.

On rp2, there's only the regular pending exception for keyboard interrupt, so this cannot interrupt e.g. code from the native emitter. (This was raised in the forum where an example was given where the entire app was emit=native and so could never be interrupted unless it called sleep or something else that did mp_handle_pending).

dpgeorge, 2022-07-05:
stm32's pendsv pending exception was something I wanted to get rid of...!  It feels like it's more complexity than it's worth.  It was there from the very beginning to make ctrl-C work on stm32 before the VM did any polling of a pending exception.  Once polling in the VM was added, this "hard ctrl-C" became less useful.

jimmo, 2022-07-05:
Yes, fair! Definitely agree on reducing complexity. It is likely to be a fairly rare situation where the app basically never runs the VM at all (or any other functions that do mp_handle_pending). In this case I think doing something along the lines of https://github.com/micropython/micropython/issues/8862 would be more worthwhile.

iabdalkader, 2023-04-03:
Just randomly found this issue, and pendsv dispatch has been in `rp2` port for while now, introduced in c001cfa603ef0dcb95196f1942ce2e4d5bd6385b I think you can close this issue.

closed 2023-04-04 by dpgeorge