micropython/micropython #8866: rp2.StateMachine:  state machine fails when I2S protocol is used

miketeachman, 2022-07-05:
A rp2 state machine instance can be made to fail when the I2S protocol is used.

**Minimal test to reproduce**
The code below uses a PIO state machine to flash the onboard LED.

- Working case:  Onboard LED will flash for STATEMACHINE_INSTANCE = 1...7
- Failure case:  Onboard LED will NOT flash for STATEMACHINE_INSTANCE = 0  (shows the Issue)

```
import time
from machine import Pin, I2S
import rp2

STATEMACHINE_INSTANCE = 0

@rp2.asm_pio(set_init=rp2.PIO.OUT_LOW)
def prog():
    pass

sm = rp2.StateMachine(STATEMACHINE_INSTANCE, prog, set_base=Pin(25))
i2s = I2S(0, sck=Pin(16), ws=Pin(17), sd=Pin(18), mode=I2S.TX, bits=16, format=I2S.STEREO, rate=22_050, ibuf=10000)

while True:
	sm.exec("set(pins, 1)")
	time.sleep(0.5)
	sm.exec("set(pins, 0)")
	time.sleep(0.5)
```

In the failure case (STATEMACHINE_INSTANCE = 0) , both the `rp2.StateMachine` instance and the `machine.I2S` instance try to use state machine 0.  Because I2S instantiation follows the `rp2.StateMachine` instantiation, the state machine configuration used by `rp2.StateMachine` is overwritten by I2S initialization.  This causes a failure in the `rp2.StateMachine` instance (LED does not flash).

The root cause of this Issue seems to emerge from the `rp2.StateMachine` implementation.   `rp2.StateMachine` does make use of the functions for cooperative claiming of hardware that are provided by the Pico SDK.  e.g. `pio_sm_is_claimed()`, `pio_sm_unclaim()`,` pio_claim_unused_sm()`.  This does not allow any other MicroPython entities (such as  I2S) to detect state machine configuration conflicts.

**Suggestions for a fix to rp2.StateMachine implementation**:
Enhance the `rp2.StateMachine` implementation to use the Pico SDK functions for cooperative claiming of PIO hardware.  The [I2S class implementation](https://github.com/micropython/micropython/blob/master/ports/rp2/machine_i2s.c#L526) shows one way to use these functions, in the functions `pio_configure()` and `pio_deinit()`.

EliAaron, 2022-07-06:
Thanks for raising the issue. I made a mistake in my original post [Timer gets stuck when thread is running - RP2040](https://forum.micropython.org/viewtopic.php?f=21&t=12639) where I wrote that the I2S stops state machines with ID=0-3, it only stops ID=0. This means that the PIO program memory is not overwritten (state machines 0-3 and 4-7 share the same program memory).

EliAaron, 2022-07-06:
I looked at the rp2 code for the I2S and PIO state machine. It seems the problem is in the PIO code, not the I2S.

In the I2S code, file: _ports/rp2/machine_i2s.c_, function `pio_configure`, there are checks for free state machine and free program memory using the sdk functions `pio_sm_is_claimed` and `pio_can_add_program`. Then `pio_claim_unused_sm` and `pio_add_program` are called.

In the PIO cod, file: _ports/rp2/rp2_pio.c_, `pio_sm_is_claimed` is not used, only `pio_can_add_program` is used in function `rp2_pio_add_program`.

I think two changes to the PIO function `rp2_state_machine_init_helper` can solve the problem:

1. Check for `pio_sm_is_claimed` and raise an error if false
2. Claim the state machine calling `pio_sm_claim`

I’m think these changes need to be made in function `rp2_state_machine_init_helper` but I’m not sure.

I will try to make the changes and test it.