micropython/micropython #8867: esp32 - uart - uasyncio streamreader, always receive b"" from RS485

straga, 2022-07-05:
after commit for uasyncio streamreader, data read from rs485 always b"". If reverse back, start works like before.
https://github.com/micropython/micropython/commit/db7682e02d3ffd3338f20effc9ad4735a48bf774

for example.
```
        self.ac_uart = UART(2, baudrate=9600, tx=14, rx=15)
        self.ac_swriter = asyncio.StreamWriter(self.ac_uart, {})
        self.ac_sreader = asyncio.StreamReader(self.ac_uart)
```

```
      while True:
            data = await self.ac_sreader.read(-1)
            log.info(f" RAW: {data}")
```

dpgeorge, 2022-07-08:
> data = await self.ac_sreader.read(-1)

Were you using this code (passing -1 to read) before db7682e?

Can you show the output of the above code before and after that patch, to see exactly what the difference is?  Also please describe what data is coming in on the RS485 line.

straga, 2022-07-08:
That works for 3 months and more. That real hardware is Solax Inverter and Meter Mbus.
Hmm, very good accident. Maybe UART hardware does pause between data and that work.
That works: read all data before the pause. Like Frame. 

Video: that works before: 
https://user-images.githubusercontent.com/34532/178025340-cac2940b-cf7b-4ddc-b5a3-d530cb654579.mp4

One UARTread request from Invertort - Second send that request to Meter :: Second read the answer from Meter -> One send that to Invertor.
[image]

I am trying to do more tests.

straga, 2022-07-08:
```
class Runner(Load):

    async def _activate(self):

        self.snif_uart = UART(1, baudrate=9600, tx=12, rx=14)
        self.snif_swriter = asyncio.StreamWriter(self.snif_uart, {})
        self.snif_sreader = asyncio.StreamReader(self.snif_uart)

        # launch(self.sniff_process)
        self.snff_old = False
        self.snff_new = False

    async def sniff_old(self):
        while self.snff_old:
            try:
                data = await self.snif_sreader.read(-1)
                log.info(f"OLD - recv: {hexh(data)}")

            except Exception as e:
                log.error(": {}".format(e))

    async def sniff_new(self):
        while self.snff_new:
            try:
                data = await self.snif_sreader.read_all(-1)
                log.info(f"New - recv: {hexh(data)}")

            except Exception as e:
                log.error(": {}".format(e))
```
Test: New that patch - Just nothing get. Same hardware, firmware - I am added second method read_all in asyncio.

```
>>> meter.snff_old = True
>>> launch(meter.sniff_old)
<Task>
>>> INFO:MODBUS:OLD - recv: f4 f5 00 40 13 00 00 01 01 fe 01 00 00 1e 00 00 80 00 00 00 00 00 00 00 01 f2 f4 fb f4 f5 01 40 13 01 00 fe 01 01 01 01 00 1e 00 01 80 00 00 02 15 00 00 00 02 0d f4 fb f4 f5 00 40 0c 00 00 01 01 fe 01 00 00 66 00 00 00 01 b3 f4 fb f4 f5 01 40 49 01 00 fe 01 01 01 01 00 66 00 01 00 00 02 1a 1c 1c 80 80 00 01 01 00 00 00 00 00 00 00 00 c0 00 00 05 00 00 00 00 00 18 17 26 23 00 00 ee 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 05 75 f4 fb f4 f5 00 40 13 00 00 01 01 fe 01 00 00 1e 00 00 80 00 00 00 00 00 00 00 01 f2 f4 fb f4 f5 01 40 13 01 00 fe 01 01 01 01 00 1e 00 01 80 00 00 02 15 00 00 00 02 0d f4 fb f4 f5 00 40 0c 00 00 01 01 fe 01 00 00 66 00 00 00 01 b3 f4 fb f4 f5 01 40 49 01 00 fe 01 01 01 01 00 66 00 01 00 00 02 1a 1c 1c 80 80 00 01 01 00 00 00 00 00 00 00 00 c0 00 00 05 00 00 00 00 00 18 17 26 23 00 00 ee 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 05 75 f4 fb 00 00 02 1a 1c 1c 80 80 00 01 01 00 00 00 00 00 00 00 00 c0 00 00 05 00 00 00 00 00 18 17 26 23 00 00 ee 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 05 75 f4 fb
INFO:MODBUS:OLD - recv: f4 f5 00 40 13 00 00 01 01 fe 01 00 00 1e 00 00 80 00 00 00 00 00 00 00 01 f2 f4 fb
INFO:MODBUS:OLD - recv: f4 f5 00 40 0c 00 00 01 01 fe 01 00 00 66 00 00 00 01 b3 f4 fb
INFO:MODBUS:OLD - recv: f4 f5 00 40 13 00 00 01 01 fe 01 00 00 1e 00 00 80 00 00 00 00 00 00 00 01 f2 f4 fb
INFO:MODBUS:OLD - recv: f4 f5 01 40 13 01 00 fe 01 01 01 01 00 1e 00 01 80 00 00 02 15 00 00 5c 56 fd 7f
INFO:MODBUS:OLD - recv: f4 f5 00 40 0c 00 00 01 01 fe 01 00 00 66 00 00 00 01 b3 f4 fb
INFO:MODBUS:OLD - recv: f4 f5 00 40 13 00 00 01 01 fe 01 00 00 1e 00 00 80 00 00 00 00 00 00 00 01 f2 f4 fb
INFO:MODBUS:OLD - recv: f4 f5 00 40 0c 00 00 01 01 fe 01 00 00 66 00 00 00 01 b3 f4 fb
INFO:MODBUS:OLD - recv: f4 f5 00 40 13 00 00 01 01 fe 01 00 00 1e 00 00 80 00 00 00 00 00 00 00 01 f2 f4 fb
INFO:MODBUS:OLD - recv: f4 f5 00 40 0c 00 00 01 01 fe 01 00 00 66 00 00 00 01 b3 f4 fb
INFO:MODBUS:OLD - recv: f4 f5 00 40 13 00 00 01 01 fe 01 00 00 1e 00 00 80 00 00 00 00 00 00 00 01 f2 f4 fb
INFO:MODBUS:OLD - recv: f4 f5 00 40 0c 00 00 01 01 fe 01 00 00 66 00 00 00 01 b3 f4 fb
INFO:MODBUS:OLD - recv: f4 f5 00 40 13 00 00 01 01 fe 01 00 00 1e 00 00 80 00 00 00 00 00 00 00 01 f2 f4 fb
INFO:MODBUS:OLD - recv: f4 f5 00 40 0c 00 00 01 01 fe 01 00 00 66 00 00 00 01 b3 f4 fb
INFO:MODBUS:OLD - recv: f4 f5 00 40 13 00 00 01 01 fe 01 00 00 1e 00 00 80 00 00 00 00 00 00 00 01 f2 f4 fb
INFO:MODBUS:OLD - recv: f4 f5 00 40 0c 00 00 01 01 fe 01 00 00 66 00 00 00 01 b3 f4 fb
INFO:MODBUS:OLD - recv: f4 f5 00 40 13 00 00 01 01 fe 01 00 00 1e 00 00 80 00 00 00 00 00 00 00 01 f2 f4 fb
meter.snff_old = False
>>>
>>> INFO:MODBUS:OLD - recv: f4 f5 00 40 0c 00 00 01 01 fe 01 00 00 66 00 00 00 01 b3 f4 fb
meter.snff_new = True
>>> launch(meter.sniff_new)
<Task>
>>>
>>>
>>>

```

straga, 2022-07-12:
I was wrong. not always receive b"", just nothing received.

straga, 2022-11-15:
For UART.
If set -1 and read data from RS485. Read 
That means to leave in the while loop forever - Stream.read.

https://docs.micropython.org/en/latest/library/machine.UART.html#class-uart-duplex-serial-communication-bus
A UART object acts like a [stream](https://docs.micropython.org/en/latest/reference/glossary.html#term-stream) object and reading and writing is done using the standard stream methods:

uart.read(10)       # read 10 characters, returns a bytes object
uart.read()         # read all available characters

if read -1:
```
    async def read(self, n=-1):
        r = b""
        while True:
            yield core._io_queue.queue_read(self.s)
            r2 = self.s.read(n)             # 1. read r2 = self.s.read(-1)
            if r2 is not None:              # 2. Yes
                if n >= 0:                      # 3. No
                    return r2
                if not len(r2):                # 4. No
                    return r
                r += r2                         # 5. Yes
                                                    # Return to 1. 
```
r: store all read, a lot byte
[image]

if read 4
```
    async def read(self, n=-1):
        r = b""
        while True:
            yield core._io_queue.queue_read(self.s)
            r2 = self.s.read(n)             # 1. read r2 = self.s.read(4)
            if r2 is not None:              # 2. Yes
                if n >= 0:                      # 3. Yes
                    return r2                   # 5. Exit
                if not len(r2):               
                    return r
                r += r2                                          
```
read(4)
Recieved b'\x01\x03\x00\xac'
Recieved b'\x00\x01D+'
Recieved b'\x01\x03\x02\x01'
Recieved b'28\x01'
Recieved b'\x01\x03\x00L'
Recieved b'\x00\x02\x05\xdc'
Recieved b'\x01\x03\x04\x00'
Recieved b'C\x00\x00\x0b'
Recieved b'\xe7'
Recieved b'\x01\x03\x00\xaf'
Recieved b'\x00\x01\xb4+'
Recieved b'\x01\x03\x02\xff

How read() ? from UART.
```
    async def read_uart(self, n=-1):
        r = b""
        while True:
            yield core._io_queue.queue_read(self.s)
            r2 = self.s.read(n)
            if r2 is not None:
                if n >= 0:
                    return r2
                if n < 0:
                    r += r2
                    return r
```
read_uart(1):
Recieved b'\x01\x03\x00<\x00\x01D\x06'
Recieved b'\x01\x03\x02\xff\xfex4'
Recieved b'\x01\x03\x00\xba\x00\x01\xa5\xef'
Recieved b'\x01\x03\x02\x00\x029\x85'
Recieved b'\x01\x03\x00l\x00\x01D\x17'
Recieved b'\x01\x03\x02\x00\x029\x85'
Recieved b'\x01\x03\x00m\x00\x02U\xd6'
Recieved b'\x01\x03\x04\x02z\x00\x00\xdaR'
Recieved b'\x01\x03\x00\xb2\x00\x01$-'
Recieved b'\x01\x03\x02\x01\x07\xf8\x16'
Recieved b'\x01\x03\x00\xa9\x00\x01T*'
Recieved b'\x01\x03\x02\x0128\x01'
Recieved b'\x01\x03\x00\xac\x00\x01D+'
Recieved b'\x01\x03\x02\x0128\x01'
Recieved b'\x01\x03\x00L\x00\x02\x05\xdc'
Recieved b"\x01\x03\x04\x00B\x00\x00Z'"
Recieved b'\x01\x03\x00\xaf\x00\x01\xb4+'

What right way for that?