micropython/micropython #8868: Formatted string literals cannot handle colon character in string literals

weinand, 2022-07-05:
I'm using `MicroPython v1.19.1-88-g74e33e714 on 2022-06-30; Raspberry Pi Pico W with RP2040`

Running the following expression in the REPL:
```python
print(f"foo {':'} bar")
```
results in this error:
[image]

Running this in Python 3.9.2 works fine:
[image]

jimmo, 2022-07-06:
Our f-string implementation cannot resolve cases where the full lexer+parser is required to understand the expression. We have some heuristics for brackets, and could probably add a heuristic for quotes, but what's happening here is that the colon is triggering the start of the formatting options section.

A terrible workaround is that you can write "\x3a" in place of ":".

We should probably add this particular case to https://docs.micropython.org/en/latest/genrst/core_language.html

dpgeorge, 2022-07-06:
> A terrible workaround is that you can write "\x3a" in place of ":".

That's actually quite a neat workaround!  Unfortunately it doesn't work in CPython.