micropython/micropython #8871: Wemos S2 Mini print() does not output to serial on latest firmware LOLIN_S2_MINI-20220618-v1.19.1.bin

wolfen351, 2022-07-06:
With a the factory default version of micropython the print() statement works fine printing at 115200 to uthe USB/Serial UART. But after updating, the board works fine, and I can use all the great new features in 1.19 i.e. SPI (yay!), but serial print does nothing. REPL still works, as in ampy is happy to put files on the board etc.

When I say it does nothing, I mean there is no output at all. No garbage, no wrong baud data, no indications at all that there is data being streamed from the board to the pc over the usb/serial interface. 

 - firmware file name: LOLIN_S2_MINI-20220618-v1.19.1.bin
 - git commit hash and port/board I don't have the commit hash. The board is an official S2 Mini from Lolin bought via Aliexpress
 - version information shown in the REPL (hit Ctrl-B to see the startup message)

* Remove all placeholder text above before submitting.

It seems to work via ampy, but in "normal mode" when the program is running via main.py there is no output.

```
PS D:\stuff\code\micropython-temp-sensor> echo "print('hi')" > hi.py
PS D:\stuff\code\micropython-temp-sensor> ampy -p COM3 put ./hi.py  
PS D:\stuff\code\micropython-temp-sensor> ampy -p COM3 run hi.py    
hi
```

robert-hh, 2022-07-06:
So in a terminal emulator like putty or teraterm you have the startup-message and the REPL prompt.
If in REPL you submit the line:
`print('Hi there')`
Is that printed? And you say, that if you put that line
`while True: print('hi');`
into a main.py file and reboot via Ctrl-D, it does not run?

wolfen351, 2022-07-06:
Yes, that is accurate. It works via REPL but not via the main.py.

wolfen351, 2022-07-06:
Also just to be clear, everything else on the board works fine, I can flash the led, communicate with SPI with an external device etc.

wolfen351, 2022-07-06:
I apologise - I found the problem. It was my terminal.ps1 program - I triple checked, and if I configure putty correctly, the print() logs appear correctly. Somehow something changed, and my powershell program can no longer read the output, but it is most likely not the device's fault. 

Did something change in the serial protocol? Like is flow control enabled in the newer version?

robert-hh, 2022-07-06:
> Did something change in the serial protocol? Like is flow control enabled in the newer version?

Not that I am aware of. Flow control must be switched off at the PC side.

closed 2022-07-06 by wolfen351