micropython/micropython #8875: Need help to complete PR #8872

ska-la, 2022-07-06:
Please, anybody, help. I have no patient to fix commits formatting. It's a third attempt. I've got advice of Damien George to do:
git push --force
but I don't understand how and where it must be done.

dlech, 2022-07-06:
Change `upstream` to `origin`.

jimmo, 2022-07-06:
> git push --force upstream master

As @dlech said, upstream isn't your fork. What's the name of your remote (likely "origin")

But also you don't want to push to the "master" branch. It likely should be "nucleo_h755zi".

You've already pushed this branch though to create the PR, so if it already has its upstream set, then just "git push -f" should work. Otherwise set its upstream first with "git branch -u origin nucleo_h755zi", then "git push -f".

Note.two distinct meanings of "upstream"... both the name of the remote, and also a branch may have an upstream configured.

ska-la, 2022-07-11:
Thanks, folks. for all tips.

closed 2022-07-11 by ska-la