micropython/micropython #8876: ESP8266 IC2 Crash

tintin10q, 2022-07-06:
I am getting the same error as in this issue: https://github.com/micropython/micropython/issues/1996

```
from machine import I2C, Pin
i2c = machine.I2C(freq=9600,sda=Pin(4), scl=Pin(5))
i2c.scan()
```

This code causes a crash.

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 30720, room 16 
tail 0
chksum 0x90
load 0x3ffe8000, len 996, room 8 
tail 12
chksum 0x80
ho 0 tail 12 room 4
load 0x3ffe83f0, len 1080, room 12 
tail 12
chksum 0xf5
csum 0xf5
```
There is also this garbage text see the image:

[image]

 - firmware file name: [v1.19.1 (2022-06-18) .bin](https://micropython.org/resources/firmware/esp8266-20220618-v1.19.1.bin)
 - version information shown in the REPL: MicroPython v1.19.1 on 2022-06-18; ESP module (1M) with ESP8266

Does anyone know what is going on? I have nothing connected to pin 4 and 5.

peterhinch, 2022-07-07:
See https://github.com/micropython/micropython/issues/1996#issuecomment-1177193026

dpgeorge, 2022-07-08:
As mentioned by @peterhinch, you need to connect pull-ups to pin 4 and 5 to get it to work.

The problem is that the `i2c.scan()` is taking a very long time because the frequency is very low (9600) and there are no pull-ups so each scan attempt times out.  This long delay means the watchdog kicks in and resets the MCU.

One possible fix in the firmware is to add a `MICROPY_EVENT_POLL_HOOK` in the `machine_i2c_scan()` function.  That prevents the watchdog from biting, and also allows ctrl-C to break the scan (which is a problem on all ports with SoftI2C).

tintin10q, 2022-07-08:
Thank you for the suggestions. However by now I have switched to an esp32 and SPI communication instead and that worked for me.

dpgeorge, 2022-08-31:
Fixed by 0b26efe73dd3396bdc2b77651a78d9f2edeb9004, which calls `MICROPY_EVENT_POLL_HOOK` during the scan.

closed 2022-08-31 by dpgeorge