micropython/micropython #8877: Add inet_pton and inet_ntop to CYW43 driver

twitch0001, 2022-07-06:
Hello! 
Recently received some Pico W devices and I have been experimenting with MDNS and Service discovery, during this development I noticed that both `socket.inet_pton` & `socket.inet_ntop` are not available for the Pico W/RP2 port. 

From a quick search it seems like that the CYW43 driver does not have support for this (unless my poor understanding of C has let me down...). 

If someone would be able to add support for `socket.inet_pton` & `socket.inet_ntop` to the rp2 driver that would be greatly appreciated!
I will give this a go but I understand someone more competant in C and MicroPython may be able to implement this feature.

Many thanks!

dpgeorge, 2022-07-07:
Can you give an example of how you would use these functions?  MicroPython can't implement everything because it won't fit on a microcontroller, so functions like these are missing by design.

If they were to be implemented, it would be in `extmod/modlwip.c`.

sosi-deadeye, 2022-07-08:
The low-level functions can be done with Micropython.
If you want to process only IPv4-Adresses, the task is easy.

```
def inet4_ntop(packed_ip: bytes) -> str:
    return ".".join(map(str, packed_ip))

def inet4_pton(ip: str) -> bytes:
    return bytes(int(c) for c in ip.split("."))
```

IPv6 is a bit complicated:
```
def inet6_ntop(packed_ip: bytes) -> str:
    """
    ipv6 address is 128 bits long
    each block has 16 bits / 2 bytes and
    is represented in hex and splitted via colon
    """
    iterator = zip(*[iter(packed_ip)]*2)
    ipv6_addr = []
    for high, low in iterator:
        ipv6_addr.append(f"{high << 8 | low:04x}")

    return ":".join(ipv6_addr)

def inet6_pton(ip: str) -> bytes:
    packed_ip = []
    for block in ip.split(":"):
        value = int(block, 16)
        packed_ip.append(value >> 8)
        packed_ip.append(value & 0xFF)

    return bytes(packed_ip)
```

All examples do not have Exception handling and do not check if the packed_ip/ip is valid. The IPv6-Version do not handle the two-colon notation ::