micropython/micropython #8896: Arduino RP2 Nano connect: Unexpected OSError (regression)

peterhinch, 2022-07-11:
With today's [nightly build](https://micropython.org/resources/firmware/ARDUINO_NANO_RP2040_CONNECT-20220708-unstable-v1.19.1-113-g9af6a275d.uf2) this occurs
```
MPY: soft reboot
MicroPython v1.19.1 on 2022-07-08; Arduino Nano RP2040 Connect with RP2040
Type "help()" for more information.
>>> import network
>>> s = network.WLAN(network.STA_IF)
>>> s.active(True)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: Firmware version mismatch. Minimum supported firmware is v1.5.0 found v1.4.8
```
Previously I was running a build dated 22 April: this error did not occur.

It's also worth noting that the support for MSC mode means that firmware installation by powering up with the button depressed does not work. The docs are therefore misleading to new users: it now seems mandatory to use **machine.bootloader()**.

peterhinch, 2022-07-11:
I think it's also fair to say that the process of updating the Nina firmware is difficult. In April I managed to install V1.4.8 but so far V1.5.0 is defeating me.

With MicroPython installed it seems to be necessary to use the command line version. I downloaded the latest version of the firmware uploader, but this isn't working for me:
```bash
$ sudo ./arduino-fwuploader firmware flash -b arduino:mbed_nano:nanorp2040connect -a /dev/ttyACM0 -m NINA@1.5.0
rp2040load 1.0.1 - compiled with go1.15.8
.....................
error during sketch flashing: exit status 1
rp2040load 1.0.1 - compiled with go1.15.8
.....................
error during sketch flashing: exit status 1
rp2040load 1.0.1 - compiled with go1.15.8
```
It produces these errors forever. I have checked that the serial port exists.

robert-hh, 2022-07-11:
I used a special sketch for rp2 and esptool. See https://github.com/micropython/micropython/pull/8654, and search for `SerialNinaPassthrough`. Not a really comfortable method, but it worked for NINA binaries built by myself.

iabdalkader, 2022-07-11:
This is not a bug, the updated firmware introduced a new sockets ABI (to fix all the issues that have been reported here), which is now used by the updated host driver, so it can't work with firmware older than `1.5.0`. That said, I do agree the update process is not easy, and I am thinking about re-implementing it in MicroPython, but can't promise it will be any time soon.

> It's also worth noting that the support for MSC mode means that firmware installation by powering up with the button depressed does not work. The docs are therefore misleading to new users: it now seems mandatory to use machine.bootloader().

I don't think this was ever possible before MSC (that's sounds more like the RPi Pico). There's a REC pin that you can pull low (or high) and then power on the board, but also you can get into the bootloader by double-pressing the reset button (with MicroPython running on the board).

>  Could we make this error message have a short link to the above docs? Is there any way to make a MicroPython app that can do Nina FW update?

@jimmo I don't mind adding that, but I've never seen links in exception messages before, so I'm not sure if that's acceptable, if you think it's okay I'll send a PR.

peterhinch, 2022-07-12:
>  I do agree the update process is not easy

This is an understatement, and it's got harder since I last did it in April.

Method 1 listed [here](https://docs.arduino.cc/software/ide-v2/tutorials/ide-v2-fw-cert-uploader#firmware-updater) has never worked in my testing. It fails with the classic: "something went wrong". Method 2 successfully downgraded Nina to V1.4.8 but didn't offer an option to upgrade to V1.5.0. My experience with the command line approach was better in that it enabled me to install an April version of MP with Nina V1.4.8 however all attempts to install more recent Nina or MP have failed.

The following is my experience of using the CLI to upgrade  to the latest nightly build with Nina 1.5.0. The initial state was a working build `ARDUINO_NANO_RP2040_CONNECT-20220422-unstable-v1.18-370-g28e7e15c0` with Nina 1.4.8.

Link rec-gnd and power up
Install blink sketch
Remove link and power cycle. Blink sketch doesn't appear to do anything.
Run
```bash
$ sudo ./arduino-fwuploader firmware flash -b arduino:mbed_nano:nanorp2040connect -a /dev/ttyACM0 -m NINA@1.5.0
```
This appeared successful.
Power down.
Link rec-gnd
Power up, a Thunar window opens.
Drag latest nightly build to window.
Power cycle
Board appears dead. Port /dev/ttyACM0 appears but I can't connect to it with mpremote or rshell.

I have struggled with this for some hours. The only way I can get a usable MP is to downgrade Nina to V1.4.8 and to downgrade MP to 22April. It is not enough just to downgrade MP, I have to downgrade both.

By contrast the hot-off-the-press RP2 Pico W "just works". I'm surprised that Arduino have released something so uniquely hard to use.

I'll try the method of @robert-hh ...

jimmo, 2022-07-13:
> @jimmo I don't mind adding that, but I've never seen links in exception messages before, so I'm not sure if that's acceptable, if you think it's okay I'll send a PR.

@iabdalkader Agree, but this is a bit of a unique case and it's not at all clear from the current message what the user can do about it.

However, I think it would be more worthwhile to add a way to update the firmware from MicroPython.

How big is the Nina firmware bin? If the device has the flash space, having a way to just copy the file to the filesystem, then adding a method to the network.WLAN object to flash-from-file would be ideal.

iabdalkader, 2022-07-13:
> How big is the Nina firmware bin? If the device has the flash space, having a way to just copy the file to the filesystem, then adding a method to the network.WLAN object to flash-from-file would be ideal

The combined firmware image is ~1.3MBs, so it should fit in the flash storage, and the firmware update can be implemented via `ioctl`s I think. That's actually how our WINC-1500 firmware update works (microchip/atmel provide a Win `.bat` file to update the firmware), it's just the matter of allocating time for this.

iabdalkader, 2022-07-20:
FYI, I was just re/testing from the git [1.5.0](https://github.com/arduino/nina-fw/releases/tag/1.5.0) release, and followed these steps with no issues:

* Enter bootloader mode (`machine.bootloader()` or double reset)
* Upload `SerialNINAPassthrough` from Arduino IDE.
* Reset the board.
* Run the following command:
```python
python ~/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyACM0 --baud 115200 --before no_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x0 NINA_W102-v1.5.0-Nano-RP2040-Connect.bin
```
* When it's done, reset the board, copy back micropython firmware.uf2.

Note still looking into the possibility of implementing this in MicroPython, but for now the above works.

Output from `esptool.py`:
```bash
Serial port /dev/ttyACM0
Connecting.....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: xx:xx:xx:xx
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 2MB
Compressed 1129472 bytes to 671360...
Wrote 1129472 bytes (671360 compressed) at 0x00000000 in 59.4 seconds (effective 152.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

iabdalkader, 2022-10-19:
>  I think it would be more worthwhile to add a way to update the firmware from MicroPython.

@jimmo  see #9677

iabdalkader, 2022-11-27:
I forgot to post an update here, so I added a new tool in `micropython-lib` called [espflash.py](https://github.com/micropython/micropython-lib/tree/master/micropython/espflash) it's a minimal implementation of the ESP-32 bootloader protocol for MicroPython, and froze it in the Nano-RP2040 firmware. The Nina WiFi firmware update process should now be incredibly easy, you just need to either copy the firmware binary to MSC, and run the example, or if you prefer to use `mpremote`, copy to same directory and run something like:
```bash
mpremote mount . run example.py
```
Note edit the file name (in `path` in the script) if needed, and the MD5 checksum, or set `md5sum` to `None` it will be calculated.