micropython/micropython #8906: Can't set hostname in AP mode

LucaDevelop, 2022-07-13:
**Board**: Raspberry Pico W
**Firmware file name**: rp2-pico-w-20220713-unstable-v1.19.1-129-g5bf376563.uf2

I can't set the "hostname" param with config() function.

This works (without hostname param):
```
import network
wlan = network.WLAN(network.AP_IF)
wlan.config(essid='somessid', key="1234567890")
wlan.active(True)
```

This doesn't work:
```
import network
wlan = network.WLAN(network.AP_IF)
wlan.config(essid='somessid', key="1234567890", hostname="somehostname")
wlan.active(True)
```
and prints this error:
```
Traceback (most recent call last):
  File "<stdin>", line 3, in <module>
ValueError: unknown config param
```

I've tried with "dhcp_hostname" also, same result.

iabdalkader, 2022-07-16:
I'd like to send this in, if no one is currently working on it, along with some proposed changes to `connect`.

scruss, 2022-08-08:
Although this was merged in [CYW43: Allow setting hostname. by iabdalkader · Pull Request #8918 · micropython/micropython](https://github.com/micropython/micropython/pull/8918), it doesn't seem to have made it into the rp2-pico-w-20220808-unstable-v1.19.1-241-g6baeded32.uf2 nightly

robert-hh, 2022-08-08:
@iabdalkader: 
Trying to build the binaries with `MICROPY_PY_NETWORK_CYW43_USE_LIB_DRIVER=0` fails.
How did you succeed and test setting hostname?

jimmo, 2023-02-01:
See https://github.com/micropython/micropython/pull/10635

maik123100, 2023-02-21:
It seems the last  time i found it was mentioned is that you should "dhcp_hostname" instead of "hostname" to set the hostname. I found it [here](https://github.com/micropython/micropython/issues/8792) and the concret link would be [this](https://github.com/micropython/micropython/pull/8831).
I might be completly mistaken but from what i saw this should resolve the issue. I am eager to hear how it turned out for you. But from my point of view it seems the [doc](https://docs.micropython.org/en/latest/library/network.WLAN.html?highlight=hostname#network.WLAN) is not 100% correct on this.

PR #10635 (2023-02-28): top: Common network hostname/country configuration.
The overall goal is to allow configuration of the hostname on Pico W, but also unify the way networking works across ports. Fixes #8906, #10397

This PR adds `network.hostname()` and `network.country()` to globally set the device hostname (for all interfaces) and country (for radio compliance). For interfaces that previously supported `nic.config(hostname)` this will continue to work, as will `pyb.country()` (and `rp2.country()`).

Additionally configure the default hostname on a per-port and per-board basis. We may want to remove this change [... cut]
closed 2023-02-28 by dpgeorge