micropython/micropython #8908: No REPL when building for Nucleo-WB55 with LTO enabled

Hendrik-K, 2022-07-14:
Hello,

I've tried building MicroPython for the Nucleo-WB55 with LTO enabled, as this was recently added.
Unfortunately the board does not display the REPL when connecting.

I'm not familiar with the limitations of LTO and the consequences, I just wanted to try it out to save some space.
Maybe I'm doing something wrong? The only thing I can think of is that my version of the toolchain causes this?
I used the GNU Arm Embedded Toolchain 7-2018-q2-update for building.

jimmo, 2022-07-14:
It would be good if you could try with a more modern toolchain. I will test here with gcc 12.1 on a wb55 nucleo and report back.

dpgeorge, 2022-07-15:
@projectgus Anything we can do with docs or compile-time warnings to help this situation?

projectgus, 2022-07-15:
Did a quick poll of a few ARM toolchain versions, in case this was some bug that only shows up sometimes, but can confirm that newer gcc is consistently fine:

OK: my local gcc 12.1, ARM gcc 10-2020-q4, 10.3-2021.07
Not OK: ARM gcc 7-2018-q2 gcc, 8-2018-q4, 9-2019-q4, 9-2020-q2

Quick search around and it looks like this bug: https://bugs.launchpad.net/gcc-arm-embedded/+bug/1747966 that was in the ARM toolchain that was only fixed in their gcc 10 release, which is consistent with what is shown here!

I applied the workaround mentioned in comment 10 (put startup files first in the link) and it works now even with the old gcc 7 2018-q2. Will submit a PR for that, although there might be some implications to changing the linker order that I've missed.

> Anything we can do with docs or compile-time warnings to help this situation?

@dpgeorge  Yes, I think a build time warning can be done quite easily (I actually have one implemented here, but then I decided to check some more compiler versions and noticed the linked bug). If it's OK with you then I suggest we look at adding the workaround first, and if any additional LTO issues in older gcc show up then add the warning as well.

projectgus, 2022-07-15:
I believe it's safe to put these object files (SRC_O) first in the link order (at least modulo any other linker bugs like this one!)

dpgeorge, 2022-07-18:
Fixed by 1d23c01dd6d2bcf64dec86ceed77db7ecac8a8d9

closed 2022-07-18 by dpgeorge