micropython/micropython #8921: esp32-c3 1.17 and 1.18 compile error

puppet13th, 2022-07-18:
i'm trying to compile firmware(1.17 and 1.18) for esp32-c3 port with these setup:
- Windows 10 with WSL(ubuntu)
- firmware tar.gz downloaded from github release(1.17/1.18/1.19.1)
- downloaded esp-idf version 4.2.3/4.3.3/4.4.1

command line used when compiling firmware 1.19.1 + esp-idf 4.3.3:
- source ~/idf/esp-idf-v4.3.3/export.sh
- make -C mpy-cross
- make -C ports/esp32 BOARD=GENERIC_C3

ai-thinker esp32-c3s + firmware 1.19.1 + esp-idf 4.3.3:
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0xf18
load:0x403ce000,len:0x6e4
load:0x403d0000,len:0x292c
entry 0x403ce000
MicroPython v1.19.1 on 2022-07-17; ESP32C3 module with ESP32C3
Type "help()" for more information.
MicroPython v1.19.1 on 2022-07-17; ESP32C3 module with ESP32C3

Type "help()" for more information.
>>> import platform
>>> platform.platform()
'MicroPython-1.19.1--IDFv4.3.3-dirty-with-newlib3.3.0'
```

using same setup i tried to compile firmware 1.17 and 1.18 and encounter these error:
```
[1309/1338] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/__/machine_pin.c.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/__/machine_pin.c.obj
/home/puppet13th/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc -DFFCONF_H=\"/home/puppet13th/micropython-1.17/lib/oofatfs/ffconf.h\" -DLFS1_NO_ASSERT -DLFS1_NO_DEBUG -DLFS1_NO_ERROR -DLFS1_NO_MALLOC -DLFS1_NO_WARN -DLFS2_NO_ASSERT -DLFS2_NO_DEBUG -DLFS2_NO_ERROR -DLFS2_NO_MALLOC -DLFS2_NO_WARN -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DMICROPY_ESP_IDF_4=1 -DMICROPY_MODULE_FROZEN_MPY="(1)" -DMICROPY_QSTR_EXTRA_POOL=mp_qstr_frozen_const_pool -DMICROPY_VFS_FAT=1 -DMICROPY_VFS_LFS2=1 -D__DBINTERFACE_PRIVATE=1 -Dvirt_fd_t="void*" -Iconfig -I/home/puppet13th/micropython-1.17 -I/home/puppet13th/micropython-1.17/lib/berkeley-db-1.xx/PORT/include -I../ -I../boards/GENERIC_C3 -I. -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/newlib/platform_include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/freertos/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/freertos/include/esp_additions/freertos -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/freertos/port/riscv/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/freertos/include/esp_additions -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/esp_hw_support/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/esp_hw_support/include/soc -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/esp_hw_support/include/soc/esp32c3 -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/esp_hw_support/port/esp32c3/. -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/esp_hw_support/port/esp32c3/private_include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/heap/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/log/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/lwip/include/apps -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/lwip/include/apps/sntp -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/lwip/lwip/src/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/lwip/port/esp32/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/lwip/port/esp32/include/arch -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/soc/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/soc/esp32c3/. -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/soc/esp32c3/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/hal/esp32c3/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/hal/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/hal/platform_port/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/esp_rom/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/esp_rom/include/esp32c3 -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/esp_rom/esp32c3 -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/esp_common/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/esp_system/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/esp_system/port/soc -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/esp_system/port/include/riscv -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/esp_system/port/public_compat -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/riscv/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/driver/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/driver/esp32c3/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/esp_pm/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/esp_ringbuf/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/efuse/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/efuse/esp32c3/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/vfs/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/esp_wifi/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/esp_event/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/esp_netif/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/esp_eth/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/tcpip_adapter/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/esp_phy/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/esp_phy/esp32c3/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/esp_ipc/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/app_trace/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/esp_timer/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/app_update/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/spi_flash/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/bootloader_support/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/bt/common/osi/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/bt/include/esp32c3/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/bt/common/api/include/api -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/bt/common/btc/profile/esp/blufi/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/bt/common/btc/profile/esp/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/bt/host/nimble/nimble/porting/nimble/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/bt/host/nimble/port/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/bt/host/nimble/nimble/nimble/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/bt/host/nimble/nimble/nimble/host/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/bt/host/nimble/nimble/nimble/host/services/ans/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/bt/host/nimble/nimble/nimble/host/services/bas/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/bt/host/nimble/nimble/nimble/host/services/dis/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/bt/host/nimble/nimble/nimble/host/services/gap/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/bt/host/nimble/nimble/nimble/host/services/gatt/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/bt/host/nimble/nimble/nimble/host/services/ias/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/bt/host/nimble/nimble/nimble/host/services/ipss/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/bt/host/nimble/nimble/nimble/host/services/lls/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/bt/host/nimble/nimble/nimble/host/services/tps/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/bt/host/nimble/nimble/nimble/host/util/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/bt/host/nimble/nimble/nimble/host/store/ram/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/bt/host/nimble/nimble/nimble/host/store/config/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/bt/host/nimble/nimble/porting/npl/freertos/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/bt/host/nimble/esp-hci/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/nvs_flash/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/mbedtls/port/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/mbedtls/mbedtls/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/mbedtls/esp_crt_bundle/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/mdns/include -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/console -I/mnt/d/mpy/IDF/esp-idf-v4.4.1/components/sdmmc/include -march=rv32imc -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Wno-error=format= -nostartfiles -Wno-format -Os -freorder-blocks -fmacro-prefix-map=/home/puppet13th/micropython-1.17/ports/esp32=. -fmacro-prefix-map=/mnt/d/mpy/IDF/esp-idf-v4.4.1=IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.4.1-dirty\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -Wno-clobbered -Wno-deprecated-declarations -Wno-missing-field-initializers -MD -MT esp-idf/main/CMakeFiles/__idf_ma [... cut ...]
3fcc4930: 0x3fc90f80 0x37354443 0x37354400 0x2e85a308 0x3f22f984 0x2e85a300 0x248d3132 0x43800000
3fcc4950: 0x7cf0bdc2 0x40800000 0x4b000000 0x3eec5b0c 0x3fbb16c3 0x3805ff67 0x3af135b4 0x3cda40e4
3fcc4970: 0x3e15dce6 0x3ea6cc7a 0x3e5c245a 0x3d9e233f 0x36f5d7bd 0x3a4beed6 0x3c98bf54 0x3e300f6e
3fcc4990: 0x3f38d0c5 0x3fb22d3b 0xbad5c4e8 0x3a5b3dd2 0x3a1c065c 0x3a500cfd 0x3daaaaab 0x3ed67f1d
3fcc49b0: 0x37d383a2 0x39679767 0x3a9c54a1 0x3bf2027e 0x3d89f001 0x383c2c75 0x38e28445 0x3a05b634

ELF file SHA256: 8277442616582117

Rebooting...
```

please advice.

dpgeorge, 2022-07-18:
> i'm trying to compile firmware(1.17 and 1.18) for esp32-c3 port

Why do you need to compile older versions?

You may have more luck getting it working using an older IDF, like 4.3.2 or 4.4.

puppet13th, 2022-07-18:
look like esp32-c3 ports ble does not play nice with uasyncio.please check [forum thread](https://forum.micropython.org/viewtopic.php?f=18&t=12696&sid=60f9d580ae78f6ce21754ae000a8a12b)

#8628 have issue with ble too.

i have tried to run ble test code(uploaded at forum) with 1.17 firmware too,and it runs flawlessly. but esp32-c3 1.17 firmware does not have enough ram to run my full code.

hopefully u can understand the problem that i described.

i will try to compile with IDF 4.3.2 then.

puppet13th, 2022-07-18:
i have managed to compiled 1.17/1.18 firmware using IDF 4.3.2 without modify 'ports/esp32/machine_pin.c' and the firmware boot normally.it's time to test which firmware is more suitable for my code. @dpgeorge thank your very much for the pointer.

closed 2022-07-23 by dpgeorge