micropython/micropython #8924: Upgrading from 1.18 to 1.19 doesn't work with Dfuse [Pyboard lite 1.0)

gudnimg, 2022-07-18:
* **Board:** Pyboard Lite v1.0
* **DFU file:** [v1.19.1 (2022-06-18) .dfu](https://micropython.org/resources/firmware/pyblitev10-20220618-v1.19.1.dfu)
* **DFU programmer:** Dfuse v3.0.6
    * I followed the instructions here: https://micropython.org/resources/Micro-Python-Windows-setup.pdf (Last two pages)
* **Operating System:** Windows 11 

The flashing works according to Dfuse and no errors are reported. But upon exiting the DFU mode, the REPL still shows the Pyboard is using Micropython 1.18

[image]

I have never upgraded Micropython before so If there is something missing in the documentation I have no idea.

gudnimg, 2022-07-18:
> Depending on the Micro Python version you upgrade to you may need to restore the flash file
system to get the latest boot.py and pybcdc.inf. See the Micro Python tutorial “Safe mode and
factory reset”.

I've tried this and used this documenation https://docs.micropython.org/en/latest/pyboard/tutorial/reset.html Unfortunately it doesn't seem to change anything for me.

jimmo, 2022-07-19:
Can you share some screenshots of dfuse before, during, and after the flashing process? Also try checking the "Verify after download" option.

jimmo, 2022-07-19:
The other thing to try would be to use the newer tool from ST -- https://www.st.com/en/development-tools/stm32cubeprog.html

This tool (confusingly) uses DFU mode, but last time I used it required a .hex file. Recent releases also provide a .hex file on https://micropython.org/download/pyblitev10/

gudnimg, 2022-07-19:
It worked!

* Used STM32CubeProgrammer instead of Dfuse
* Used `.hex` file instead of `.dfu`

Note: When the board enters DFU mode, `USB` needs be selected here:
[image]

Then hitting the blue `Download` button takes care of updating the firmware.

gudnimg, 2022-07-19:
Since my issue is fixed I'm closing this Github issue 🎉 

Another issue could be created to update the documentation to replace deprecated Dfuse with STM32CubeProgrammer?

closed 2022-07-19 by gudnimg