micropython/micropython #8928: webrepl: Standardise on all networking-capable boards

jimmo, 2022-07-19:
Follow-up to https://github.com/micropython/micropython/pull/8893#issuecomment-1181751258

We need to decide:
 - Should we freeze the webrepl Python files on all boards that include the webrepl C support (which is now controlled by `MICROPY_PY_NETWORK`)
 - or. Make webrepl something that can be installed optionally (upip, etc) and document that, and leave it to specific boards that want to freeze it.

(Following our general principle of only including the bare minimum by default, I have a weak preference for the second option, especially if we make package installation more seamless... see notes in #8914)

jimmo, 2022-07-19:
> As the person writing the getting started documentation, I have a medium-strong preference for this to be available by default on firmware built for Pico W.

Sure. Out of curiosity how are you documenting the requirement that webrepl is accessed over http, but this is basically impossible right now except from a local copy? (#5266)
We need to fix this of course, either by hosting webrepl on the device or by making the device use wss websockets, but I'm not sure either will be soon.

aallan, 2022-07-19:
I'm not documenting it at all right now. 🤷‍♂️

jimmo, 2022-07-20:
> I'm not documenting it at all right now. 🤷

Unfortunately it's a bit of a show-stopper to the whole webrepl feature...

I had a go at solving it based on the idea I suggested in #5266 and if we can get this merged for 1.20 then perhaps you could document that instead.

lurch, 2022-07-20:
Just to clear up any potential confusion, this is the "getting started documentation" that @aallan is referring to: https://datasheets.raspberrypi.com/picow/connecting-to-the-internet-with-pico-w.pdf (which is linked to from e.g. https://www.raspberrypi.com/products/raspberry-pi-pico/ amongst other places).

PR #10592 (2023-02-01): top: Make all network-capable boards use bundle-networking.
This will ensure that any board with networking support gets:
 - webrepl
 - mip
 - urequests
 - ntptime

Fixes #8928 #8961

This PR has the following effect:
esp32: No change
esp8266: No change
mimxrt-all-boards-with-ethernet: Adds mip,webrepl,urequests,ntptime (@robert-hh)
stm32-PYBD: Adds mip, urequests, ntptime
stm32-Portenta: Adds mip (@iabdalkader)
stm32-all-other-boards-with-ethernet: Adds mip,webrepl,urequests,ntptime
rp2-nano-connect: Adds mip (@iabdalkader)
rp2-wiznet-boards: Adds mip,webrepl,urequests,ntptime
rp2-pico-w: Add [... cut]
closed 2023-02-01 by dpgeorge