micropython/micropython #8937: extmod/uasyncio: gather gets stuck if you pass an empty list in v1.19

caligotori, 2022-07-20:
Hello, folks

        This is pretty straightforward in v1.17 and v1.18 passing an empty list to `gather` will return (or yield) whereas in v1.19 it gets stuck for ever. Not sure if this is a new intended behavior (which is not in the docs?). I also haven't checked the diffs, but there are a lot of changes to uasyncio related to how tasks are cancelled.

Cheers,

jimmo, 2022-07-20:
Thanks @caligotori for the report. I can see how this broke due to the cancellation improvements. Will send a PR to fix and add test.

AmirHmZz, 2022-12-26:
@jimmo I think this issue should be closed as completed due to https://github.com/micropython/micropython/pull/8939