micropython/micropython #8938: W5500-EVB-PICO:  setblocking, SOL_SOCKET, SO_REUADDR NOT working; Cannot use Asyncio start_server()

UncleBee1885, 2022-07-20:
I'm currently working with the [Wiznet W5500-EVB-PICO](https://www.wiznet.io/product-item/w5500-evb-pico/). For those who aren't familiar, it's essentially a Raspberry Pi Pico and a [Wiznet Ethernet Hat](https://www.wiznet.io/product-item/wiznet-ethernet-hat/) combined onto a single board and, as the name implies, gives you a wired ethernet connection.

I'm currently using the following firmware: [v1.0.5](https://github.com/Wiznet/RP2040-HAT-MicroPython/releases/tag/v1.0.5)
Version information from the REPL:  
> MicroPython v1.18-10-g5db278f1d-dirty on 2022-05-10; Raspberry Pi Pico with RP2040

### I'm trying to use Asyncio to start a server. 

When I run [this code](https://github.com/UncleBee1885/bug_examples/blob/main/async_test_server.py)  I receive the following error(s):

> Task exception wasn't retrieved
> future: <Task> coro= <generator object 'start_server' at 2000d700>
> Traceback (most recent call last):
>   File "uasyncio/core.py", line 1, in run_until_complete
>   File "uasyncio/stream.py", line 1, in start_server
> OSError: [Errno 107] ENOTCONN

For reference, similar versions of the above code have worked on both an ESP8266 and Raspberry Pi Pico W.
The Raspberry Pi Pico W version that I've used can be found [here](https://gist.github.com/aallan/3d45a062f26bc425b22a17ec9c81e3b6#file-async_webserver-py).

Upon further inspection it seems that using the following seem to be causing issues:

- **socket.setblocking(False)**
- **socket.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)**

To this hypothesis, I used some known-good [example code](https://github.com/UncleBee1885/bug_examples/blob/main/HTTP_Server.py) for a small non-asynchronus HTTP server and added the following lines of code (one at a time) to see if I could invoke the error.
`s.setblocking(False)`
`s.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)`

With "s.setblocking(False)" uncommented on line 51, the following error occurs:

>Traceback (most recent call last):
  File "<stdin>", line 80, in <module>
  File "<stdin>", line 51, in main
OSError: [Errno 107] ENOTCONN

With "s.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)" uncommented at line 52, the following error occurs:

> Traceback (most recent call last):
  File "<stdin>", line 80, in <module>
  File "<stdin>", line 52, in main
AttributeError: type object 'socket' has no attribute 'SOL_SOCKET'

It seems that [start_server()](https://github.com/micropython/micropython/blob/master/extmod/uasyncio/stream.py#L155) uses both of these lines of code and, they don't seem to be compatible with the W5500 wrapper/driver that is used in this firmware. 

I respectfully request that someone look into this issue. I'll provide any additional information that is required. I'm a hardware guy who recently started doing software so I'll try my best.

dpgeorge, 2022-07-21:
Thanks for the comprehensive report.

The main issue is that you are using firmware from an external source, ie not from this repo directly.  The first thing you should do is switch to a firmware built directly from this repo, then it will most likely work.

Right now we don't have a board definition for the W5500-EVB-PICO, but I think @jimmo is working on one.

omogenot, 2022-07-22:
Hi,
I allow myself to intervene there as I succeeded in building the firmware (1.19.1 as of the time of writing) for W5500_EVB_PICO board, and the SO_REUSEADDR is working.
I wrote a linux shell script to setup the environment:
```
#! /bin/bash
# Get lastest Micropython
echo "Get latest micropython"
git clone -b master https://github.com/micropython/micropython.git
cd micropython
# Make SDK environment
echo "Get submodules"
make -C ports/rp2 submodules
make -C ports/rp2 BOARD=W5100S_EVB_PICO submodules
echo "make tools"
make -C mpy-cross                                                 
# Create new board
echo "Create new W5500 evb pico board"
mkdir ports/rp2/boards/W5500_EVB_PICO
cp -r ports/rp2/boards/W5100S_EVB_PICO/. ports/rp2/boards/W5500_EVB_PICO/.
sed -i  's/W5100S/W5500/' ports/rp2/boards/W5500_EVB_PICO/board.json
sed -i  's/w5100s/w5500/' ports/rp2/boards/W5500_EVB_PICO/board.json
sed -i  's/W5100S/W5500/' ports/rp2/boards/W5500_EVB_PICO/mpconfigboard.cmake 
sed -i  's/W5100S/W5500/' ports/rp2/boards/W5500_EVB_PICO/mpconfigboard.h
```

In order to create the WS5500_EVB_PICO firmware, use the following command:
`make BOARD=W5500_EVB_PICO`

After that you can find the file `firmware.uf2` in the `micropython/ports/rp2/build-W5500_EVB_PICO` directory.

Hope this helps,

Olivier.

UncleBee1885, 2022-10-10:
> Hi, I allow myself to intervene there as I succeeded in building the firmware (1.19.1 as of the time of writing) for W5500_EVB_PICO board, and the SO_REUSEADDR is working. I wrote a linux shell [...]

Since the release of the official board definition for the W5500, I've been unable to insert my own MAC address using a method that had previously worked using the non-official firmware: [https://github.com/Wiznet/RP2040-HAT-MicroPython/releases/tag/v1.0.5](url) .  

Any ideas on how to change the MAC address?

Using the W5500 firmware built from this repository, it writes a MAC Address of 02:9A:0F:74:61:9A which is NOT the factory default nor is it the address I'm trying to program.

omogenot, 2022-10-11:
@UncleBee1885 ,
Since the version you are referring to, I suppose that the driver has been updated from the so-called `wiznet` stack to the `lwIP` stack. As a consequence some functions have changed a bit.
Specifically the mac address management which now can be accessed using the `config` method of the network interface:
```
>>> import network
>>> nic = network.WIZNET5K()
>>> nic.config("mac")
b'\x02\x9a\x9fta\x9a'
>>> nic.config(mac=b"\x04\x9a\x0F\x00\x23\x56")
>>> nic.config("mac")
b'\x04\x9a\x0f\x00#V'
```
Here we go !
Hope this helps.

AWilco, 2023-04-03:
I had the same bug using the v2.0.0 release from the Wiznet https://github.com/Wiznet/RP2040-HAT-C repo. Switching to 1.91.1 for Wiznet 5100s from the official repo fixed the issue for me.