micropython/micropython #8943: Add ESPNow/WiFi support

davefes, 2022-07-22:
For the last 6 months I have been running an ESPNow repeater link for water tank level monitoring and a portable ESPNow link for antenna testing. The later required RSSI capability, which was recently included.
 
I have done quite a bit of testing on the [firmware-esp32-GENERIC.bin] v1.19.1 image, specifically the ESPNow/WiFi feature, to give added capability to two datalogging applications that currently use mainline. I have another 2 links that I will migrate to ESPNow/WiFi.

I will be able to reduce the number of ESP32s, that I have scattered about the property, thanks to having ESPNow/WiFi capability.

I believe that providing ESPNow/WiFi in mainline would be a big boost for IoT applications.

Applicable to boards with WiFi-capability.

straga, 2022-07-24:
I too used ESPNow, it works very well. For RS485 Modbus Gateway.

AmirHmZz, 2022-07-26:
I think any **updates**/**questions**/**issues**/**bugs** about ESPNow implementation in Micropython would be better to be in #6515.

davefes, 2022-07-26:
I'll consider re-posting there.

closed 2022-07-26 by davefes