micropython/micropython #8948: docs: Small formatting problem

lurch, 2022-07-22:
The text at the bottom of https://docs.micropython.org/en/latest/library/index.html#extending-built-in-libraries-from-python isn't quite rendering correctly:
[image]
Note how the backticks are showing up in the last paragraph, instead of `import module` and `import umodule` being displayed in code-style formatting.

I don't know enough about RST to fix this myself, but the corresponding source is at https://github.com/micropython/micropython/blob/master/docs/library/index.rst#extending-built-in-libraries-from-python
```
This applies to both the Python standard libraries (e.g. ``os``, ``time``, etc),
but also the MicroPython libraries too (e.g. ``machine``, ``bluetooth``, etc).
The main exception is the port-specific libraries (``pyb``, ``esp``, etc).

*Other than when you specifically want to force the use of the built-in module,
we recommend always using ``import module`` rather than ``import umodule``.*
```

dlech, 2022-07-22:
I don't think restructured text allows nested styles. The easiest fix is probably just to remove the `*`s from the last line.

Carglglz, 2022-07-22:
Or just add a few more `*`s:

This applies to both the Python standard libraries (e.g. ``os``, ``time``, etc),
but also the MicroPython libraries too (e.g. ``machine``, ``bluetooth``, etc).
The main exception is the port-specific libraries (``pyb``, ``esp``, etc).

*Other than when you specifically want to force the use of the built-in module,
we recommend always using* ``import module`` *rather than* ``import umodule``.

```
This applies to both the Python standard libraries (e.g. ``os``, ``time``, etc),
but also the MicroPython libraries too (e.g. ``machine``, ``bluetooth``, etc).
The main exception is the port-specific libraries (``pyb``, ``esp``, etc).

*Other than when you specifically want to force the use of the built-in module,
we recommend always using* ``import module`` *rather than* ``import umodule``.
```