micropython/micropython #8949: ESP32 machine.SDCard() Instantiation throws error on soft reboot: SPI bus already initialized; ESP_ERR_INVALID_STATE

jamesquilty, 2022-07-23:
We're having trouble with microSD card access with MicroPython versions 1.16 through 1.19 on two different revisions of a single-board PCB based on the ESP32-WROVER-B (SPI RAM) with direct connections from the ESP32 pins to the pins of a [MEM2075 microSD card connector](https://www.mouser.com/datasheet/2/837/MEM2075-2887971.pdf) (PDF). On first power-up, or after a hard reset, the card can be mounted on the REPL with the commands 

```python
from machine import SDCard

# PCB Revision A pin configuration:
sd = SDCard(slot=2, sck=13, miso=15, mosi=23, cs=12)

# PCB Revision B pin configuration:
sd = SDCard(slot=2, sck=13, miso=34, mosi=23, cs=12)
```

The cards can be accessed normally. Once a soft reboot is executed, the code to instantiate the SDCard() class results in an error. An example from a pristine MicroPython 1.19.1 ([esp32spiram-20220618-v1.19.1.bin](https://micropython.org/resources/firmware/esp32spiram-20220618-v1.19.1.bin) from <https://micropython.org/download/esp32spiram/>) on a Revision A board:

```python
MicroPython v1.19.1 on 2022-06-18; ESP32 module (spiram) with ESP32
Type "help()" for more information.
>>> from machine import SDCard
>>> sd = SDCard(slot=2, sck=13, miso=15, mosi=23, cs=12)
>>>
MPY: soft reboot
MicroPython v1.19.1 on 2022-06-18; ESP32 module (spiram) with ESP32
Type "help()" for more information.
>>> from machine import SDCard
>>> sd = SDCard(slot=2, sck=13, miso=15, mosi=23, cs=12)
E (xxxxx) spi: spi_bus_initialize(462): SPI bus already initialized.
E (xxxxx) sdspi_host: spi_bus_initialize failed with rc=0x103
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: (-259, 'ESP_ERR_INVALID_STATE')
```

where we are not using `sd.deinit()` because the [MicroPython documentation says `sd.deinit()` is only implemented on the cc3200 port](http://docs.micropython.org/en/latest/library/machine.SD.html).

 This problem does not appear to affect re-instantiating `SDCard()` after wake from deep sleep, only soft reboot. That said, access to the microSD card is an integral part of remote operation of our device and we feel this problem blocks us from migrating to MicroPython 1.19.1, keeping us on version 1.14. This is, needless to say, a pretty serious problem for us.

 We're aware that we're not using the slot 2 pins recommended in [class SDCard - ESP32](http://docs.micropython.org/en/latest/library/machine.SDCard.html#esp32) but we're uncertain whether changing the pin connections on the PCB will resolve the problem. We're using every pin on the ESP32, so changing the microSD card connections will be a non-trivial task. We'll do it if we know it will resolve the `SPI bus already initialized` error.

 We were using MicroPython 1.14 ([esp32spiram-idf4-20210202-v1.14.bin](https://micropython.org/resources/firmware/esp32spiram-idf4-20210202-v1.14.bin)) when the boards were developed and did not, and do not, see this error with microSD card access that we see with later versions. The issue causing the errors above appears to have been introduced in version 1.16 of the firmware. Hopefully this provides a clue to the change(s) causing the problem and the potential resolution?

 Previous GitHub Issues and MicroPython Forum posts of similar errors on instantiating `SDCard()` after soft reboot coincide in time with the release of version 1.16:

- SD card not accessible from rshell. SD card not mounted after soft_reset <https://forum.micropython.org/viewtopic.php?f=18&t=10882&p=59841>
- Best practice for remounting sdcard and checking availability <https://forum.micropython.org/viewtopic.php?f=2&t=11205&p=61967>
- Oddities with SD <https://forum.micropython.org/viewtopic.php?f=16&t=12622&p=68456>
- #7414 _micropython 1.16 SD won't mount after umount_ <https://github.com/micropython/micropython/issues/7414>
- #7352 _micropython 1.15 esp32 reboot at SD deinit_, comment 863908895 <https://github.com/micropython/micropython/issues/7352#issuecomment-863908895>
- #4722 _sdcard esp32 does not work TTGO T8 V1.8_ <https://github.com/micropython/micropython/issues/4722>

jamesquilty, 2022-08-24:
@dpgeorge, in #8994 you wrote:
> ... if code expects to run after a soft reset then it must be aware that hardware peripherals can be in a variety of states when MicroPython code starts running.

How does that work in practice for `machine.SDCard()`, please? The previously instantiated object has been lost to the reset, trying to create a new object throws an error... there doesn't appear to be any way to reconnect to a previously mounted card.

dpgeorge, 2022-08-26:
I have tried to test this scenario and it works OK for me, the SDCard can be recreated after a soft reset.  The point is that the SDCard class implements the `__del__` destructor, so when a soft reset occurs the GC will sweep any SDCard instances and automatically deinit them (there is actually a `SDCard.deinit()` method on esp32).

I tested on a Lolin32 Pro which has an ESP32-Wrover module on it.  It has an SD card slot on it.  Here is the REPL session which does a few soft resets and recreates the SD card:
```
MPY: soft reboot
MicroPython v1.19.1-332-g09879f99c on 2022-08-26; ESP32 module with ESP32
Type "help()" for more information.
>>> import machine
>>> sd=machine.SDCard()
>>> sd.info()
(1977614336, 512)
>>> import os
>>> os.mount(sd, '/sd')
>>> machine.SDCard()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: (-259, 'ESP_ERR_INVALID_STATE')
>>> 
MPY: soft reboot
MicroPython v1.19.1-332-g09879f99c on 2022-08-26; ESP32 module with ESP32
Type "help()" for more information.
>>> import machine
>>> sd=machine.SDCard()
>>> sd.info()
(1977614336, 512)
>>> import os
>>> os.mount(sd, '/sd')
>>> 
MPY: soft reboot
MicroPython v1.19.1-332-g09879f99c on 2022-08-26; ESP32 module with ESP32
Type "help()" for more information.
>>> import machine
>>> sd=machine.SDCard()
>>> sd.info()
(1977614336, 512)
>>> 
```

jamesquilty, 2022-08-26:
@dpgeorge Thanks for trying to reproduce the problem! The [LOLIN D32 Pro](https://doc.riot-os.org/group__boards__esp32__wemos-lolin-d32-pro.html) connects the ESP32 to the microSD card reader by the default pins for slot 2 according to the [periph_conf.h File Reference](https://doc.riot-os.org/boards_2esp32-wroom-32_2include_2periph__conf_8h.html). We're using non-recommended pins for `sck`, `miso` and `cs`:

```python
# MicroPython default (and preferred) pin configuration:
sd = SDCard(slot=2, sck=18, miso=19, mosi=23, cs=5)

# PCB Revision B pin configuration:
sd = SDCard(slot=2, sck=13, miso=34, mosi=23, cs=12)
```

and we reproduce the problem by using the same command sequence you used (transcript below) but we can't recover with `sd.deinit()`.

 I think this narrows the problem to how the `__del__` destructor handles configurations with non-standard pins, and I'm _guessing_ that a change to how  the SDCard class handles this in was made in a version subsequent to MicroPython 1.14. I'm hoping that this will be easier and quicker to fix in the MicroPython firmware than it would be to change the ESP32 connections and PCB routing?

```python
MicroPython v1.19.1 on 2022-06-18; ESP32 module (spiram) with ESP32
Type "help()" for more information.
>>> import machine
>>> sd = machine.SDCard(slot=2, sck=13, miso=34, mosi=23, cs=12)
>>> sd.info()
(31914983424, 512)
>>> import os
>>> os.mount(sd, '/sd')
>>> sd = machine.SDCard(slot=2, sck=13, miso=34, mosi=23, cs=12)
E (25440) sdspi_host: Bus already initialized. Call `sdspi_host_init_dev` to attach an sdspi device to an initialized bus.
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: (-259, 'ESP_ERR_INVALID_STATE')
>>>
MPY: soft reboot
MicroPython v1.19.1 on 2022-06-18; ESP32 module (spiram) with ESP32
Type "help()" for more information.
>>> import machine
>>> sd = machine.SDCard(slot=2, sck=13, miso=34, mosi=23, cs=12)
E (40080) spi: spi_bus_initialize(462): SPI bus already initialized.
E (40080) sdspi_host: spi_bus_initialize failed with rc=0x103
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: (-259, 'ESP_ERR_INVALID_STATE')
>>> sd.deinit()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'sd' isn't defined
```

```python
MicroPython v1.19.1 on 2022-06-18; ESP32 module (spiram) with ESP32
Type "help()" for more information.
>>> import machine
>>> sd = machine.SDCard(slot=2, sck=13, miso=34, mosi=23, cs=12)
>>> sd.info()
(31914983424, 512)
>>> sd.deinit()
>>>
MPY: soft reboot
MicroPython v1.19.1 on 2022-06-18; ESP32 module (spiram) with ESP32
Type "help()" for more information.
>>> import machine
>>> sd = machine.SDCard(slot=2, sck=13, miso=34, mosi=23, cs=12)
E (35730) spi: spi_bus_initialize(462): SPI bus already initialized.
E (35730) sdspi_host: spi_bus_initialize failed with rc=0x103
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: (-259, 'ESP_ERR_INVALID_STATE')
```

jamesquilty, 2022-08-26:
OK, so doing a bit of digging in the git history of [ports/esp32/machine_sdcard.c](https://github.com/micropython/micropython/blob/master/ports/esp32/machine_sdcard.c) I found that the known-working 1.14 release shipped commit 1662a0b06f2ad58aa4a632a88d0b3836c08bded2 and there have been three subsequent commits containing changes to machine_sdcard.c. Commit 4eb13c50cd87eff8374b9bffc3d7c15b1bf5e35f makes changes to SD bus deinit and shipped in 1.16 - the first version that exhibits the problems we see.

So, commit 4eb13c50cd87eff8374b9bffc3d7c15b1bf5e35f is strongly implicated as the cause of the problem. It's a rather simple commit, so hopefully it will be easy to fix deinit when `SDCard` is instantiated with non-recommended pin configurations?

dpgeorge, 2022-08-26:
@jamesquilty what IDF version do you use?  I use 4.4.1.

jamesquilty, 2022-08-26:
We're not using the ESP-IDF or building custom versions of the firmware at present, we're using the binary from the download page <https://micropython.org/download/esp32spiram/> which says it's "compiled using ESP-IDF v4.x".

Would it help resolve the problem with `SDCard()` if we built a custom version of the firmware?

dpgeorge, 2022-08-29:
I have been able to reproduce this problem.  The point is one needs to use the SPI bus (eg `slot=2`), not SDIO bus (which is the default).

It definitely does look related to 4eb13c50cd87eff8374b9bffc3d7c15b1bf5e35f and #7352.

dpgeorge, 2022-08-30:
See #9147 for a fix.

dpgeorge, 2022-08-30:
Should be fixed by d108fc9c4793e1846b2ddfad925fb17bbc44cd51

@jamesquilty please test with the latest "unstable" release on the download page (but you may need to wait about 6 hours for it to appear).

closed 2022-08-30 by dpgeorge

jamesquilty, 2022-09-01:
@dpgeorge, commit [d108fc9](https://github.com/micropython/micropython/commit/d108fc9c4793e1846b2ddfad925fb17bbc44cd51) does indeed fix the error on soft reboot, tested with [esp32spiram-20220831-unstable-v1.19.1-358-g0b26efe73.bin](https://micropython.org/resources/firmware/esp32spiram-20220831-unstable-v1.19.1-358-g0b26efe73.bin) from the [ESP32 with SPIRAM](https://micropython.org/download/esp32spiram) page. This fixes the biggest problem, thanks! 

We're still seeing some oddness with microSD cards, for example a Core panic (which I'll raise separately) and more importantly some cards, specifically the Samsung EVO Plus 32GB,  fail to mount on our Revision B PCB while they mount fine on Revision A PCB:

```python
# PCB Revision A pin configuration:
sd = SDCard(slot=2, sck=13, miso=15, mosi=23, cs=12)

# PCB Revision B pin configuration:
sd = SDCard(slot=2, sck=13, miso=34, mosi=23, cs=12)
```

SanDisk 32 GB cards mount fine on both revisions. I'd like to clarify a couple of things, please.

The Lolin unit which you mention above appears to be different to the [Lolin D32](https://doc.riot-os.org/group__boards__esp32__wemos-lolin-d32-pro.html) which appears to use the SPI bus `slot=2` pins to connect to its microSD card reader. Your Lolin uses `slot=1` (default) and the SDIO bus Is that correct?

In principle I think we can revise our PCB to use the SDIO bus (`slot=1`), but is there any advantages to doing so compared to using the SPI bus with the recommended `slot=2` pins?

dpgeorge, 2022-09-13:
> specifically the Samsung EVO Plus 32GB, fail to mount on our Revision B PCB while they mount fine on Revision A PCB:

The difference looks like MISO which is now pin 34.  Pin 34 is input-only so may have different electrical properties to the input/output pins.  That could be the cause of the problem you see.

> The Lolin unit which you mention above appears to be different to the [Lolin D32](https://doc.riot-os.org/group__boards__esp32__wemos-lolin-d32-pro.html) which appears to use the SPI bus `slot=2` [...]

I was testing the code using both the SDIO bus and the SPI bus.

> In principle I think we can revise our PCB to use the SDIO bus (`slot=1`), but is there any advantages to doing so compared to using the SPI bus with the recommended `slot=2` pins?

Using SDIO should be faster because it uses a wider bus for data transfer (4 bits maybe?).  Also the SDIO protocol might be better supported by certain SD cards.  I would suggest to use SDIO if possible.