micropython/micropython #8952: Resquest Slave-mode be added to micropython (#3935 follow-up)

mmliam, 2022-07-23:
My personal interest has been for a gateway between communication protocols. I've been reduced to using a UART on the ESP32 for inter-unit communication, now limited to two protocols. The slave mode would allow for the creation of a multi-protocol gateway, with each protocol hosted on a separate unit.

In addition it would permit inter-unit communication between multiple ESP8266 NodeMCU's, which is now complicated by being limited to the single UART dedicated to USB.

beyonlo, 2022-07-26:
I'm interested on this feature as well.

kwagyeman, 2023-10-09:
@dpgeorge @robert-hh

We need to add this feature to the STM32 and MIMXRT so that can port our interface libraries to the machine module. I am willing to do the work. Is it possible you can think of what the API changes would be? It should just be a parameter added to the init call. If you were able to give some clarity on this I can make it happen.

quic-DevJoshi, 2024-04-24:
Hello @mmliam,

I hope this message finds you well. I’m reaching out to you regarding your potential use case for an I2C slave. I’m currently have a preliminary version of an I2C slave working for the rp2 port, and I am hoping to integrate this support into MicroPython.

My current implementation is tailored to my specific needs, which involve an 8-byte register map shared between a MicroPython script and the kernel. The Interrupt Service Routine (ISR) for I2C continuously services I2C Master's requests and updates this register map, and the MicroPython script can then manipulate the data within the map. This setup essentially mimics a TI fuel gauge.

However, I’m interested in developing a more versatile solution that could cater to a broader range of use cases. To achieve this, I’m gathering information on potential applications for an I2C slave. If you could share your use case, it would very helpful.

I hope my explanation is clear, and I look forward to hearing from you.

jonnor, 2024-08-11:
This issue seems to be a duplicate of #3935

closed 2024-08-11 by mattytrentini