micropython/micropython #8963: RP2: Pico W still hanging shortly after start of main.py when connecting to wif

peterharperuk, 2022-07-26:
Follow up to https://github.com/micropython/micropython/issues/8904
We still seem to have a problem when running some scripts in main.py that connect to an access point when powered by USB o a mac.
It looks like we missed the main source of delays cyw43_delay_ms. Putting the hook in there seems to improve things.

dpgeorge, 2022-07-27:
My guess is that the original patch where the slow poll hook was used was changing the timing of when things occurred on the Pico W vs on the Mac.

closed 2022-07-27 by dpgeorge

Deltafire, 2022-10-11:
I've just encountered this problem while running a slightly modified version of the sample webserver as main.py. The webserver works, but the USB won't initialise. Output from dmesg (Linux):

> [ 4061.576633] usb 1-3: new full-speed USB device number 70 using xhci_hcd
> [ 4061.576891] usb 1-3: Device not responding to setup address.
> [ 4061.784992] usb 1-3: Device not responding to setup address.
> [ 4061.992638] usb 1-3: device not accepting address 70, error -71
> [ 4061.992869] usb usb1-port3: unable to enumerate USB device

Adding sleep(1) to the start of the script seems to fix it.

Using firmware rp2-pico-w-20221011-unstable-v1.19.1-532-g8e912a501.uf2.

lurch, 2022-10-12:
@Deltafire "The webserver works" sounds like a very different problem to "Pico W still hanging", so it might be best to create a new issue and include your example code? :wink:

commit 45ab801c (2022-07-27): rp2/cyw43_configport: Add event hook into cyw43_delay_ms.

Still see some USB issues apparently caused by delays loading wifi
firmware.  cyw43_delay_ms is used to wait in the driver, so we should call
the event hook in there.

Fixes #8963.