micropython/micropython #8978: use argparse in mpremote

dlech, 2022-07-28:
`mpremote` doesn't handle invalid args gracefully. This could be improved by using `argparse` instead of manually parsing `sys.argv`.

For example, I forgot the second arg for `cp` and got an `IndexError` crash:

```
$ ./tools/mpremote/mpremote.py a1 cp ports/stm32/build-LEGO_HUB_NO6/firmware.dfu.gz 
Traceback (most recent call last):
  File "/home/david/work/pybricks/micropython/./tools/mpremote/mpremote.py", line 6, in <module>
    sys.exit(main.main())
  File "/home/david/work/pybricks/micropython/tools/mpremote/mpremote/main.py", line 573, in main
    do_filesystem(pyb, args)
  File "/home/david/work/pybricks/micropython/tools/mpremote/mpremote/main.py", line 338, in do_filesystem
    pyboard.filesystem_command(pyb, args, progress_callback=show_progress_bar, verbose=verbose)
  File "/home/david/work/pybricks/micropython/tools/mpremote/mpremote/../../pyboard.py", line 583, in filesystem_command
    if srcs[0].startswith("./") or dest.startswith(":"):
IndexError: list index out of range
```

It was difficult to figure out what I did wrong to trigger the error.

dpgeorge, 2022-07-29:
I'm not sure `argparse` can support the way `mpremote` handles arguements, that it has chained commands?  It may be simpler just to handle it explicitly.

jimmo, 2023-06-05:
We did this!

closed 2023-06-05 by jimmo