micropython/micropython #8982: Please Add 'add_done_callback' to ASyncIO?

Jibun-no-Kage, 2022-07-29:
Please Add 'add_done_callback' to ASyncIO?

This was added in Python 3.7 and allows for ASyncIO created tasks to self clean up when needed.

peterhinch, 2022-07-29:
Please could you clarify what this offers over:
```python
async def foo():
    try:
        # code
    finally:
        done_callback()
```

sosi-deadeye, 2022-07-29:
This should run with CPython (tested with 3.10.5) and Micropython (v1.19.1).
The goal is to run the **finished** method and printing the **result** of the finished Task instance.

If `add_done_callback` is implemented, we don't need the `add_done_callback` helper function.

Another quirk is the different way how the result can be accessed.
Micropython saves the result in attribute `data`. (simpler, lesser code)
With CPython you have to call the `result` method to get the result. (complex, more code)

```python
import time

try:
    import uasyncio as asyncio
except ImportError:
    import asyncio

def add_done_callback(task, callback):
    """
    Function to add_done_callback to task.
    """

    async def watcher(task_instance):
        print("Started observer")
        while True:
            await asyncio.sleep(0.1)
            # await gives control to running loop
            # mandatory to allow asyncio loop
            # running other tasks
            if task_instance.done():
                callback(task_instance)
                return

    asyncio.create_task(watcher(task))

class Task:
    def __init__(self):
        self.running = False
        self.task = None

    def _add_method(self, callback):
        if hasattr(self.task, "add_done_callback"):
            # for CPython
            self.task.add_done_callback(callback)
        else:
            # for Micropython
            add_done_callback(self.task, callback)

    def finished(self, task):
        """
        This method should be called after the
        Task is done. Printing the result.

        CPython's Task instance has a result method
        uPython's Task instance has a data attribute
        """
        try:
            print(f"Task finished with {task.result()}")
        except:
            print(f"Task finished with {task.data}")
        self.running = False

    async def worker(self):
        """
        A worker which should run concurrently by task
        """
        print("Long running Task")
        await asyncio.sleep(5)
        return 42  # <- we want this result

    async def main(self):
        """
        Mainloop where other things could be done
        """
        self.running = True
        self.task = asyncio.create_task(self.worker())
        self._add_method(self.finished)
        while self.running:
            await asyncio.sleep(1)

if __name__ == "__main__":
    asyncio.run(Task().main())
else:
    print("Run following:")
    print(f"{__name__}.asyncio.run({__name__}.Task().main())")

```
CPython:
```
Long running Task
Task finished with 42
```

Micropython:
```>>> import t
Run following:
t.asyncio.run(t.Task().main())
>>> t.asyncio.run(t.Task().main())
Long running Task
Started observer
Task finished with 42
>>> ```

peterhinch, 2022-07-30:
I still think there is an easy workround. This produces the expected `42, True`:
```python
import uasyncio as asyncio

async def done_task(task):
    await asyncio.sleep(0)  # Scheduler hands control to foo which terminates
    print(task.data, task.done())

async def foo():
    try:
        await asyncio.sleep(1)
        return 42
    finally:
        asyncio.create_task(done_task(asyncio.current_task()))

async def main():
    asyncio.create_task(foo())
    await asyncio.sleep(2)

asyncio.run(main())
```