micropython/micropython #8986: Multiple crashes when using UART (ESP32-S on ESP32-CAM)

AwesomeCronk, 2022-07-30:
I'm using MicroPython 1.19.1 on the ESP32-CAM (ESP32_SPIRAM). I used the precompiled binary `esp32spiram-20220618-v1.19.1.bin` on micropython.org.

I am seeing multiple issues when playing around with UART. A lot of the errors are Guru meditation errors (LoadProhibited), with both low addresses and really scrambled addresses in the register dump. Just now I got a different error: `MemoryError: NLR jump failed, val=0x3f817780`. What triggers the errors is a uart instance `.init(...)` after de-initializing or `.read()`.

Here's the latest crash, along with all the code running when it happened:
```python
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4540
ho 0 tail 12 room 4
load:0x40078000,len:12788
load:0x40080400,len:4176
entry 0x40080680
MicroPython v1.19.1 on 2022-06-18; ESP32 module (spiram) with ESP32
Type "help()" for more information.
>>> from machine import UART
>>> rx = UART(1)
>>> rx.init(9600, bits=8, parity=0, stop=1, rx=14)
>>> rx.any()
0
>>> rx.any()
15
>>> rx.read()
b'ok now see this'
>>> rx.read()
>>> rx.read()
b'ok now see this'
>>> while True:
...     if rx.any(): print(rx.read())
... 
MemoryError: NLR jump failed, val=0x3f817780
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4540
ho 0 tail 12 room 4
load:0x40078000,len:12788
load:0x40080400,len:4176
entry 0x40080680
MicroPython v1.19.1 on 2022-06-18; ESP32 module (spiram) with ESP32
Type "help()" for more information.
>>> 
```

I'll grab a dump next time the guru stops meditating. :stuck_out_tongue:

AwesomeCronk, 2022-07-30:
Ok I interrupted the guru, this time by typing in the wrong function name twice. Note that the first boot log is a SW_CPU_RESET, that's because I poked at it immediately after the last crash occurred.
```python
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4540
ho 0 tail 12 room 4
load:0x40078000,len:12788
load:0x40080400,len:4176
entry 0x40080680
MicroPython v1.19.1 on 2022-06-18; ESP32 module (spiram) with ESP32
Type "help()" for more information.
>>> from machine import UART
>>> rx = UART(1)
>>> rx.init(9600, bits=8, parity=0, stop=1, rx=14)
>>> rx.deinit()
>>> rx.init(1, bits=8, parity=0, stop=1, rx=14)
E (35580) uart: uart_wait_tx_done(995): uart driver error
>>> rx.init(9600, bits=8, parity=0, stop=1, rx=14)
E (41050) uart: uart_wait_tx_done(995): uart driver error
>>> rx.init(9600, bits=8, parity=0, stop=1, rx=14)
E (44710) uart: uart_wait_tx_done(995): uart driver error
>>> rx.deint()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'UART' object has no attribute 'deint'
>>> rx.deinnt()
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400e111a  PS      : 0x00060130  A0      : 0x800e137f  A1      : 0x3ffcd120  
A2      : 0x3ffcd140  A3      : 0x3ffba114  A4      : 0x3ffba0c0  A5      : 0x400e145c  
A6      : 0x000000fa  A7      : 0x00000000  A8      : 0xffffffff  A9      : 0x0000002c  
A10     : 0xffffffff  A11     : 0x3ffba114  A12     : 0x3ffc3abc  A13     : 0x000000a5  
A14     : 0x00000014  A15     : 0x0000000c  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000003  LBEG    : 0x40094564  LEND    : 0x40094580  LCOUNT  : 0xffffffff  

Backtrace:0x400e1117:0x3ffcd120 0x400e137c:0x3ffcd140 0x400daeaa:0x3ffcd170 0x400d94f7:0x3ffcd190 0x400d954d:0x3ffcd1b0 0x400ee0c1:0x3ffcd1e0 0x400ee369:0x3ffcd270 0x400d5214:0x3ffcd2b0

ELF file SHA256: 46bca36b7d6020a6

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4540
ho 0 tail 12 room 4
load:0x40078000,len:12788
load:0x40080400,len:4176
entry 0x40080680
MicroPython v1.19.1 on 2022-06-18; ESP32 module (spiram) with ESP32
Type "help()" for more information.
>>> 
```

robert-hh, 2022-07-30:
The default pins for UART 1 are GPIO9 and GPIO10, which  according to the schematics are used for the PSRAM. So you have to use different pins for UART 1.

AwesomeCronk, 2022-07-30:
According to ESP32 Quick Reference:
> Any GPIO can be used for hardware UARTs using the GPIO matrix, so to avoid conflicts simply provide tx and rx pins when constructing.

I was telling UART 1 to use pin 14 as rx, but it seems that even though I was using 14, overriding UART 1 interrupted MicroPython's ability to read/write the SPIRAM.

I erased the flash on the ESP32 and re-imaged MicroPython to it. Using UART 2 and rx=14, everything seems more stable. `.deinit()` still triggers `E (47210) uart: uart_wait_tx_done(995): uart driver error`, but the crashes while reading seem to have vanished. Thanks for catching that @robert-hh!

closed 2022-07-30 by AwesomeCronk