micropython/micropython #8994: ESP32: Wifi peripheral configuration is not reset before/after a soft reset or light/deep sleep.

glenn20, 2022-07-31:
After a soft reset, micropython restarts with the wifi peripheral configuration continuing as set prior to the reset.

This **may** be regarded as a feature by some (eg. continued connection to wifi Access Point) after reset, however, it 

- requires additional unexpected and undocumented protection code in startup scripts
  - eg. `sta.connect(ssid, pwd)` will raise an exception after a soft reset if already connected().
- difficult to diagnose errors can arise in circumstances where the peripheral is left in a bad state or a state unexpected by the users or the micropython software after reset.

I have made the mistake of chasing apparent bugs in my apps (or while developing PR #6515) only for them to resolve after a hard reset. I have also dealt with a number of users reporting issues which have turned out to be due to the wifi peripheral being in a bad state and not being re-initialised after a soft reset. I suspect this may also be contributing to user frustration with apparent lack of reproducibility or reliability.

I propose to prepare a PR which:
- calls esp_wifi_stop() and esp_wifi_deinit() before soft reset
  - also before deepsleep/lightsleep() as required according to [W-fi Bluetoth Sleep Modes](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/sleep_modes.html#wi-fi-bluetooth-and-sleep-modes)

An alternative may be to call `esp_wifi_deinit()` on bootup before `esp_wifi_init()`.

I am aware that some may rely on the current (undocumented) behaviour, though I believe that for predictability and reproducibility it is much more desirable to allways start micropython with the peripheral in a known state.

sosi-deadeye, 2022-07-31:
It's hard to write code for an unknown state.

I've tried it many times with the ESP32 and often, it ends up in a combination of `CTRL+C` + `CTRL+D` and finally pressing the reset button.

In my humble opinion everything, which does have problems with initialization after a `soft-reset`, `deepsleep` or `lightsleep` must be _deactivated_. Even those people who require WLAN could `reactivate` it after a `lightsleep` and they also benefit from it, because of the known state. Same with #8949. Having a defined state, requires lesser logic in application code.

dpgeorge, 2022-08-09:
> This **may** be regarded as a feature by some

For me this is a big, intentional feature!  Eg for automated network testing.

>  eg. `sta.connect(ssid, pwd)` will raise an exception after a soft reset if already connected().

We should fix that so it doesn't raise an exception.

> difficult to diagnose errors can arise in circumstances where the peripheral is left in a bad state or a state unexpected by the users or the micropython software after reset.

But peripherals can get into a bad state during normal running (without a soft reset) and so code needs to handle that situation anyway.

-----------------

The idea with soft reset is that it resets Python state but not necessarily hardware state.  Eg soft reset won't put GPIO back to a default state, or disable USB.  Any hardware that's left untouched after a soft reset should still be in a valid state, and if code expects to run after a soft reset then it must be aware that hardware peripherals can be in a variety of states when MicroPython code starts running.

If you need a blank slate then hard reset should be used instead of soft reset.

karfas, 2022-08-19:
> For me this is a big, intentional feature!  Eg for automated network testing.
> 
> >  eg. `sta.connect(ssid, pwd)` will raise an exception after a soft reset if already connected().
> 
> We should fix that so it doesn't raise an exception.
> 
I don't think so. When it is a feature, the user program has complete responsibility for checking startup conditions and/or peripheral status.

A connect() call for an already connected interface is most likely a program error and therefore deserves an exception.