micropython/micropython #8998: Support for QUIC and HTTP/3

veni-vidi-code, 2022-07-31:
The HTTP/3 standard has been recently finalized and will go towards using QUIC instead of TCP. Adding support for QUIC would be a great addition to those Micropython Ports with network abilities in my opinion.

The relevant RFCs are:
- [9114 (HTTP/3)](https://www.rfc-editor.org/info/rfc9114)
- RFCs 8999 - 9002 for QUIC
  - [8999](https://datatracker.ietf.org/doc/html/rfc8999)
  - [9000](https://datatracker.ietf.org/doc/html/rfc9000)
  - [9001](https://datatracker.ietf.org/doc/html/rfc9001)
  - [9002](https://datatracker.ietf.org/doc/html/rfc9002)
  
There also is alread a finished libary for C Python: https://github.com/aiortc/aioquic

Adding this to Micropython could be done with an additional package aswell, but seeing as QUIC already has a share of the total internet traffic of ~ 30% as of writing this (https://radar.cloudflare.com/) it is relevant enough to get implemented properly in the main lib in my eyes.

AmirHmZz, 2022-08-05:
@veni-vidi-code Can you give some examples of use-cases for this feature?

veni-vidi-code, 2022-08-05:
@AmirHmZz 
HTTP/3 and QUIC are in itself just protocols that replace HTTP/1 HTTP/2 and TCP. Basically it would mean a different way to transfer data via the internet. 

Some of the benefits are that they connect faster, allow multiplex, way less performance lose for a packet loss, encrypt all the data (other than e.g. TCP) and can reconnect even when the client side changes its IP (which results in less problems with mobile devices switching between connections).

So the use cases would be e.g. website loading (would be faster and more secure), for moblie devices that use an internet connection (switching between different wifis or to a mobile data connection would mean no problem anymore), for web servers on picos with e.g. live updating dashboard (that often use TCP) it would mean more performance, the possibility to choose which parts of the stream are more important in case of a packet loss etc.

I think i should also name the one big problem:
QUIC basically uses UDP (thats why it can be really easily be added in software to every hardware). Some Firewally do not like this due to it beeing impossible for them to see whats in these encrypted packages so they block them. This will however most likely change soon anyways due to widespread implementation and usage of HTTP/3.

AmirHmZz, 2022-08-05:
@veni-vidi-code Thanks for your answer but I don't get the point. I know that QUIC is going to be a better choice for backend servers or even CDNs, but how a Microcontroller can benefit these advantages?

veni-vidi-code, 2022-08-05:
@AmirHmZz QUIC does not only need a server side implementation, it also needs a client side implementation. 

Lets say you have a project that is portable and needs to connect to a server (there are enough out there). It would switch between Networks quite often so it would benefit from beeing able to speak QUIC with the server instead of TCP. Also the traffic would be encrypted without needing to do it on a programm to programm basis. For such a mobile project QUIC would most likely be better. But thinking about it w bit more, it might actually not be a good idea to include it in micropython itself due to its size limitations and this not beeing code that might be used by everyone, hence i will close this

closed 2022-08-05 by veni-vidi-code