micropython/micropython #9003: Pico W ENOMEM error when (re)using PIO (rp2_pio_init, CYW43)

scruss, 2022-08-01:
First reported here: [Pico W ENOMEM error when using PIO - Raspberry Pi Forums](https://forums.raspberrypi.com/viewtopic.php?t=338158).

A program that uses PIO (such as the example code: [micropython/pio_1hz.py at master · micropython/micropython](https://github.com/micropython/micropython/blob/master/examples/rp2/pio_1hz.py)) can't be reliably stopped and restarted on a Pico W, while it can on a Pico W.

Expected result: being able to stop and start PIO example at will

Actual result: after a small number of attempts `OSError: [Errno 12] ENOMEM`

The cause appears to be identified as [PIO.remove_program()](https://docs.micropython.org/en/latest/library/rp2.PIO.html?highlight=remove_program#rp2.PIO.remove_program) not being usable when the CYW43 driver is loaded, as the CYW43 itself is running a PIO program.

Indeed, comments inside [rp2_pio_init()](https://github.com/micropython/micropython/blob/0c45a28d24e1a7f9ea5912c928d7b5a860a6514a/ports/rp2/rp2_pio.c#L104) suggest this may be an issue:

```
void rp2_pio_init(void) {
    // Reset all PIO instruction memory.
    #if MICROPY_PY_NETWORK_CYW43
    // TODO: cannot reset PIO memory when CYW43 driver is enabled and active
    // because it uses a PIO for the bus interface.
```
Version:
```
MicroPython v1.19.1 on 2022-07-29; Raspberry Pi Pico W with RP2040
rp2-pico-w-20220729-unstable-v1.19.1-223-g963e599ec.uf2
```

sbrooke, 2022-09-26:
Trying to pull internal temp on a Pico W and then making a requests (Curl) call to a server before going to sleep. Fails after the 5th run with this error. I'm running v1.19.1-443.

scruss, 2022-09-27:
> making a requests (Curl) call to a server before going to sleep. Fails after the 5th run with this error. I'm running v1.19.1-443.

urequests objects can be very large, and can quickly cause ENOMEM if they're not cleaned up by the gc. Your many not be the same issue, especially if you're not explicitly using PIO.

davidoccam, 2022-10-17:
trying 
github blink example on
version = v1.19.1 on 2022-09-23 (GNU 12.1.0 MinSizeRel)
machine = Raspberry Pi Pico W with RP2040
machine.freq = 125000000

gc.collect() #precaution

sm = rp2.StateMachine(0, blink, freq=23413, set_base=Pin(7))

sometimes works  but
intermittently gives  OSError: [Errno 12] ENOMEM on the above line

sbrooke, 2022-11-26:
My problem was because of this:

`r = urequests.post(url, data=data)
    r.close()`

Where I wasn't calling the r.close() so the urequests was staying open.

pdw-mb, 2023-01-12:
I think the replies so far on this topic are unrelated to the original issue.  I can see a couple of possible fixes:

* The CYW43 driver gets installed in `PIO1` by default, and only uses `PIO0` if there's no space.  Given that the driver is initialised as part of micropython startup, I can't see why this would happen, so it should be safe to call `pio_clear_instruction_memory(pio0)`.  This would probably address the issue for the majority of user code.
* Figure out where the CYW43 driver is loaded, and avoid clearing that.  I think this could be done by scanning the whole PIO address space with `can_add_program_at_offset` with a 1 byte program immediately after loading the CYW43 driver.

In any event, I think it would be worth documenting this as it presents a surprising gotcha for anyone migrating working code from a Pico to a Pico W.  e.g. https://stackoverflow.com/questions/74883568/oserror-errno-12-enomem-with-pi-pico-w

dpgeorge, 2023-02-03:
The original problem described here is indeed a problem (as per that comment in `rp2_pio_init()`).

See #10652 for a comprehensive fix, which tracks both state machine and PIO memory resource use, and frees them on soft reset.

dpgeorge, 2023-02-07:
Fixed by 0afe60b876e20684cc0313922a34ee65e7db898a

MarusGradinaru, 2025-04-15:
Cand someone tell me is this bug was fixed in the v1.24.1 of MicroPython for Raspberry Pi Pico W ?
I have a PIO LED blinker and I didn't receive this error so far...