micropython/micropython #9006: Lightsleep with sleep period set to a value never waking

guillochon, 2022-08-01:
I'm currently running a program on a Pico W where I am trying to `lightsleep(5000)`. Unfortunately, the program seems to never wake up again after this call. Using a `time.sleep` instead and the Pico does wake up as expected, but of course with much more power draw.

I am running this nightly: https://micropython.org/resources/firmware/rp2-pico-w-20220729-unstable-v1.19.1-223-g963e599ec.uf2

robert-hh, 2022-08-01:
Confirmed, and it is related to the WiFi being active or not. With WiFi enabled, only lingtsleep times of about 200ms are possible. Without WiFi, there seems to be not limit besides the limitation of the alarm counter, which is 2**32 µs.

guillochon, 2022-08-01:
As a workaround, should I just wrap `lightsleep(100)` in a for loop? Or is there no benefit at the point since the chip is waking up so frequently?

Also I should note that I disconnect the WLAN with a `.disconnect()` call before I run the `lightsleep`. Do I need to do more to deactivate the wifi so that the `lightsleep` will succeed?

robert-hh, 2022-08-01:
Neither method works  here. Lightsleep in a loop fails as well when WiFi is enabled, and sometimes also without WiFi. And disabling WiFi with .active(False) does also not help.

mikerr, 2022-08-03:
It doesn't work on my non wifi PIco here either,

with nightly https://micropython.org/resources/firmware/rp2-pico-20220802-unstable-v1.19.1-229-g579f33050.uf2

On either Pico or Pico W lightsleep hangs on any value higher than 1 'lightsleep(1)'

'lightsleep()' does of course go full dormant as it should

guillochon, 2022-08-14:
Giving this a bump as this is a big issue for a project of mine...battery life is on the order of days without `lightsleep` whereas it would be several weeks I believe.

robert-hh, 2022-08-14:
It seems for any sleep time longer than 1 ms the USB interface gets into a weird state. In REPL, push enter on the terminal revives it. But any print from the device side is lost or behaves strange.

guillochon, 2022-08-24:
@robert-hh where are you removing that `clock_stop` call? I'm using USB to charge the device but I'm not communicating over it so I think it'd be OK (for me at least) if USB takes some time to recover.

robert-hh, 2022-08-24:
File modmachine.c, line 144.

guillochon, 2022-08-27:
@robert-hh I'm willing to give that a try...but how do I make a `.uf2` file for my Pico W?

robert-hh, 2022-08-28:
You have to set up a build environment, preferably with a Linux machine - maybe virtual one or Windows WSL. Instructions are here: https://github.com/micropython/micropython/blob/master/README.md and more specific for the Pico here: https://github.com/micropython/micropython/blob/master/ports/rp2/README.md. If you have all set-up, just run:

make BOARD=PICO_W

There may be docker images to make the set-up easier, but a shot search did only reveal docker containers for ESP8266 and ESP32.

guillochon, 2022-08-28:
Alright, made the code change to `modmachine.c`, compiled the uf2 successfully in Ubuntu in WSL, installed the firmware to my Pico W. Same problem, never waking up from `lightsleep`. :(

For reference here's the code I am trying to run, not sure if there's anything obvious here as to why it might not be working but maybe someone will spot something:

```
import time
from collections import deque

import ds18x20
import network
import onewire
import sys
from machine import Pin
from umqtt.robust2 import MQTTClient

ssid = "abcd"
password = "abcd"
device_name = b"Recirculation Probe"

sleep_time = 5000  # ms
inactive_sleep_time = 15000  # ms
awake_dt = 3  # deg C / s
history_len = 12
max_inactive_time = 15 * 60  # s
min_active_time = 15  # s
min_plausible_temp = 10  # C
heartbeat_time = max_inactive_time  # s
on_usb = False
check_probe = True

temperature = None
active = False
time_since_active = 0  # s
active_start_time = time.time()
inactive_start_time = sys.maxsize

if on_usb:
    lightsleep = time.sleep_ms
else:
    lightsleep = machine.lightsleep

wlan = network.WLAN(network.STA_IF)

def connect_wifi():
    wlan.active(True)
    # wlan.config(dhcp_hostname=device_name)  # Not yet supported on Pico W https://github.com/micropython/micropython/pull/8918
    wlan.connect(ssid, password)

    # Wait for connect or fail
    max_wait = 60
    while max_wait > 0:
        if wlan.status() < 0 or wlan.status() >= 3:
            break
        max_wait -= 1
        print("Waiting for connection...")
        time.sleep(1)

    # Handle connection error
    if wlan.status() != 3:
        raise RuntimeError("Network connection failed.")
    else:
        print("Connected.")
        status = wlan.ifconfig()
        print("ip = " + status[0])

    if not c.connect(clean_session=False):
        print("New session being set up")

    led.on()

def disconnect_wifi():
    print("Disconnecting from WiFi")
    c.disconnect()
    wlan.active(False)
    wlan.disconnect()

    led.off()

led = Pin("LED", Pin.OUT)
led.on()

c = MQTTClient(device_name, "homebridge.local")
c.DEBUG = True
# Information whether we store unsent messages with the flag QoS==0 in the queue.
c.KEEP_QOS0 = False
# Option, limits the possibility of only one unique message being queued.
c.NO_QUEUE_DUPS = True
# Limit the number of unsent messages in the queue.
c.MSG_QUEUE_MAX = 2

connect_wifi()
active = True

ds_pin = Pin(2)
ds_sensor = ds18x20.DS18X20(onewire.OneWire(ds_pin))

if check_probe:
    while 1:
        try:
            rom = ds_sensor.scan()[0]
        except IndexError:
            print("Couldn't find temperature probe, trying again in 5 seconds.")
            time.sleep_ms(5000)
        else:
            break

temperature_history = deque((), history_len)
last_temperature = None
time_of_heartbeat = 0
heartbeat_value = False
temp_plausible = True
while 1:
    if check_probe:
        ds_sensor.convert_temp()

        temperature = ds_sensor.read_temp(rom)

        temp_plausible = temperature > min_plausible_temp

        if temp_plausible:
            temperature_history.append(temperature)

        print(temperature)
    else:
        temperature = -1

    active_time = time.time() - active_start_time
    inactive_time = time.time() - inactive_start_time

    max_temp = -1000
    min_temp = 1000
    for i in range(len(temperature_history)):

        t = temperature_history.popleft()
        if t > max_temp:
            max_temp = t
        if t < min_temp:
            min_temp = t
        temperature_history.append(t)

    if (
        (max_temp - min_temp >= awake_dt)
        or (active and active_time <= min_active_time)
        or (not active and inactive_time > max_inactive_time)
    ):
        if not active:
            connect_wifi()
            active = True
            active_start_time = time.time()
            inactive_start_time = sys.maxsize
    else:
        if active:
            disconnect_wifi()
            active = False
            inactive_start_time = time.time()

    if temp_plausible and active and last_temperature != temperature:
        if check_probe:
            c.publish(device_name + b"/temperature", str(temperature))
        if time.time() - time_of_heartbeat > heartbeat_time:
            heartbeat_value = not heartbeat_value
            c.publish(device_name + b"/heartbeat", str(heartbeat_value).lower())
        led.toggle()
        time.sleep_ms(100)
        led.toggle()
        time.sleep_ms(sleep_time - 100)
    else:
        if active:
            time.sleep_ms(sleep_time)
        else:
            lightsleep(inactive_sleep_time)

    if temp_plausible and check_probe:
        last_temperature = temperature

    # At this point in the code you must consider how to handle
    # connection errors.  And how often to resume the connection.
    if active:
        if c.is_conn_issue():
            while c.is_conn_issue():
                # If the connection is successful, the is_conn_issue
                # method will not return a connection error.
                c.reconnect()
            else:
                c.resubscribe()

        c.check_msg()  # needed when publish(qos=1), ping(), subscribe()
        c.send_queue()  # needed when using the caching capabilities for unsent messages

c.disconnect()
```

robert-hh, 2022-08-28:
That's not so good. But it was anyhow not the real fix. Until that is fixed, you could try to use sleep_ms with a clock freq of 24Mhz for the time of sleeping, reducing the current consumption of a Pico or a Pico_W without WiFi to about 6 mA (from ~18mA).

guillochon, 2022-08-28:
How do I set the clock frequency? It appears `sleep_ms` only takes one argument...is it another function?

guillochon, 2022-08-28:
OK I see, I can change it with `machine.freq`. It looks like although I can legally set it to 24 MHz, the Pico stops functioning at that low a frequency. In some trial-and-error testing I was able to drop it to 55 MHz and still have it be responsive, which is slightly less than half the native frequency. So that helps (should roughly double the battery life), but uncertain why I wasn't able to take it all the way to 24...

Any other ways I can save power here? I read I can maybe disable parts of the chip that I'm not using?

robert-hh, 2022-08-29:
I tested with a Pico and a Pico_W, going down to 24 MHz. I was surprised that it could not be less. The data sheet show figures down to ~10 MHz. But I only checked that REPL is responsive, that time.sleep_ms() works as it should, and that a small performance test code ran. The performance drop was equivalent to the frequency ratio.
Disabling parts of the chip or board that are not used is what machine.lightsleep() tries at the moment, by switching of some of the clocks. For CMOS circuits that is sufficient to stop the power consumption. You should also look at your board's configuration for GPIO pins driving loads. These should whenever possible be set to a non-driving state.

guillochon, 2022-08-29:
Yeah the REPL stopped working (Thonny said the USB connection had died) when I set it to 50 MHz. All I can think is that the temperature probe (the ds18x20, which uses onewire) might be interfering with things? I am not entirely sure where it's locking up given I lose REPL when it happens.

robert-hh, 2022-08-29:
I have the suspicion that something is generally wrong with USB and Pico being not properly clocked. Sometimes I see strange effects even in default operation modes, like requiring several resets before the USB interface comes up. In bootloader mode, the USB interface always works. But that  is a different firmware. Something to dive into ....

guillochon, 2022-08-31:
Well, sadly longer term testing here has not been fruitful...the Pico runs stably for a few hours, but seems to eventually seize up when running with these lower frequencies. So I've reverted back to the full clock speed for now. Hopefully a `lightsleep` fix comes eventually, until then I'll be patient 😄

felixdoerre, 2022-09-26:
@guillochon As an alternative workaround, you could try `wlan.deinit()`, that would reset the WLAN chip and should allow you to go into lightsleep mode. A minimal example worked fine for me for some hours:

```
import time, machine, base, socket

while True:
    led = machine.Pin("LED", machine.Pin.OUT)
    led.on()
    wlan = base.init_wlan();
    s = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
    s.sendto("value", ("255.255.255.255", 8000))
    s.close()
    wlan.disconnect()
    wlan.deinit()
    led.off()

    machine.lightsleep(60000)
```

I've got the debug pins of my pico hooked up so I can pull stack traces when the pico hangs in lightsleep mode, and from what I observed the pico hangs when the wlan chip is not deinitialized and he recieves packets while in lightsleep:
```
#0  cyw43_post_poll_hook () at ../mpnetworkport.c:70
#1  0x1004ba6c in cyw43_poll_func () at .../micropython/lib/cyw43-driver/src/cyw43_ctrl.c:259
#2  0x10037336 in PendSV_Handler () at ../pendsv.c:80
#3  <signal handler called>
#4  machine_lightsleep (n_args=1, args=<optimized out>) at ../modmachine.c:184
#5  0x10013908 in fun_builtin_var_call (self_in=0x1007e1ac <machine_lightsleep_obj>, n_args=1, n_kw=0, args=0x20041ef8) at .../micropython/py/objfun.c:131
#6  0x1001d064 in mp_call_function_n_kw (fun_in=0x1007e1ac <machine_lightsleep_obj>, n_args=1, n_kw=n_kw@entry=0, args=args@entry=0x20041ef8) at .../micropython/py/runtime.c:693
#7  0x1001d186 in mp_call_method_n_kw (n_args=<optimized out>, n_kw=0, args=args@entry=0x20041ef0) at .../micropython/py/runtime.c:709
#8  0x20003806 in mp_execute_bytecode (code_state=code_state@entry=0x20041ed8, inject_exc=inject_exc@entry=0x0) at .../micropython/py/vm.c:1042
#9  0x10013a2e in fun_bc_call (self_in=0x2000b070 <gc_heap+13140>, n_args=0, n_kw=1, args=0x1001d065 <mp_call_function_n_kw+32>) at .../micropython/py/objfun.c:288
#10 0x1001d064 in mp_call_function_n_kw (fun_in=fun_in@entry=0x2000b070 <gc_heap+13140>, n_args=n_args@entry=0, n_kw=n_kw@entry=0, args=args@entry=0x0) at .../micropython/py/runtime.c:693
#11 0x1001d08c in mp_call_function_0 (fun=fun@entry=0x2000b070 <gc_heap+13140>) at .../micropython/py/runtime.c:667
#12 0x100329ae in parse_compile_execute (source=source@entry=0x20041fb0, input_kind=<optimized out>, exec_flags=exec_flags@entry=22) at .../micropython/shared/runtime/pyexec.c:120
#13 0x10032c14 in pyexec_friendly_repl () at .../micropython/shared/runtime/pyexec.c:672
#14 0x100364e6 in main (argc=<optimized out>, argv=<optimized out>) at ../main.c:198
```
stepping around a bit shows, that the interrupt handler tries to get data from the wifi chip, but fails:
```
Thread 1 hit Breakpoint 2, cyw43_ll_sdpcm_poll_device (self=self@entry=0x2003da30 <cyw43_state>, len=len@entry=0x20041da8, buf=buf@entry=0x20041dac) at .../micropython/lib/cyw43-driver/src/cyw43_ll.c:1025
1025   .../micropython/lib/cyw43-driver/src/cyw43_ll.c: No such file or directory.
(gdb) p/x spi_int
$1 = 0x4140
```
ultimately failing to get the wifi chip to clear the interrupt line and therefore hanging in `PendSV`+interrupt handler never reaching the main code to continue. If I disable the interrupt (i.e. by skipping `cyw43_post_poll_hook` execution resumes, but ofc the wifi driver is not usable anymore without deinit/init). So I guess we would need someone with more detailed knowledge of the wifi chip to understand how to put it in sleep mode and more importantly how to recover from it afterwards.

Plaque-fcc, 2022-11-02:
> It doesn't work on my non wifi PIco here either,
>
> with nightly https://micropython.org/resources/firmware/rp2-pico-20220802-unstable-v1.19.1-229-g579f33050.uf2
>
> On either Pico or Pico W lightsleep [...]

Can confirm: with values bigger than 1, my non-WiFi RP2040 board hangs up as well, until reset.

felixdoerre, 2022-11-03:
Interesting. I have no non-wifi board to test with and guessing what goes wrong is a bit hard without a stack trace. Have you tried the firmware with the fix I proposed in #9438? Although I triggered the hand on a pico-W board deterministically with wifi packages, on a non-wifi board there might be interrupts as well.

Plaque-fcc, 2022-11-04:
> Interesting. I have no non-wifi board to test with and guessing what goes wrong is a bit hard without a stack trace. Have you tried the firmware with the fix I proposed in #9438? Although I triggered the hang on a pico-W board deterministically with wifi packages, on a non-wifi board there might be interrupts as well.

I think I should try it on my non-WiFi board.

Plaque-fcc, 2022-11-04:
Actually, on the fix proposal by @felixdoerre: your fix actually works.
But only for the first time. For instance, after hard-resetting the board, I try `machine.lightsleep(3)`, it works fine. If I repeat it, the µC gets stuck until reset.

Plaque-fcc, 2022-11-06:
Actually, on the fix proposal by @felixdoerre: your fix #9438 actually works for a non-WiFi board.
But only for the first time. For instance, after hard-resetting the board, I try `machine.lightsleep(3)`, it works fine. If I repeat it, the µC gets stuck until reset.

blongden, 2022-12-03:
Experiencing the same issue - unable to wake from lightsleep (wifi is connected, I have not tried it without wifi yet to see if that makes a difference). Keeping an eye on this thread!

Plaque-fcc, 2022-12-04:
> Experiencing the same issue - unable to wake from lightsleep (wifi is connected, I have not tried it without wifi yet to see if that makes a difference). Keeping an eye on this thread!

Can you test the fix #9438 on your Pico W? I only have a regular Pico, and this fix didn't help me, but I can't test the WiFi-enabled build myself.

jmodrako, 2023-02-14:
Does to #9438 is included in the latest Micropython firmware for Pico W? As I can't get working `lightsleep`...

jmodrako, 2023-02-19:
Does anyone have working sleeping in the latest firmware?

felixdoerre, 2023-02-19:
If by "latest" you mean `1.19.1` as downloadable from the website, then no, the PR is not included in there. If you mean, built from current master, then I guess it should work.

Apart from that, the code that you posted in https://github.com/micropython/micropython/pull/9438#issuecomment-1429827995 should work even on older firmware. I've never observed lightsleep not working as expected when the WiFi chip was inactive. However in the code I see no way how you would be able to tell without debugger if lightsleep is working or not, apart from maybe pinging and checking how long the pico is unavailable between reconnects.

With the PR, lightsleep should wait until the next WiFi package arrives or the specified timeout, whichever is earlier, so in a busy network it might look as if lightsleep is waiting not long enough.

jmodrako, 2023-02-20:
> as downloadable from the website

Did you mean this site: https://micropython.org/download/rp2-pico-w/

felixdoerre, 2023-02-27:
@jmodrako not sure if you noticed, but I was able to reproduce the problem and submitted a PR: https://github.com/micropython/micropython/pull/10803

jmodrako, 2023-02-27:
WOW! Oneliner ❤️ I will test it tomorrow and let you know how it works ;)

aleppax, 2023-03-10:
I've been lucky enough not to encounter this problem until now. I was using nightly build 1.1.19-8xx being xx two numbers I don't remember, maybe 53 (it was flashed on February), but my colleague was having this very issue with a older build (same version): same code, different behaviour on two Pico. Therefore we both updated three days ago to the latest build and this led to the problem occurring also to my Pico. Recently  wifi drivers have been replaced with open source ones, could it be related? I'll try to build a source before those changes.
I've tested the PR by @felixdoerre and not at the first occurrence of lightsleep but after an hour or so, it still hangs not exiting from sleep.
Also I can't manage to achieve current absorption less than 2mA without tampering with pin 23.

optimal-system, 2024-02-26:
Hi everybody
Any news on the lightsleep bug ? It's still not working with latest firmware on both Pico H and W.
It only work with value 1, but that's useless.
```
from machine    import lightsleep
for i in range (1000) :
    print(i)
    lightsleep(2)
```
I really need to save power for my project. Any suggestions ?

felixdoerre, 2024-02-26:
What behavior are you experiencing? Is the lightsleep hanging? Note, that when you lightsleep, the usb connection might break down, as the device will also not respond to usb commands. For me, on the webrepl, this snippet works fine. Have you tried blinking the LED instead if printing something?

optimal-system, 2024-02-26:
Thanks for your lightspeed reply,

I tried without success :     Pin("LED", Pin.OUT).toggle()
and also write to a log file

optimal-system, 2024-02-26:
Hi again,
Can someone explain me what is needed (or on the contrary what not to do) to get lightsleep() running correctly.
felidoerre pointed at the USB cnx and the print(). I don't have that in my project but I have 
- several GPIO pins activated (out and in / pull up)  
- write() statements for log file in csv and other config file in json 
- neopixels
- wifi connexions (not active when I call lightsleep)
It's not easy for me to disable all that stuff and still have my program doing something...
Thanks in advance if you give me a insight.

optimal-system, 2024-02-27:
I might have found the problem : The power supply !

Obviously if you want to use lightsleep(), you don't have illimited power so I tried my program with a PowerPack.
When the USB cable is connected to AC adapter there is no problem with lightsleep() but with a connexion to a PowerPack, there is no waking up !

Thus I'm going to try to get power from the VBUS/40 Pin.

Anybody found that problem before ?

robert-hh, 2024-02-27:
I recall people reporting, that power packs switch of the output if the drawn current is too low. That will not change if you feed into the VBus pin instead of the USB port.

optimal-system, 2024-02-27:
Well, what is the use of lightsleep() if you cannot use it with limited power supply (power pack) ?
Maybe it would be fine with a regular battery (with no switch off), but it's cumbersome to recharge and expensive to replace.
My Pico will be running 5s every 20mn and I would enjoy to save maximum power to not have to use AC adapter.

robert-hh, 2024-02-27:
There are RP2040 boards with support for Lithium batteries including a charging circuit, like the Adafruit Feather RP2040 or the Waveshare RP2040-Plus. Probably there are more. You can connect a battery matching your needs.

dhalbert, 2024-02-27:
It is a peculiarity of most power packs. Some power packs do support low-current draw:
https://voltaicsystems.com/always-on-batteries/
https://support.anker.com/s/article/What-is-Trickle-Charging-Mode

optimal-system, 2024-02-27:
Thanks a lot for your advices. 
I checked Adafruit Feather RP2040 and the Waveshare RP2040-Plus. That seemed the best way to go but unfortunately they don't have wifi or BLE connexion. I could not find any micro controller with Built-in lipo charger AND wifi/ble.  
Then I checked the second solution with the Voltaic "Always On" battery. That should work but it's a bit expensive for my project.
Maybe the simplest hack is to activate lightsleep() for x seconds, then ask for power by turning on the buildin led for 1ms, and restart sleeping... Just need to find the good timing and measure if that really save power in comparison with time.sleep()

optimal-system, 2024-03-04:
This post is probably a bit out of topic but I wanted to close my intervention.
After quite a few tests I arrived to these results :
- PowerPack in general are not at alla suitable for Pico.  You must use a PP with low-current (like Voltaic).  
- Direct connection to VSys is better than a PP. 9V and 18650 batteries were tested. You can also use a 12V car battery if space is not an issue.  A basic converter with a transistor and 2 capacitors is sufficient to get clean 5V.  There are also several solutions available to monitor battery charge and discharge.
**- Using machine.lightsleep() is a bit better than time.sleep() but you still run only for a few hours, not days.**  
As far as I can tell and unfortunately, there is no way to run a Raspberry (Pico or Pi) for a long period on batteries. Thus you must be connected to power or use a solar panel.

RYoder97, 2024-11-07:
I am having the exact opposite problem (sort of).

I'm using a Pi Pico WH on a breadboard. Board is connected to a USB power tester to measure draw and voltage.

```lightsleep()``` and ```deepsleep()``` work, but I cannot seem to wake the device using GPIO afterwards.

Setting a value for lightsleep (ex. ```lightsleep(10000)```) does nothing in terms of power consumption, and my interrupt code works. But since there's no dip in power, I assume it's not even sleeping at all with this.

Setting a value for deepsleep (ex. ```deepsleep(10000)```) seems to disconnect the USB connection between the Pi and my laptop, but still no reduction in power.

Leaving ```lightsleep()``` and ```deepsleep()``` with NO time parameter always results in a drop in power draw, but I can't do anything with the board.

What is going on? Is this Pi bad? I've tried re-flashing a couple of times.

```python

import machine
from machine import Pin, deepsleep, lightsleep

def wake_up(pin):
    # Handle wake-up logic here
    print("Button pressed, waking up!")

# Configure buttons
button = Pin(10, Pin.IN, Pin.PULL_UP) # Example pin

# Set up interrupt
button.irq(trigger=Pin.IRQ_FALLING, handler=wake_up)

# Go to sleep until a button is pressed
deepsleep(10000)

```

Plaque-fcc, 2024-11-08:
@RYoder97 no, you seem to have the very same problem.