micropython/micropython #9007: Pico PWM bug: PWM.deinit() does not reset pin FUNCSEL

rkompass, 2022-08-02:
The title tells the problem and the following code illustrates it.
```
from machine import Pin, PWM, mem32
from array import array
from time import sleep

def pin_func(pin):    # report OEOVER, OUTOVER, FUNCSEL  for the pin on RPi 2040
    func = {1: 'SPI', 2: 'UART', 3: 'I2C', 4:'PWM', 5:'SIO', 6:'PIO0', 7:'PIO1', 9:'USB'}
    oeover = {0:'OE<-func', 1:'OE<-~func', 2:'OUT disabl', 3:'OUT enable'}           # descriptions
    outover = {0:'OUT<-func', 1:'OUT<-~func', 2:'OUT low', 3:'OUT high'}
    baddr=bytes(array('O', [pin]))
    pin = mem32[int.from_bytes(baddr, 'little')+4]                                   # pin number
    ctrl = mem32[0x40014004 + 0x08*pin]                                              # GPIOpin_CTRL
    return oeover[ctrl>>12 & 0b11], outover[ctrl>>8 & 0b11], func[ctrl & 0b11111]    # extract relevant bits from CTRL

pin9 = Pin(9, Pin.IN)
print(pin_func(pin9))    #  ('OE<-func', 'OUT<-func', 'SIO')

pwm0 = PWM(pin9)                                   # create PWM object from a pin
pwm0.freq(1000)                                    # set frequency
pwm0.duty_u16(30_000)                              # set duty cycle, range 0-65535
sleep(1)
pwm0.deinit()                                      # turn off PWM on the pin
print(pin_func(pin9))    #  ('OE<-func', 'OUT<-func', 'PWM')  # !!!! pwm isn't turned off  !!!!

sleep(1)
Pin(9, Pin.IN)                                     # <-  this turns PWM off
print(pin_func(pin9))    #  ('OE<-func', 'OUT<-func', 'SIO')  
```

rkompass, 2022-08-03:
Perhaps I should add that a LED at pin 9 started to shine with half intensity, and it continued to be on even after pwm0.deinit.
That's how I noted the problem.

JoelGodin, 2022-08-18:
Same issue here. I can [pin].deinit() but still control the duty cycle of that pin.  PWM is still initialized.

gitsmol, 2023-07-19:
Can confirm this issue is present on ESP32. I'm connecting a number of buzzers to the board and notice that sounding one of them sounds all of them after they have been initialized. Deinitializing does not prevent this. Pin(X, Pin.IN) does.

daid, 2026-02-28:
> Can confirm this issue is present on ESP32. I'm connecting a number of buzzers to the board and notice that sounding one of them sounds all of them after they have been initialized. Deinitializing does not prevent this. Pin(X, Pin.IN) does.

Thanks for this workaround.

For an ESP32 fix, I think the `mp_machine_pwm_deinit` function just needs a call to `esp_rom_gpio_pad_select_gpio` (and maybe `gpio_set_direction`?) I didn't try that yet, as I don't have a micropython build environment setup for ESP32

daid, 2026-02-28:
> Just re-checked the issue with a recent build, and the problem seems to be gone. PWM had a major reshape since the issue was raised. Closing it now. [@rkompass](https://github.com/rkompass) If you think the issue is still valid, please re-open it.

I don't know if it's still the case for the RP2, but the ESP32 definitely still has this issue in the latest build.

daid, 2026-03-01:
> `pwm_pin.deinit()` turns off the PWM signal at pwm_pin as documented, but keeps the assignment as PWM pin. It does not re-configure the pin to any other mode. What mode should that be? GPIO, UART, I2S, I2C .... ? If the Pin is to be used in a different mode, it must be done deliberately by the Python code.

I respectfully disagree with your reasoning. I expect a deinit to do the inverse on an init. Not keep some random state that causes extremely unpredictable behavior.
As for what to revert to, gpio input. Just like pins are per default after reset.