micropython/micropython #9009: "default" rules from py/mkrules.mk place build artefacts outside build directory

musteresel, 2022-08-02:
For the "default" build rules from `py/mkrules.mk` for both a binary and a static library we do not take into account the build directory, thus placing build artefacts (the binary and/or libmicropython.a) potentially outside of the build directory. I stumbled across this because my source directory (where the repository is checked out, which happens to be the current working directory of the build process) is read-only (at least for the user running the build); therefore I see permission denied errors.

That's not really a critical issue I guess ...  but IMO it'd be cleaner to force the build output to be inside the build directory, e.g. by prefixing `$(BUILD)` (if set) at those places:

https://github.com/micropython/micropython/blob/579f330508e4ba46f4a71193582c36bdf7aa56bd/py/mkrules.mk#L201
https://github.com/micropython/micropython/blob/579f330508e4ba46f4a71193582c36bdf7aa56bd/py/mkrules.mk#L234

AFAIK at least the unix port uses the `PROG` make variable and the associated rule; so changing this might break stuff (which assumes the final binary to show up in the current working directory rather than the build directory). Let me know if you want me to change it.

dlech, 2022-08-02:
This has been a constant papercut bug for me - not bad enough to fix it but always an annoyance. So I would be very happy to see this change made.

musteresel, 2022-08-02:
Working on it...

dpgeorge, 2022-08-06:
> This has been a constant papercut bug for me

@dlech can you explain what the issue is for you?

dlech, 2022-08-06:
The problem happens when you need to build the same unix variant for different targets. For example, we build the unix port for the host arch and also cross-compile for embedded arches. So when you build for embedded, it writes over the executable for the host arch even though they have different build directories. The same thing happens if you want to have a debug build and release build for the host arch at the same time. I realize that you can work around this by changing the `PROG` variable in addition to the `BUILD` variable on the command line, but that is just one more thing to remember and in the case of the embedded builds, an extra rename to change it back to the default name after copying to the embedded device.

dpgeorge, 2022-08-08:
OK thanks for the explanation.  I agree that is annoying.

dpgeorge, 2022-08-11:
Fixed by b2e82402688b53829f37475583231b067b9faea7

closed 2022-08-11 by dpgeorge

jimmo, 2022-10-06:
@musteresel @dlech -- this took some getting used to (I build different variants of the unix port many times each day) but overall I much prefer the way it works now. Thank you for making this happen!!