micropython/micropython #9010: Pico type only determined by which type of Pico firmware image installed?

Jibun-no-Kage, 2022-08-02:
Micropython only knows model of Pico by which firmware is installed?  I tested this, and it is true, you can load non-wifi image on a Pico W, and a Pico W image to Pico (original).  And the query of model descriptor is wrong, for example...

>>> import sys
>>> sys.implementation._machine
Raspberry Pi Pico W on RPI2040

Even though the actual hardware is NOT Pico W? This can't be the indented result, right?  As noted above the reverse is true as well.  The official Pico W documentation notes the actual method to determine the Pico model type, Pico vs. PicoW.  Is this a bug?

Reference [url]https://datasheets.raspberrypi.com/picow/connecting-to-the-internet-with-pico-w.pdf[/url], page , how to determine model type of Pico is in section 2.4.

robert-hh, 2022-08-03:
I do not consider that this as a bug. There is no automatic firmware load process which tells the type of the target board or a unique firmware for all models. The user is responsible for loading the proper firmware, and sometimes people load intentionally the "wrong" firmware to a device, knowing that may behave different. Telling at which hardware the firmware is running might be possible in special cases like Pico W vs Std. Pico, but generalization is hardly possible. Even in that special case, the detection mechanism is weak. There might be a board with the same ADC3 level as the Pico W, but being no Pico W.
So even assuming such a mechanism exists, what should the 'wrong' firmware do: Refuse to work? Print an error message?

Jibun-no-Kage, 2022-08-05:
Not a bug? But a missed opportunity then.  Specific to Pico, yes, an error that the image is not correct would be a very good thing.  The fact that the official documentation notes that Pico vs. Pico W identification is possible strongly suggests that the next model of Pico will have an explicit identification method.  As you may recall the first few Raspberry Pi devices had a similar issue, where you had to query the processor and memory size to figure out what you had under the hood, but the Pi foundation corrected this.  Pico seems to be going through the same process.  Once the RTS OS for Pico is released, I am sure it will be able to identify the actual hardware characteristics in a specific and qualified manner per model of Pico release.

Because MicroPython has a ESP legacy, in which specific hardware characteristic identification is still a bit dodgy, I can understand how generalization is and for ESP maybe never realistically possible.  But that should not mean that Pico is crippled by that same legacy.

Note, I said Pico, not RP2040 based devices in total.  Will all RP2040 based devices have self-identification?  IMHO they all should.  But will they, likely not.  If for no other reason than to avoid the mess with ESP scope.  Countless forum questions, google queries and debate if not discussion on which model some has or does not have what memory it has or does not have, etc., is all wasted time and effort because some did not think it would be a good idea to standardize device identification.

From historical perspective, PC compatibles had the same basic issue in the early years, it was not easy to figure out what a given PC really had under the hood, Microsoft (following Apple mindset for hardware qualification, I know this because I worked at Apple at the time) finally had to force hardware developers to standardize and qualify hardware identification, or the confusion would have continued to this day.

IMHO, a missed opportunity that would have saved time and effort, expense, etc.  Oh well! :)

Jibun-no-Kage, 2022-08-07:
Hey, just for reference, there does seem to be a valid way to determine the Pico device type.  At least across the Pico devices I have.  For example:

`machine.ubinascii.hexlify(unique_id()).decode()`

Always seems to return 'e6605481db' the following for Pico device I have:

`   e6605481db4e2f37'
    e6605481db704c36
    e6605481db2ab634
`

Always seems to return 'e6614c311b the following for Pico W devices I have:

`e6614c311b307d33`

I wish I had one more Pico W to confirm the sequence.  Maybe others could confirm this?

robert-hh, 2022-08-07:
using `import machine;import binascii;binascii.hexlify(machine.unique_id()).decode()`
Pico:
'e66058388340512e'
'e660583883412b2e'
Pico W:
'e6614c311b777932'
Arduino Nano Connect:
'50157249485d691c'
Sparkfun Pro Micro 2040:
'e469b03567342f37'
So maybe it's only the first 4 or 5 digits that tell the hardware type.
Even if there are unique hardware type identifiers, these must be administered and documented. Does that kind of administration exist?

Jibun-no-Kage, 2022-08-08:
True, these may not be truly qualified by type, only but processor id or serial number via 2040.  That is the question that I think should be posed to the Raspberry Pi foundation to address.  Figure the MicroPython development team has more weight that I would, so if it came from them, likely get better consideration.  

For now, I am using the values I have in a lookup enum/dict, as well as query of ADC3 and GPIO25 pulled low voltage reading (per the official Pico C/C++ SDK) to qualify what the MicroPython image 'should' be installed, if the firmware image reports different I throw an exception.  Reasonable approach for now.

dpgeorge, 2022-08-08:
> True, these may not be truly qualified by type, only but processor id or serial number via 2040. That is the question that I think should be posed to the Raspberry Pi foundation to address

The value returned by `machine.unique_id()` is actually the 64-bit unique ID from the external SPI flash.  The RP2040 itself does not have any unique id.  According to the docs:

> all instances of RP2040 silicon are identical and have no persistent state

jimmo, 2022-08-08:
There seems to be two separate points here:

1. Should we allow you to install the "wrong" firmware on a given board.

2. Should there be a way to query what hardware you're on at runtime (distinct to what the firmware says)?

I agree with @robert-hh on (1) especially given that reliable detection isn't possible, and I'm fairly sure (2) is unfortunately out of our control on Pico / Pico W (modulo tricks with ADC3 & GPIO25).

Jibun-no-Kage, 2022-08-08:
Ah, so we are really stuck with no options, until the Pico has a designed method to qualify model/type.  I have no problem for now using the ADC/GPIO trick, to determine what modules I need to load.  Just means I that I will likely have to roll my own firmware image, to conserve resources, but we all end up at the point anyway when or if we need to optimize.

Funny that Pi foundation had the same basic issue with the original Pi.  They had to create qualified method.  Although a true serial number exists from the BROADCOM processor, I am sure for licensing purposes.

I think I what I have done is about as good as can be for now, which I did for ESP32 modules, I maintain a reference list in my master image, and my boot script ensures the MAC is unique per the table, as well as the host name in due course.  But what I have done only works for a given environment, and a given set of devices, and of course a given common firmware image.

I still think a simple warning that the wrong image was loaded is a good idea... once you can with accurately determine the device type.

closed 2022-08-21 by Jibun-no-Kage