micropython/micropython #9013: Duplicate WIZNET5K driver/library files in MicroPython tree

omogenot, 2022-08-03:
I noticed that the wiznet5k files are duplicated in the tree. As far as I can see, the one located in lib/wiznet5k are the latest and "linked" to WIZNET development branch (i.e. [wiznet5k @ 0803fc5]), whereas a similar set of files is located in drivers/wiznet5k and seems very old :
Original source: https://github.com/Wiznet/W5500_EVB/tree/master/ioLibrary Taken on: 30 August 2014
Shall not this older version of files be removed, making sure that some builds which could be dependent on these can use the newer ones instead?

Apparently, the only port relying on this old version is stm32. Can anyone try to use lib/wiznet5k instead of drivers/wiznet5k in ports/stm32/Makefile and ports/stm32/network_wiznet5k.c files? I have no STM32 platform to test.

robert-hh, 2022-08-03:
It does not even compile. Fixing the Makefile was the easy part. But then are some compile errors compiling stm32/modnwwiznet5k.c.

robert-hh, 2022-08-03:
Also extmod/networkl_wiznet5k.c should be use instead stm32/modnwwiznet5k.c. But that is a major change.

omogenot, 2022-08-03:
@robert-hh 
Which board compilation are you working on? The pybv10 (default) compiles OK for me. If I can help making your board version compile at least. I cannot test, I have no stm32 based board.

robert-hh, 2022-08-03:
I have a PYBV11 & can test that.

omogenot, 2022-08-03:
Hum... I did not change anything but the makefile `stm32/Makefile` and `stm32/network_wiznet5k.c` to change the reference from `drivers/wiznet5k` to `lib/wiznet5k`, and rename `drivers/wiznet5k` as `drivers/wiznet5k.old` and `pybv11` compiles just fine :-(
```
LINK build-PYBV11/firmware.elf
   text    data     bss     dec     hex filename
 368152      16   27448  395616   60960 build-PYBV11/firmware.elf
GEN build-PYBV11/firmware0.bin
GEN build-PYBV11/firmware1.bin
GEN build-PYBV11/firmware.dfu
GEN build-PYBV11/firmware.hex
```
Strange...

omogenot, 2022-08-03:
@robert-hh 
I renamed modnwwiznet5k.c to .old and it still compiles as if it is not part of the makefile...
More and more strange...
Do you want the .dfu file on the store link to test?

robert-hh, 2022-08-03:
Are you sure you compiled as
`make MICROPY_PY_NETWORK_WIZNET5K=5500 `
Because it cannot work. The file names in the lib have different case, like Ethernet instead of ethernet, and so on.

robert-hh, 2022-08-03:
Yes, you can send the DFU file.

omogenot, 2022-08-03:
@robert-hh 
Ok I found that make option just at the same time you sent this message. I will go through it and let you know when it's finished... I'll post the dfu with the WIZNET option when it's done.

robert-hh, 2022-08-03:
There is a branch with the changes for files & settings here: https://github.com/robert-hh/micropython/tree/stm32_wizlib
Just 2 compile errorrs left.

omogenot, 2022-08-03:
@robert-hh 
I am at 1 link error left, I'm working on. Most of the modifications consist in upgrading the Makefile (and probably remove unused files).

omogenot, 2022-08-03:
@robert-hh 
My Makefile block related to WIZnet contains:
```
ifneq ($(MICROPY_PY_NETWORK_WIZNET5K),0)
WIZNET5K_DIR=lib/wiznet5k
INC += -I$(TOP)/$(WIZNET5K_DIR) -I$(TOP)/$(WIZNET5K_DIR)/Ethernet
CFLAGS_MOD += -DMICROPY_PY_NETWORK_WIZNET5K=$(MICROPY_PY_NETWORK_WIZNET5K) -D_WIZCHIP_=$(MICROPY_PY_NETWORK_WIZNET5K)
CFLAGS_MOD += -DWIZCHIP_PREFIXED_EXPORTS=1
ifeq ($(MICROPY_PY_LWIP),1)
# When using MACRAW mode (with lwIP), maximum buffer space must be used for the raw socket
CFLAGS_MOD += -DWIZCHIP_USE_MAX_BUFFER
endif
SRC_MOD += extmod/network_wiznet5k.c # modnwwiznet5k.c
SRC_MOD += $(addprefix $(WIZNET5K_DIR)/,\
	Ethernet/w$(MICROPY_PY_NETWORK_WIZNET5K)/w$(MICROPY_PY_NETWORK_WIZNET5K).c \
	Ethernet/wizchip_conf.c \
	Ethernet/socket.c \
	Internet/dns/dns.c \
	Internet/DHCP/dhcp.c \
	)
endif
```
The only remaining link error is `machine_spi_type`. I don't know where it's coming from.

robert-hh, 2022-08-03:
You have to add it in modmachine.h, just like the other declarations like e.g., as machine_i2s_type.

robert-hh, 2022-08-03:
This line:
`	Ethernet/w$(MICROPY_PY_NETWORK_WIZNET5K)/w$(MICROPY_PY_NETWORK_WIZNET5K).c \`
should be
`	Ethernet/W$(MICROPY_PY_NETWORK_WIZNET5K)/w$(MICROPY_PY_NETWORK_WIZNET5K).c \`

robert-hh, 2022-08-03:
No, it's different. machine_spi_type does not exist. It is called machine_hard_spi_type.

robert-hh, 2022-08-03:
So you may add to modmachine.h:
```
extern const mp_obj_type_t machine_hard_spi_type;

#define machine_spi_type machine_hard_spi_type
```

omogenot, 2022-08-03:
@robert-hh 
I added 
```
#define machine_spi_type    machine_hard_spi_type
```
to `stm32/modmachine.h` et voilà !! It compiles and links

omogenot, 2022-08-03:
@robert-hh 
Arghh. I type too slowly !!

omogenot, 2022-08-03:
@robert-hh 
OK. Here are my mods:
* Remove `drivers/wiznet5k` directory
* Remove `stm32/modnwwiznet5k.c` file
* Remove `stm32/network_wiznet5k.c`file
* Change Makefile (as discussed above)
* Update `stm32/modmachine.h`
Let me know if I can help any further. Good testing :-)

robert-hh, 2022-08-03:
I'm wodering how you came across these compile errors:
```
./../extmod/network_wiznet5k.c: In function 'wiznet5k_config_interrupt':
../../extmod/network_wiznet5k.c:193:5: error: implicit declaration of function 'mp_hal_pin_interrupt'; did you mean 'mp_hal_pin_input'? [-Werror=implicit-function-declaration]
  193 |     mp_hal_pin_interrupt(
      |     ^~~~~~~~~~~~~~~~~~~~
      |     mp_hal_pin_input
CC build-PYBV11/pins_PYBV11.c
../../extmod/network_wiznet5k.c:196:20: error: 'MP_HAL_PIN_TRIGGER_FALL' undeclared (first use in this function); did you mean 'MP_HAL_PIN_MODE_ALT'?
  196 |         (enabled)? MP_HAL_PIN_TRIGGER_FALL : MP_HAL_PIN_TRIGGER_NONE,
      |                    ^~~~~~~~~~~~~~~~~~~~~~~
      |                    MP_HAL_PIN_MODE_ALT
../../extmod/network_wiznet5k.c:196:20: note: each undeclared identifier is reported only once for each function it appears in
CC build-PYBV11/frozen_content.c
../../extmod/network_wiznet5k.c:196:46: error: 'MP_HAL_PIN_TRIGGER_NONE' undeclared (first use in this function); did you mean 'MP_HAL_PIN_PULL_NONE'?
  196 |         (enabled)? MP_HAL_PIN_TRIGGER_FALL : MP_HAL_PIN_TRIGGER_NONE,
      |                                              ^~~~~~~~~~~~~~~~~~~~~~~
      |                                              MP_HAL_PIN_PULL_NONE
```

omogenot, 2022-08-03:
@robert-hh 
Hum .... Funny. I cleaned up everything and tried again `make BOARD=PYBV11 MICROPY_PY_NETWORK_WIZNET5K=5500` and it compile and linked perfectly.
I push the .dfu file to the store link for you to test... Oops, looks like I have no rights to write to the storage site.
I try to clean it again an recompile with `make BOARD=PYBV11 MICROPY_PY_NETWORK_WIZNET5K=5200` to see if there is a difference.

omogenot, 2022-08-03:
@robert-hh 
Works as well.
```
make BOARD=PYBV11 clean                           
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
rm -rf build-PYBV11 

make BOARD=PYBV11 MICROPY_PY_NETWORK_WIZNET5K=5200
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
mkdir -p build-PYBV11/genhdr
GEN build-PYBV11/genhdr/pins.h
GEN stmconst build-PYBV11/modstm_qstr.h
GEN build-PYBV11/genhdr/pybcdc.inf
GEN build-PYBV11/genhdr/pybcdc_inf.h
GEN build-PYBV11/genhdr/pllfreqtable.h
GEN build-PYBV11/genhdr/mpversion.h
GEN build-PYBV11/genhdr/qstr.i.last
GEN build-PYBV11/genhdr/qstr.split
GEN build-PYBV11/genhdr/qstrdefs.collected.h
QSTR updated
GEN build-PYBV11/genhdr/qstrdefs.generated.h
GEN build-PYBV11/genhdr/moduledefs.split
GEN build-PYBV11/genhdr/moduledefs.collected
Module registrations updated
GEN build-PYBV11/genhdr/moduledefs.h
GEN build-PYBV11/genhdr/root_pointers.split
GEN build-PYBV11/genhdr/root_pointers.collected
Root pointer registrations updated
GEN build-PYBV11/genhdr/root_pointers.h
GEN build-PYBV11/genhdr/compressed.split
GEN build-PYBV11/genhdr/compressed.collected
Compressed data updated
GEN build-PYBV11/genhdr/compressed.data.h
mkdir -p build-PYBV11/build-PYBV11
mkdir -p build-PYBV11/drivers/bus
mkdir -p build-PYBV11/drivers/dht
mkdir -p build-PYBV11/drivers/memory
mkdir -p build-PYBV11/extmod
mkdir -p build-PYBV11/lib/libm
mkdir -p build-PYBV11/lib/littlefs
mkdir -p build-PYBV11/lib/oofatfs
mkdir -p build-PYBV11/lib/stm32lib/CMSIS/STM32F4xx/Source/Templates
mkdir -p build-PYBV11/lib/stm32lib/CMSIS/STM32F4xx/Source/Templates/gcc
mkdir -p build-PYBV11/lib/stm32lib/STM32F4xx_HAL_Driver/Src
mkdir -p build-PYBV11/lib/wiznet5k/Ethernet
mkdir -p build-PYBV11/lib/wiznet5k/Ethernet/W5200
mkdir -p build-PYBV11/lib/wiznet5k/Internet/DHCP
mkdir -p build-PYBV11/lib/wiznet5k/Internet/dns
mkdir -p build-PYBV11/py
mkdir -p build-PYBV11/shared/libc
mkdir -p build-PYBV11/shared/netutils
mkdir -p build-PYBV11/shared/readline
mkdir -p build-PYBV11/shared/runtime
mkdir -p build-PYBV11/shared/timeutils
mkdir -p build-PYBV11/usbdev/class/src
mkdir -p build-PYBV11/usbdev/core/src
AS ../../lib/stm32lib/CMSIS/STM32F4xx/Source/Templates/gcc/startup_stm32f405xx.s
CC ../../lib/stm32lib/CMSIS/STM32F4xx/Source/Templates/system_stm32f4xx.c
CC system_stm32.c
AS resethandler.s
AS ../../shared/runtime/gchelper_m3.s
CC ../../py/mpstate.c
CC ../../py/nlr.c
CC ../../py/nlrx86.c
CC ../../py/nlrx64.c
CC ../../py/nlrthumb.c
CC ../../py/nlraarch64.c
CC ../../py/nlrpowerpc.c
CC ../../py/nlrxtensa.c
CC ../../py/nlrsetjmp.c
[... 313 lines cut ...]
CC fatfs_port.c
CC lcd.c
CC accel.c
CC servo.c
CC dac.c
CC adc.c
CC ../../lib/oofatfs/ff.c
CC ../../lib/oofatfs/ffunicode.c
CC ../../lib/littlefs/lfs2.c
CC ../../lib/littlefs/lfs2_util.c
CC ../../extmod/network_wiznet5k.c
CC ../../lib/wiznet5k/Ethernet/W5200/w5200.c
CC ../../lib/wiznet5k/Ethernet/wizchip_conf.c
CC ../../lib/wiznet5k/Ethernet/socket.c
CC ../../lib/wiznet5k/Internet/dns/dns.c
CC ../../lib/wiznet5k/Internet/DHCP/dhcp.c
CC build-PYBV11/pins_PYBV11.c
LINK build-PYBV11/firmware.elf
   text    data     bss     dec     hex filename
 379892      92   27616  407600   63830 build-PYBV11/firmware.elf
GEN build-PYBV11/firmware0.bin
GEN build-PYBV11/firmware1.bin
GEN build-PYBV11/firmware.dfu
GEN build-PYBV11/firmware.hex
```

robert-hh, 2022-08-03:
You have built for W5200. I have a W5500 breakout.

omogenot, 2022-08-03:
@robert-hh 
Works for 5500 as well:
```
make BOARD=PYBV11 clean                           
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
rm -rf build-PYBV11 

make BOARD=PYBV11 MICROPY_PY_NETWORK_WIZNET5K=5500
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
mkdir -p build-PYBV11/genhdr
GEN build-PYBV11/genhdr/pins.h
GEN stmconst build-PYBV11/modstm_qstr.h
GEN build-PYBV11/genhdr/pybcdc.inf
GEN build-PYBV11/genhdr/pybcdc_inf.h
GEN build-PYBV11/genhdr/pllfreqtable.h
GEN build-PYBV11/genhdr/mpversion.h
GEN build-PYBV11/genhdr/qstr.i.last
GEN build-PYBV11/genhdr/qstr.split
GEN build-PYBV11/genhdr/qstrdefs.collected.h
QSTR updated
GEN build-PYBV11/genhdr/qstrdefs.generated.h
GEN build-PYBV11/genhdr/moduledefs.split
GEN build-PYBV11/genhdr/moduledefs.collected
Module registrations updated
GEN build-PYBV11/genhdr/moduledefs.h
GEN build-PYBV11/genhdr/root_pointers.split
GEN build-PYBV11/genhdr/root_pointers.collected
Root pointer registrations updated
GEN build-PYBV11/genhdr/root_pointers.h
GEN build-PYBV11/genhdr/compressed.split
GEN build-PYBV11/genhdr/compressed.collected
Compressed data updated
GEN build-PYBV11/genhdr/compressed.data.h
mkdir -p build-PYBV11/build-PYBV11
mkdir -p build-PYBV11/drivers/bus
mkdir -p build-PYBV11/drivers/dht
mkdir -p build-PYBV11/drivers/memory
mkdir -p build-PYBV11/extmod
mkdir -p build-PYBV11/lib/libm
mkdir -p build-PYBV11/lib/littlefs
mkdir -p build-PYBV11/lib/oofatfs
mkdir -p build-PYBV11/lib/stm32lib/CMSIS/STM32F4xx/Source/Templates
mkdir -p build-PYBV11/lib/stm32lib/CMSIS/STM32F4xx/Source/Templates/gcc
mkdir -p build-PYBV11/lib/stm32lib/STM32F4xx_HAL_Driver/Src
mkdir -p build-PYBV11/lib/wiznet5k/Ethernet
mkdir -p build-PYBV11/lib/wiznet5k/Ethernet/W5500
mkdir -p build-PYBV11/lib/wiznet5k/Internet/DHCP
mkdir -p build-PYBV11/lib/wiznet5k/Internet/dns
mkdir -p build-PYBV11/py
mkdir -p build-PYBV11/shared/libc
mkdir -p build-PYBV11/shared/netutils
mkdir -p build-PYBV11/shared/readline
mkdir -p build-PYBV11/shared/runtime
mkdir -p build-PYBV11/shared/timeutils
mkdir -p build-PYBV11/usbdev/class/src
mkdir -p build-PYBV11/usbdev/core/src
AS ../../lib/stm32lib/CMSIS/STM32F4xx/Source/Templates/gcc/startup_stm32f405xx.s
CC ../../lib/stm32lib/CMSIS/STM32F4xx/Source/Templates/system_stm32f4xx.c
CC system_stm32.c
AS resethandler.s
AS ../../shared/runtime/gchelper_m3.s
CC ../../py/mpstate.c
CC ../../py/nlr.c
CC ../../py/nlrx86.c
CC ../../py/nlrx64.c
CC ../../py/nlrthumb.c
CC ../../py/nlraarch64.c
CC ../../py/nlrpowerpc.c
CC ../../py/nlrxtensa.c
CC ../../py/nlrsetjmp.c
[... 314 lines cut ...]
CC lcd.c
CC accel.c
CC servo.c
CC dac.c
CC adc.c
CC ../../lib/oofatfs/ff.c
CC ../../lib/oofatfs/ffunicode.c
CC ../../lib/littlefs/lfs2.c
CC ../../lib/littlefs/lfs2_util.c
CC ../../extmod/network_wiznet5k.c
CC ../../lib/wiznet5k/Ethernet/W5500/w5500.c
CC ../../lib/wiznet5k/Ethernet/wizchip_conf.c
CC ../../lib/wiznet5k/Ethernet/socket.c
CC ../../lib/wiznet5k/Internet/dns/dns.c
CC ../../lib/wiznet5k/Internet/DHCP/dhcp.c
CC build-PYBV11/pins_PYBV11.c
LINK build-PYBV11/firmware.elf
   text    data     bss     dec     hex filename
 379968      92   27596  407656   63868 build-PYBV11/firmware.elf
GEN build-PYBV11/firmware0.bin
GEN build-PYBV11/firmware1.bin
GEN build-PYBV11/firmware.dfu
GEN build-PYBV11/firmware.hex
```
Try may be 'make submodules' in case you working tree contains older git clones (we never know).

robert-hh, 2022-08-03:
You have built without LWIP.

robert-hh, 2022-08-03:
When building w/o lwip, it succeeds. On network start, I get the following error with 'nic.active(True)`:
```
OSError: 16
MPY: enabling IRQs
```

omogenot, 2022-08-03:
@robert-hh 
Sorry. I have to stop for tonight. I'll resume tomorrow.
To be continued...

robert-hh, 2022-08-03:
It seems to be a problem with every attempt trying to communicate with the W5500. The message `MPY: enabling IRQs` is in shared/runtime/pyexec.c, function pyexec_friendly_repl(). So OSError: 16 is the topic to look at,

robert-hh, 2022-08-04:
The problem is, that a wrong data type is supplied for SPI  to the WIZNET driver. Then, the spi_transfer() function does not have the reference it needs, and no SPI transfer takes place.

omogenot, 2022-08-04:
@robert-hh 
I came to the same conclusion this morning, and was having a look at that. Unfortunately, I am currently on vacation and I have limited access to my computer and can't really spend a lot of time for the moment on that topic. I'll come back as soon as I can. :-)

robert-hh, 2022-08-04:
I found the reason: It is twice a substitution.
In stm32/spi.c, line 684, the spi object is returned as self->spi, where self is the argument of the network_wiznet constructor. That is the pointer to be used in SPI transfer calls. and it is store in the wiznetwiznet5k_obj. BUT, in wiz_spi_write() machine_hard_spi_transfer() is called with that value for the transfer, and that uses <input Ptr>->spi.
machine_hard_spi_transfer() cannot be changed, 
One coulf change the function spi_from_mp_obj(), which is also used by modnwcc3k.c. That is a expired WiFi adapter.

robert-hh, 2022-08-04:
The actual states is here: https://github.com/robert-hh/micropython/tree/stm32_wizlib
A sample start script is:
```
import network
import machine, pyb
spi=machine.SPI(1)
nic = network.WIZNET5K(spi, machine.Pin("X5"), machine.Pin("X4"))
nic.active(1)
nic.ifconfig(('10.0.0.139', '255.255.255.0','10.0.0.240', '10.0.0.240'))
```
This is without LWIP. For LWIP, the pin interrupt functionality has to be added.

robert-hh, 2022-08-04:
I added the definitions for LWIP and the interrupt pin. Of course, the fixes made in the PR for extmod_wirznet5k.c are not included.

omogenot, 2022-08-04:
@robert-hh 
You're the best. Congratulations.

robert-hh, 2022-08-04:
I made a single commit for adding the extmod_wirznet5k.c changes. That can be dropped later, if required. Ping times with interrupt are not overly great, like 9 ms, but better that without interrupt, which are ~40ms.

omogenot, 2022-08-05:
This has been taken care of by PR #9021.

closed 2022-08-05 by omogenot