micropython/micropython #9016: Pi Pico W fails to connect to open Wifi network

robertjensen, 2022-08-05:
When trying to connect to an open network (auth type 0), the following reply is returned:
```
wlan.connect('device')
Traceback (most recent call last):
File "<stdin>", line 1, in <module>
OSError: [Errno 1] EPERM
```

Using an empty string as password gives the same result:
```
wlan.connect('device', '')
Traceback (most recent call last):
File "<stdin>", line 1, in <module>
OSError: [Errno 1] EPERM
```

For completeness, this is how the 'device' network reports itself in a scan:
```
wlan.scan()
[.... (b'device', b'\x00,\xc8\x87\r@', 11, -77, 0, 1),
(b'eduroam', b'\x00,\xc8\x87\x0ba', 11, -84, 5, 2),
(b'device', b'\x00,\xc8\x87\x0b`', 11, -86, 0, 1),
...]

```

The device is verified to work on a normal WPA network, where it connects with no problems.

adamkaplan, 2022-08-18:
I ran into this same problem and found other unresolved citations elsewhere. I suspect the issue is related to [this line of code](https://github.com/georgerobotics/cyw43-driver/blob/2ab6ca93f9cd044bc6f35c1403b1284e4161294a/src/cyw43_ll.c#L2018) in the driver used for the CYW43 network card on the Pico W.

If I could get micropython to build on my Mac I would be able to confirm this suspicion.

robertjensen, 2022-08-19:
Thank you for finding this!

I am on Linux and I think I would be able to compile from source, but I am
not familiar with the MicroPython source code, do you have an idea of what
I should add to handle the case of open access point?

robertjensen, 2022-08-19:
I have tried to clone micropython and see if I could find the code you mention, but the repo does not contain this particular file:

/micropython/drivers/cyw43$ ls
cyw43_ctrl.c  cyw43.h  cyw43_ll.h  cyw43_lwip.c  cywbt.c  libcyw43.a  README.md

robertjensen, 2022-08-19:
Found the code. I have tried to re-compile with the catch-all commented out. The unit no longer crashes, but it also does not connect :-(

adamkaplan, 2022-08-19:
> I have tried to clone micropython and see if I could find the code you mention, but the repo does not contain this particular file
I should have mentioned that it was included via submodule from the external driver repository.

> Found the code. I have tried to re-compile with the catch-all commented out. The unit no longer crashes, but it also does not connect :-(

That's progress! It stands to reason that they haven't tested this code path since it was broken for so long. I see other lines setting up authentication down below the return.

My next hunch would be that these are [configuring an auth method without providing a key](https://github.com/georgerobotics/cyw43-driver/blob/2ab6ca93f9cd044bc6f35c1403b1284e4161294a/src/cyw43_ll.c#L2057). But at this point the docs for the underlying hardware & C APIs would be the next stop.

adamkaplan, 2022-08-21:
I was able to build micropython by fixing a missing time.h import in the SSL submodule (different issues). I also can’t get it to work be removing that return -1.

i did some debugging by commenting out most of the method code and only keeping the minimum calls, hard coded with open WiFi values and cross referenced with examples from the ZeroWifi driver (which is deprecated). I can see that a 802.11 status is returned from the AP that appears to be code 5, meaning “security disabled”. I don’t know if that is a good or bad thing to get after initiating the connection, but after that the interface goes down (-1 from the WLAN status method).

I spent a few hours on this. Debugging is impossible using Pico given the lack of debug output or breakpoints. And my lack of expertise in 802.11… need more examples.

robertjensen, 2022-08-21:
Thank you so much for your efforts. Fixing this is also significantly out
of my comfort zone, so I guess we can only hope a more experienced
maintainer steps in with some help, I assume?

sandu-bogdan, 2022-08-23:
I have the same bug. I cannot connect to the Wireless if i not have a security on SSID.

cebess, 2022-08-29:
I have the same issue. I don't believe that microPython on the Pico W is completely defined yet.

adamkaplan, 2022-08-29:
Technically this is a problem with the driver that Raspberry Pi has licensed from the manufacturer of the wifi chip. I have opened up a sibling issue on that project https://github.com/georgerobotics/cyw43-driver/issues/27

gwo0d, 2022-11-23:
I have also noticed this issue.

gwo0d, 2022-11-23:
Patched [here](https://github.com/georgerobotics/cyw43-driver/pull/45).

robertjensen, 2022-11-23:
This is amazing news! Thank you so very much! Could you give a brief hint as to how to make a local checkout that includes the patch, I have not fully understood how the cyw43 is pulled in during the install process?

gwo0d, 2022-11-23:
@robertjensen your best bet is to clone the `cyw43-driver` repository directly from my PR to get a copy with my patch.

Then, run the build as usual (as explained in the `/ports/rp2/README.md`) file. However, after you run the `make BOARD=PICO_W submodules` command, stop and copy your `cyw43-driver` clone from my pull and use it to replace `lib/cyw43-driver` in your local MicroPython repo.

Having done that, run `make BOARD=PICO_W` and upload the firmware to your Pi as usual.

Note: there is still an issue with the MicroPython network module which means you will only be able to connect to the network using the security parameter like this:
```python
import network

wlan = network.WLAN(network.STA_IF)
wlan.active(True)
wlan.connect(<your SSID>, security=0)
...
```

I'll be looking into this issue asap to find a patch.

Hope this helps!

adamkaplan, 2022-11-24:
@gwo0d nice! Thanks for fixing this

gwo0d, 2022-11-24:
@LukeBriggsDev thanks for this! Really nice update.

TheSeaUrchin, 2022-12-05:
> @robertjensen your best bet is to clone the `cyw43-driver` repository directly from my PR to get a copy with my patch.
> 
> Then, run the build as usual (as explained in the `/ports/rp2/README.md`) file. However, after you run the `make BOARD=PICO_W submodules` command, stop and copy your `cyw43-driver` clone from my pull and use it to replace `lib/cyw43-driver` in your local MicroPython repo.
> 
> Having done that, run `make BOARD=PICO_W` and upload the firmware to your Pi as usual.
> 
> Note: there is still an issue with the MicroPython network module which means you will only be able to connect to the network using the security parameter like this:
> 
> ```python
> import network
> 
> 
> ssid = 'Your SSID'
> wlan = network.WLAN(network.STA_IF)
> wlan.active(True)
> wlan.connect(ssid, security=0)
> ...
> ```

I tried doing exactly this but I'm getting the same results

caprizone6, 2022-12-13:
> I tried doing exactly this but I'm getting the same results

The new commit in the [Pull request](https://github.com/georgerobotics/cyw43-driver/pull/45) works.

closed 2022-12-20 by dpgeorge