micropython/micropython #9035: Building Micropython - Failed to resolve component 'esp_adc_cal'

DevynCJohnson, 2022-08-09:
**Versions**:

- Latest pull from the master-branch of both "esp-idf" and "micropython".
- xtensa-esp32s2-elf-gcc: 11.2.0
- gcc (on the Ubuntu system): 11.3.0
- cmake: 3.22.1

**Build Process**:

I am running the below Linux shell script (on Ubuntu 22.04) to build Micropython for the ESP32 S2 WROVER. However, the build fails when running "`make`" on the last line of the below script.

```
git clone --recursive https://github.com/espressif/esp-idf.git &
git clone --recursive https://github.com/micropython/micropython.git &
wait
[ ! -d ./esp-idf ] && exit 1
[ ! -d ./micropython ] && exit 1
cd ./esp-idf && ./install.sh && cd ../micropython && . ../esp-idf/export.sh && cd ./mpy-cross && make && cd ../ports/esp32
export BOARD='ESP32_S2_WROVER'
export MICROPY_BOARD="${BOARD}"
export IDF_TARGET='esp32s2'
make
```

**Output from "make"**:

```
idf.py -D MICROPY_BOARD=ESP32_S2_WROVER -B build-ESP32_S2_WROVER  build
Executing action: all (aliases: build)
Running cmake in directory /media/collier/PROJECTS/ESP32/Micropython/micropython/ports/esp32/build-ESP32_S2_WROVER
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DMICROPY_BOARD=ESP32_S2_WROVER -DCCACHE_ENABLE=0 /media/collier/PROJECTS/ESP32/Micropython/micropython/ports/esp32"...
-- Found Git: /usr/bin/git (found version "2.37.0") 
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/collier/.espressif/tools/xtensa-esp32s2-elf/esp-2022r1-RC1-11.2.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/collier/.espressif/tools/xtensa-esp32s2-elf/esp-2022r1-RC1-11.2.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/collier/.espressif/tools/xtensa-esp32s2-elf/esp-2022r1-RC1-11.2.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s2
CMake Error at /media/collier/PROJECTS/ESP32/Micropython/esp-idf/tools/cmake/build.cmake:245 (message):
  Failed to resolve component 'esp_adc_cal'.
Call Stack (most recent call first):
  /media/collier/PROJECTS/ESP32/Micropython/esp-idf/tools/cmake/build.cmake:281 (__build_resolve_and_add_req)
  /media/collier/PROJECTS/ESP32/Micropython/esp-idf/tools/cmake/build.cmake:576 (__build_expand_requirements)
  /media/collier/PROJECTS/ESP32/Micropython/esp-idf/tools/cmake/project.cmake:434 (idf_build_process)
  CMakeLists.txt:58 (project)

-- Configuring incomplete, errors occurred!
See also "/media/collier/PROJECTS/ESP32/Micropython/micropython/ports/esp32/build-ESP32_S2_WROVER/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1, output of the command is in the /media/collier/PROJECTS/ESP32/Micropython/micropython/ports/esp32/build-ESP32_S2_WROVER/log/idf_py_stderr_output_1020906 and /media/collier/PROJECTS/ESP32/Micropython/micropython/ports/esp32/build-ESP32_S2_WROVER/log/idf_py_stdout_output_1020906
make: *** [Makefile:38: all] Error 2
```

jimmo, 2022-08-09:
@DevynCJohnson I don't see any where in your commands where you're checking out the IDF at a supported git hash (e.g. v4.2)
https://github.com/micropython/micropython/tree/master/ports/esp32#setting-up-esp-idf-and-the-build-environment

DevynCJohnson, 2022-08-10:
Thanks, that worked.