micropython/micropython #9039: The letter b is not shown on the serial REPL

vuolen, 2022-08-10:
I am running `esp32-20220618-v1.19.1.bin` on an `esp32-devkitc-v4` board.  Startup message:
```
MicroPython v1.19.1 on 2022-06-18; ESP32 module with ESP32
Type "help()" for more information.
```
I typed in the letter b in the REPL and pressed enter. It returned an error message with the letter b missing. Then I typed in `len("`, then pressed the b key three times, and then typed in `")`. As a result I get that the length of the string is 3, even though I cannot see the letters in the REPL.

```
>>> 
Traceack (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name '' isn't defined
>>> len("")
3
```

What could be causing this? Everything else seems to work fine.

robert-hh, 2022-08-10:
Strange. Which terminal emulator and which character set do you use with micropython?

vuolen, 2022-08-11:
I emulate with `screen`, how do I figure out the character set?

robert-hh, 2022-08-11:
Dont't know. Is the upper case B echoed? Are other characters suppressed. Do you get b echoed using mpremote as terminal?

vuolen, 2022-08-11:
Upper case B is echoed, I cant see any other characters being suppressed. Seems to work normally with mpremote.

robert-hh, 2022-08-12:
Looks like a problem of screen & Mac. So no issue here, If you have a dumb USB/UART adapter, you could connect at that adapter TX and RX and see, if all characters are echoed.

closed 2022-08-27 by robert-hh