micropython/micropython #9044: various ways to read() don't check for valid utf-8

jepler, 2022-08-12:
Over in https://github.com/adafruit/circuitpython/issues/6752 a user reported that various ways to read from files didn't actually validate that the content was utf-8. In our case, this leads to a subsequent disconnection in the webrepl when the value is printed.

I prepared a possible fix relative to CircuitPython at https://github.com/adafruit/circuitpython/pull/6754 -- is there interest in changing this behavior in micropython? if so, I can prepare a PR for y'all including hopefully adding tests.

basically, each code path in file reading will now check that the content is valid utf-8.

robert-hh, 2022-08-12:
IMHO it is not valid to assume that a file content always has to be utf-8 encoded. It may be anything. So such a check would only be reasonable if the file is opened requiring a specific encoding, like utf-8.

jimmo, 2022-08-12:
My interpretation of https://docs.python.org/3/library/functions.html#open is that by not specifying an encoding to `open()` we get the default encoding. We only support one encoding, so it makes sense that we're implicitly opening the file in utf-8 mode, as as such it does make sense that we should be doing the validation if it's a text-mode stream.

jepler, 2022-08-12:
Weird stuff starts happening when a str object has non-utf8 content. The original poster over in circuitpython found, for instance, that this value doesn't subscript right ... here, "notutf8.txt" contains the single byte \xf7:
```
>>> with open("notutf8.txt") as f: content = f.read()
... 
>>> content
'\x07'
>>> content[0]
'\x07\x00\x00\x00'
>>> ord(content[0])
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: ord() expected a character, but string of length 4 found
```

stopping weirdness like this "at the source" will hopefully save somebody time down the line, too.

robert-hh, 2022-08-12:
Doesn't the documentation say that the default of the parameter encoding is None = No encoding? And obviously that applies only to text files.

dlech, 2022-08-12:
The CPython documentation says:

>  In text mode, if encoding is not specified the encoding used is platform-dependent: locale.getpreferredencoding(False) is called to get the current locale encoding. 

The MicroPython documentation does say anything about it, AFAICT.

robert-hh, 2022-08-12:
So for MicroPython the default encoding is: No encoding.

jimmo, 2022-08-17:
> So for MicroPython the default encoding is: No encoding.

What would it mean for there to be "no encoding" for a file opened in text mode?

MicroPython's default (and only) encoding is "utf-8". To illustrate this, if I write a file out to the filesystem containing utf-8 encoded data, then read it back in with the default options, it is interpreted as utf-8. If there was no encoding, then no interpretation of the `\xe2...` into the single unicode character `\u2b50` (i.e. ⭐) would have happened.

```
MicroPython v1.19.1-290-ga1a5a6539 on 2022-08-17; PYBv1.1 with STM32F405RG
Type "help()" for more information.
>>> f = open('data', 'wb')
>>> f.write(b'hello \xe2\xad\x90 world')
15
>>> f.close()
>>>
>>> f = open('data', 'r')
>>> f.read()
'hello \u2b50 world'
>>> _[6]
'\u2b50'
>>> len(_)
1
>>> f.close()
```

In Python (and MicroPython) a (utf-8) string that appears to contain `\xe2` for example, is actually encoded as `b'\xc3\xa2'`
See https://www.utf8-chartable.de/unicode-utf8-table.pl?start=128&number=128&utf8=string-literal&unicodeinhtml=hex

```
>>> s = 'hello \xe2\xad\x90 world'   # Will be stored internally as utf-8: \xe2 -> \xc3a2
>>> s[6]  # string substring understands the encoding
'\xe2'
>>> uctypes.bytes_at(uctypes.addressof(s), 18)    # no conversion, just get the raw bytes for this object
b'hello \xc3\xa2\xc2\xad\xc2\x90 world'
```

@jepler I think your fix makes sense, but I perhaps it should be more general than just for `stream.c`.

i.e. be stricter about this and make `mp_obj_new_str_from_vstr` itself always do `utf8_check` (conditional on `MICROPY_PY_BUILTINS_STR_UNICODE_CHECK`). i.e. any method that creates or modifies a `str` object should maintain the invariant that the string is utf-8, and it should be impossible to create an invalid string.

Note that `mp_obj_str_make_new` already does this (which is why `bytes.decode` will catch this). But for example `mp_obj_new_str` and `mp_obj_new_str_copy` don't.

As an optimisation, in many cases `mp_obj_new_str_from_vstr`, `mp_obj_new_str`, and `mp_obj_new_str_copy` are working with data from already-valid string objects. Perhaps we should make those three methods always check, but then add `_safe` versions for places where you know you have valid data (which would likely be static/private to objstr.c).

```c
STATIC mp_obj_t mp_obj_new_str_copy_safe(const mp_obj_type_t *type, const byte *data, size_t len) {
}
mp_obj_t mp_obj_new_str_copy(const mp_obj_type_t *type, const byte *data, size_t len) {
    #if MICROPY_PY_BUILTINS_STR_UNICODE_CHECK
    if (!utf8_check(str_data, str_len)) {
        mp_raise_msg(&mp_type_UnicodeError, NULL);
    }
    #endif
    return mp_obj_new_str_copy_safe(type, data, len);
}
```

It was easy to find an example of being able to create an invalid string due to `mp_obj_new_str` being used incorrectly, e.g. on Pico W via using `sta_if.config(ssid=b'..invalid utf8 encoding..')` then reading it back with `sta_if.config('ssid')`.

jimmo, 2022-08-22:
See https://github.com/micropython/micropython/pull/9083 for a proposed fix.

dpgeorge, 2022-08-26:
Fixed by above PR.

closed 2022-08-26 by dpgeorge