micropython/micropython #9047: Default SPI pins incorrect according to SDK

mcriley821, 2022-08-13:
When creating a default SPI0 instance, the default MOSI, MISO, and SCK pins do not reflect any datasheets nor the C/C++ SDK for the pico and pico w. 
I'm new to embedded development, but not to software development in general. Embarrassingly, it took me longer than it should have to figure out why `machine.SPI(0)` was not communicating on the GPIOs specified in the datasheet.

## Observation:
```python
>>> import machine
>>> machine.SPI(0)
SPI(0, baudrate=992063, polarity=0, phase=0, bits=8, sck=6, mosi=7, miso=4)
```

## SDKs v1.4.0:
Both clearly outline default SPI0 as SCK=18, MOSI=19, and MISO=16

[pico SDK permalink: pico-sdk/src/boards/include/boards/pico.h](https://github.com/raspberrypi/pico-sdk/blob/2e6142b15b8a75c1227dd3edbe839193b2bf9041/src/boards/include/boards/pico.h#L49-L63)
[pico w SDK permalink: pico-sdk/src/boards/include/boards/pico_w.h](https://github.com/raspberrypi/pico-sdk/blob/2e6142b15b8a75c1227dd3edbe839193b2bf9041/src/boards/include/boards/pico_w.h#L46-L61)

## Documentations:
[Raspberry Pi Pico Datasheet](https://datasheets.raspberrypi.com/pico/pico-datasheet.pdf#page=5)
[Raspberry Pi Pico W Datasheet](https://datasheets.raspberrypi.com/picow/pico-w-datasheet.pdf#page=6)

## Issue:
The issue arises from the hardcoded pin values here: https://github.com/micropython/micropython/blob/6f4d424f461bede1afb69637763f4fce5f27cd90/ports/rp2/machine_spi.c#L42-L46

## Suggested fix:
```C
 #ifndef MICROPY_HW_SPI0_SCK 
 #define MICROPY_HW_SPI0_SCK     PICO_DEFAULT_SPI_SCK_PIN  
 #define MICROPY_HW_SPI0_MOSI    PICO_DEFAULT_SPI_TX_PIN  
 #define MICROPY_HW_SPI0_MISO    PICO_DEFAULT_SPI_RX_PIN  
 #endif 
```

## Firmware:
Using UF2: unstable-v1.19.1-240 nightly build from https://micropython.org/download/rp2-pico-w/
From REPL: MicroPython v1.19.1 on 2022-08-05; Raspberry Pi Pico W with RP2040

pdg137, 2022-11-24:
Looks like this issue got "fixed" in this commit:

https://github.com/micropython/micropython/commit/29437205f29976db71dc12d61ce756732e6f813a

However, this is a breaking change (it broke my code, for example, in a place that wasn't even using SPI, by overriding my usage of GP16), and I don't understand the benefit of having these defaults. Why doesn't Micropython just require the SPI pins to be specified explicitly? You have to know the pin numbers to use the ports, so what are you saving by not having to type them into your code?

jimmo, 2022-11-25:
@pdg137 The RP/Pico documentation describes default SPI and I2C pins, so the idea is to match that (exactly as @mcriley821 describes). This is also consistent with other ports -- for example on STM32 you cannot specify pins if you're using hardware the I2C/SPI peripherals.

On something like a Pico, you're right about having to know the pins anyway, but this isn't always the case -- e.g. if you have a dedicated Grove/Gravity/Stemma/Qwiik/whatever port labelled "I2C0" then it "just works" when you use `machine.I2C(0)`.

This was a difficult one, and with any of the three options we had (do nothing, make the change we did, remove default pins) we would have made someone unhappy. Fortunately, although surprising and likely difficult to diagnose, the software fix is straightforward and doesn't require any physical/electrical changes.

@mcriley821 Thanks for reporting. Closing this because I think your original issue is now addressed by https://github.com/micropython/micropython/commit/29437205f29976db71dc12d61ce756732e6f813a. Please re-open if I've misunderstood.

closed 2022-11-25 by jimmo

DavidEGrayson, 2022-11-25:
There are many SPI devices with a clock input and a data input but without a data output, so it would be good if the SPI library supported using them without requiring the user to jump through any extra hoops.

The current implementation insists on choosing a valid pin for SCK, MISO, and MOSI, and then it calls `gpio_set_function` to configure each of those pins.

How about this: If the `sck`, `miso`, and `mosi` arguments are all `None`, then the SPI library should pick default pins for SCK, MISO, and MOSI.  If any pin argument is not `None`, then that means the user wants to control what pins are being used, and the SPI library should refrain from picking any defaults.  Any pin not specified by the user will not be configured.