micropython/micropython #9052: Unable to Flash MicroPython on ESP32-C3

vivekmistry, 2022-08-14:
Hello,

I have built firmware.bin using the following commands with a small modification, at the last step while executing "make" command, it fails while compiling "usb_serial_jtag.c" because at the line it is expecting the value "USB_SERIAL_JTAG_INTR_SOF" instead of "ETS_USB_SERIAL_JTAG_INTR_SOURCE":

#!/bin/bash

export BUILD_DIR=$(pwd)

echo "--- CLONING MICROPYTHON ---"
git clone --depth 1 https://github.com/micropython/micropython.git

echo "--- CLONING ESP-IDF ---"
cd $BUILD_DIR/micropython/
git clone -b v4.3.2 --recursive https://github.com/espressif/esp-idf.git

echo "--- INSTALL ESP-IDF ---"
cd $BUILD_DIR/micropython/esp-idf
./install.sh
. ./export.sh

echo "--- MPY-CROSS ---"
cd $BUILD_DIR/micropython/mpy-cross
make

echo "--- ESP32 SUBMODULES ---"
cd $BUILD_DIR/micropython/ports/esp32
make submodules

echo "--- PATCH MAKEFILE ---"
cp $BUILD_DIR/micropython/ports/esp32/Makefile $BUILD_DIR/micropython/ports/esp32/MakefileOld
echo "BOARD = GENERIC_C3_USB" > $BUILD_DIR/micropython/ports/esp32/Makefile
cat $BUILD_DIR/micropython/ports/esp32/MakefileOld >> $BUILD_DIR/micropython/ports/esp32/Makefile

echo "--- MAKE ---"
make

The binary file "firmware.bin" is generated and it is written to ESP32-C3 successfully. However, while flashing it gives following error:

ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xe (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0xf18
load:0x403ce000,len:0x6cc
load:0x403d0000,len:0x292c
entry 0x403ce000

And, if I try to write & flash binary from micropython site(https://micropython.org/download/esp32c3-usb/) it works without any issue. Please advise, what is wrong above?

jimmo, 2022-08-15:
> git clone -b v4.3.2 --recursive https://github.com/espressif/esp-idf.git

You need to use IDF v4.4 + to use the C3.

vivekmistry, 2022-08-15:
Now use -v4.4 and got this error:

ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xe (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0xe3c
load:0x403ce000,len:0x6dc
load:0x403d0000,len:0x28c4
entry 0x403ce000

igrr, 2022-08-19:
This part looks a bit weird,
```
cp $BUILD_DIR/micropython/ports/esp32/Makefile $BUILD_DIR/micropython/ports/esp32/MakefileOld
echo "BOARD = GENERIC_C3_USB" > $BUILD_DIR/micropython/ports/esp32/Makefile
cat $BUILD_DIR/micropython/ports/esp32/MakefileOld >> $BUILD_DIR/micropython/ports/esp32/Makefile
```

You can instead use `make BOARD=GENERIC_C3_USB`.

Please check if the sdkconfig file contains `CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y`. If not, please run `idf.py fullclean` and then build again.