micropython/micropython #9065: Use SDCard as external heap for mbedTLS

AmirHmZz, 2022-08-17:
I just wonder If it possible to use SDCard as an external heap memory for some features like mbedTLS or even Micropython heap instead. This feature seems to be something like [ReadyBoost](https://en.wikipedia.org/wiki/ReadyBoost) feature of Microsoft. The main reason that this comes to my mind is in some countries access to a suitable PSRAM for a specific MCU is impossible. For e.g. in our country ESP32 supporting PSRAMs are unavailable in stocks and it is also impossible to ship them from mouser or any other website. I just wonder to know is it possible to something like this and If the answer is yes, how much work this might need to be implemented ?

AmirHmZz, 2022-08-17:
There's a positive point about using SDCard as a memory for mbedTLS : There's no need to `gc` to `collect()` allocated objects from SDCard. The whole allocated heap can be deallocated when TLS connection is done at once. As long as we have more space on SDCards in comparison with PSRAMs there's no need to `gc.collect()` on heaps allocated on SDCard frequently.

jimmo, 2022-08-17:
I should add that if you really wanted to do this (seriously though, don't) then a better approach would be to use the spi flash rather than sd card, because at least then it's memory mapped for read.

So for example, whenever mbedtls _wrote_ to a buffer, you'd have to rewrite the code to go via a wrapper than turned it into flash write operations. But then at least you could still take read-only pointers to it.

But I'm just saying this for completeness. It's not a good idea.